Koenkk/zigbee2mqtt #20446: [New device support]: Schneider Electric Exxact Wiser Floor Heating Thermostat

Witomas, 2023-12-30:
### Link

https://www.se.com/se/sv/product/WDE002497/uppkopplad-termostat-exxact-elektronisk-16a-golv-rum-kan-styras-via-wiser-by-seappen-zigbee-3-0-vit/

### Database entry

{"id":32,"type":"Router","ieeeAddr":"0x187a3efffe7fbf39","nwkAddr":6302,"manufId":4190,"manufName":"Schneider Electric","powerSource":"Mains (single phase)","modelId":"WDE002497","epList":[1,2,3,5,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":769,"inClusterList":[0,3,4,513,516,2821,65027,65302,65315],"outClusterList":[3,10,25,1026,1030],"clusters":{"genBasic":{"attributes":{"modelId":"WDE002497","manufacturerName":"Schneider Electric","powerSource":1,"zclVersion":8}},"hvacUserInterfaceCfg":{"attributes":{"keypadLockout":0,"tempDisplayMode":0}},"hvacThermostat":{"attributes":{"57362":0,"57873":0,"localTemp":1468,"pICoolingDemand":0,"systemMode":4,"pIHeatingDemand":100,"occupiedHeatingSetpoint":2200}},"schneiderSpecificPilotMode":{"attributes":{"pilotMode":1}}},"binds":[{"cluster":513,"type":"endpoint","deviceIeeeAddress":"0x00124b0026b82ff2","endpointID":1}],"configuredReportings":[{"cluster":513,"attrId":8,"minRepIntval":0,"maxRepIntval":3600,"repChange":10,"manufacturerCode":null},{"cluster":513,"attrId":18,"minRepIntval":0,"maxRepIntval":3600,"repChange":10,"manufacturerCode":null}],"meta":{}},"2":{"profId":260,"epId":2,"devId":770,"inClusterList":[0,3,1026],"outClusterList":[3],"clusters":{"genBasic":{"attributes":{}}},"binds":[{"cluster":1794,"type":"endpoint","deviceIeeeAddress":"0x00124b0026b82ff2","endpointID":1}],"configuredReportings":[],"meta":{}},"3":{"profId":260,"epId":3,"devId":770,"inClusterList":[0,3,1026],"outClusterList":[3],"clusters":{"genBasic":{"attributes":{}}},"binds":[],"configuredReportings":[],"meta":{}},"5":{"profId":260,"epId":5,"devId":13,"inClusterList":[0,3,4,1794],"outClusterList":[],"clusters":{"genBasic":{"attributes":{}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"zclVersion":8,"interviewCompleted":true,"meta":{"configured":1253322000},"lastSeen":1703950094549,"defaultSendRequestWhen":"immediate"}

### Comments

Tried to copy EKO07259 converter, System mode On/Off works, but further features are out of my expertise.
Please help.

### External converter

```shell
/** EKO07259 converter */
```

jopppis, 2024-01-07:
I got the same device and tried to figure out what is actually missing. It seems that there is probably no electricity metering and the manual mentions that changing the program is not supported except after factory reset and when connected to the Schneider Hub.

There seems to be three msTemperatureMeasurement clusters so maybe that means there should be three different temperature readings (device temp, floor sensor and something else maybe)?

pekkeri, 2024-02-03:
Endpoint 2 seems to have room temperature (internal temperature measurement) in msTemperatureMeasurement cluster.

Endpoint 3 seems to have floor temperature (external temperature measurement) in msTemperatureMeasurement cluster.

Endpoint 1 contains the thermostat function itself. Temperature shown in cluster there will be either internal or external measurement depending on the configured program (presetting) of the device.

There is no real electricity metering. Calculated metering is in Endpoint 5, but it is based only on the value of heating power entered during setup.

Program change means how it works (floor, room, floor/room etc.). This configuration can be set from device itself, and does not make much sense to change afterwards.

In hvacThermostat there is extra attribute 57362 showing in logs:
`Received Zigbee message from '0x8c6fb9fffe2d0125', type 'attributeReport', cluster 'hvacThermostat', data '{"57362":0}' from endpoint 1 with groupID 0`

Witomas, 2024-02-11:
> Endpoint 2 seems to have room temperature (internal temperature measurement) in msTemperatureMeasurement cluster.
> Endpoint 3 seems to have floor temperature (external temperature measurement) in msTemperatureMeasurement cluster.

@jopppis , @pekkeri , did you have time to compile a working converter? I would very much appreciate if you would share...

pekkeri, 2024-02-11:
This is so far working ok for me, but I'm not sure if it's perfect yet.

```

const {} = require('zigbee-herdsman-converters/lib/modernExtend');
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
        zigbeeModel: ['WDE002497'],
        model: 'WDE002497',
        vendor: 'Schneider Electric',
        description: 'Smart thermostat',
        fromZigbee: [fz.stelpro_thermostat, fz.metering, fz.schneider_pilot_mode, fz.wiser_device_info, fz.hvac_user_interface, fz.temperature],
        toZigbee: [tz.thermostat_occupied_heating_setpoint, tz.thermostat_system_mode, tz.thermostat_running_state,
            tz.thermostat_local_temperature, tz.thermostat_control_sequence_of_operation, tz.schneider_pilot_mode,
            tz.schneider_thermostat_keypad_lockout, tz.thermostat_temperature_display_mode, tz.temperature],
        exposes: [e.binary('keypad_lockout', ea.STATE_SET, 'lock1', 'unlock')
            .withDescription('Enables/disables physical input on the device'),
        e.enum('schneider_pilot_mode', ea.ALL, ['contactor', 'pilot']).withDescription('Controls piloting mode'),
        e.enum('temperature_display_mode', ea.ALL, ['celsius', 'fahrenheit'])
            .withDescription('The temperature format displayed on the thermostat screen'),
        e.climate().withSetpoint('occupied_heating_setpoint', 4, 30, 0.5).withLocalTemperature()
            .withSystemMode(['off', 'heat']).withRunningState(['idle', 'heat']).withPiHeatingDemand(),
        e.temperature()],
        configure: async (device, coordinatorEndpoint, logger) => {
            const endpoint1 = device.getEndpoint(1);
            const endpoint2 = device.getEndpoint(2);
            const endpoint5 = device.getEndpoint(5);
            await reporting.bind(endpoint1, coordinatorEndpoint, ['hvacThermostat']);
            await reporting.thermostatPIHeatingDemand(endpoint1);
            await reporting.thermostatOccupiedHeatingSetpoint(endpoint1);
            await reporting.temperature(endpoint2);
            await endpoint1.read('hvacUserInterfaceCfg', ['keypadLockout', 'tempDisplayMode']);
        },

};

module.exports = definition;

```

Witomas, 2024-02-12:
Great thanks @pekkeri!

jopppis, 2024-02-12:
> This is so far working ok for me, but I'm not sure if it's perfect yet.

I am using a similar config but I would really like to add at least the floor sensor to that. I have tried but so far not been able to configure it successfully.

> Endpoint 2 seems to have room temperature (internal temperature measurement) in msTemperatureMeasurement cluster.
Endpoint 3 seems to have floor temperature (external temperature measurement) in msTemperatureMeasurement cluster.
How did you determine which is which from those?

> Program change means how it works (floor, room, floor/room etc.). This configuration can be set from device itself, and does not make much sense to change afterwards.
I would like this feature to be able to sometimes set the setpoint for floor temperature (like when electricity is extremely cheap and I want to reserve as much heat as possible in the floor before prices go up) while most of the time using air temperature for setpoint. This might not be possible though.

They publish [a compliance document](https://csa-iot.org/csa_product/exxact-connected-thermostat-room-floor-white-7/) that has some information but I was not able to extract anything useful from there.

pekkeri, 2024-02-12:
> > This is so far working ok for me, but I'm not sure if it's perfect yet.
> 
> I am using a similar config but I would really like to add at least the floor sensor to that. I have tried but so far not been able to configure it successfully.

Try to change the lines with endpoint 2 to endpoint 3 (or just the number (2) in first line),
from:
```
const endpoint2 = device.getEndpoint(2);
await reporting.temperature(endpoint2);
```
to:
```
const endpoint3 = device.getEndpoint(3);
await reporting.temperature(endpoint3);
```
Then you should have an additional temperature measurement which hopefully shows the floor temp. It must be of course connected and configured. Might also need to have a "program mode" of room thermostat with floor temp limiting.

> 
> > Endpoint 2 seems to have room temperature (internal temperature measurement) in msTemperatureMeasurement cluster.
> > Endpoint 3 seems to have floor temperature (external temperature measurement) in msTemperatureMeasurement cluster.
> 
> How did you determine which is which from those?

I pointed warm breath to the thermostat and looked which one was changing. I queried these via z2m dev console.

> I would like this feature to be able to sometimes set the setpoint for floor temperature (like when electricity is extremely cheap and I want to reserve as much heat as possible in the floor before [...]

You probably cannot change the program mode via Zigbee. I use it in floor thermostat mode because I use heat pump for primary heating. 

However, it does not matter if it's in floor or room thermostat mode. Just give it a high setpoint when you want to charge the heat and return the setpoint to normal when you want it to work normally. I do currently just +2...+3 C, but I think nothing prevents you to use max setpoint?

If you have configured the floor max limit correctly, you don't need to worry about overheating. Then the floor temp is just nice info, how well it has charged the heat.

I found a [very nice blueprint](https://community.home-assistant.io/t/blueprint-that-uses-nordpool-and-lets-you-turn-on-devices-on-the-cheapest-hours-and-make-automations-based-on-that-information/646360) which makes it easy to control by price!

> 
> They publish [a compliance document](https://csa-iot.org/csa_product/exxact-connected-thermostat-room-floor-white-7/) that has some information but I was not able to extract anything useful from there.

Yes, it doesn't seem to have all info, but at least what items are supported. Thanks!

Witomas, 2024-08-11:
[Device added](https://github.com/Koenkk/zigbee2mqtt.io/blob/master/docs/devices/WDE002497.md)

closed 2024-08-11 by Witomas

idoo, 2024-11-18:
Seems configuration missing two attributes  `schneider_electric_wde002497_instantaneous_demand` and  `schneider_electric_wde002497_summation_delivered` 
It shows 0 all the time