Koenkk/zigbee2mqtt #20449: SONOFF ZBCurtain: invert_cover option does not work

booroondook-eng, 2023-12-30:
### What happened?

I need to reverse direction of my Sonoff ZBCurtain. When I use offcial eWelink App, then reverse (invert) option works normally, a I need.
But in Zigbee2MQTT this option (it has name "invert_cover") does not work. I tried to set it to "on" and to "off", but motor direction does not change.

### How to reproduce it (minimal and precise)

1. Press Close, see moving direction
2. Press Open, see moving direction
3. Set "invert_cover" from off to on and repeat steps 1 and 2.

### Zigbee2MQTT version

1.34.0 commit: aae7312

### Adapter firmware version

6.7.9.0 build 405

### Adapter

 EZSP v8 (Sonoff ZB Bridge with Tasmota ZBBridge 13.3.0)

closed 2024-06-28 by booroondook-eng

booroondook-eng, 2024-07-25:
> I'm having the same issue.
> 
> @booroondook-eng did you manage to solve it?

Ha-ha. I changed physical placement.

david-storm94, 2024-07-25:
Haha. I'll probably do the same. No reason to over complicate things. I checked the manual as well I also don't see any mention of inverting the direction there. 

I appreciate the quick answer! :)