Koenkk/zigbee2mqtt #20454: ELGO 900091 (recognized as AwoX Modell 33943/33944/33946) does not completely turn off

kianusch, 2023-12-31:
### What happened?

[https://www.zigbee2mqtt.io/devices/900091.html
](https://www.zigbee2mqtt.io/devices/900091.html)

This light has two light "modes" - there is the "main" light with adjustable color temp. & brightness - and also a "RGB LED  ring".

Controlling color temp/brightness works (via color_hs) on the main light work.
Controlling  RGB LEDs ring (via color_xy and rgb). works as well.

But when turning off the light ether via brighness set to 0 or via state-OFF, only the "main" light is turned off - and the LED stays on.

### What did you expect to happen?

Turning off the light via "state":"OFF" both the main light and RGB LED should turn off.

### How to reproduce it (minimal and precise)

Turn on the Light - Turn off the Light.

### Zigbee2MQTT version

1.34.0 commit: aae7312

### Adapter firmware version

20230507

### Adapter

slaesh's CC2652

kianusch, 2023-12-31:
Turning off the light via the provided APP (I belive this communicates over BT) works as expected.

closed 2024-07-29 by github-actions[bot]

GilDev, 2025-01-15:
Just found this issue, I opened one for the same problem a few weeks ago (#25251). Have you ever found a solution for this?

kianusch, 2025-01-28:
With Z2M version 2.0.0 it seems to work as expected:

Sending {"state": "OFF"} turns off the main light as well as the "ambiance" light.
Respectively - sending {"state": "ON"} turns the main light as well as the "ambiance" light on.

Sending {"brightness": 0} turns of the main light, "ambiance" light stays on. As the brightness of the ambiance light can not be controlled - nor can the ambiance light be turn off, as long as the main light stays on.

Also the ambiance light can only be turned off when the main light is also turned off.

GilDev, 2025-01-28:
Thanks for your reply. Toggling the state in the UI (“Exposes” tab) still doesn’t work in 2.0.0 for me, same from HA which is pretty normal.

How are you sending your JSON commands? I went on the MQTT integration, then “Configure” and I tried to publish the payloads you gave to `zigbee2mqtt/<my_friendly_name>` but it doesn’t seem to do anything, can’t find my commands in Z2M’s logs, have I missed something?

kianusch, 2025-01-28:
... be aware that sending "weird" color codes (e.g., {"color":{"rgb":"0,0,0"}}) to the lamp can cause it to crash. In such cases, you will need to turn off the power to the light to reset it.

I use home-assistant - but made also some test by sending mqtt commands directly to the lamp (using MQTT Explorer) - and also Z2M web interface works as well.

Also - I updated the firmware of the lamp using AwoX HomeControl ... My current firmware is: 2.7.5_1446

kianusch, 2025-01-28:
You are missing 'set" in the topic ... zigbee2mqtt/<my_friendly_name>**/set** .

GilDev, 2025-01-28:
Oh indeed I was missing the “set”, thanks!

But still the same issue, I have the same firmware version as you, this is what I get in the logs by sending such “state” payloads:
```
info 2025-01-29 00:51:17z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/PlafonnierSalleEau', payload '{"brightness":117,"color":{"h":0,"hue":0,"s":100,"saturation":100,"x":0.536,"y":0.433},"color_mode":"xy","color_temp":251,"color_temp_startup":65535,"effect":null,"linkquality":156,"power_on_behavior":"previous","state":"ON","state_1":"OFF"}'
info 2025-01-29 00:51:21z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/PlafonnierSalleEau', payload '{"brightness":117,"color":{"h":0,"hue":0,"s":100,"saturation":100,"x":0.536,"y":0.433},"color_mode":"xy","color_temp":251,"color_temp_startup":65535,"effect":null,"linkquality":160,"power_on_behavior":"previous","state":"OFF","state_1":"OFF"}'
```

The backlight will never turn off (other than by sending a direct Zigbee command through the dev console)…

kianusch, 2025-01-29:
This would be my MQTT sending via mosquitto_pub on linux:

`mosquitto_pub -h 192.168.2.12 -u mqtt -P password   -t "zigbee2mqtt/AZ.Decke/set" -m '{"state":"OFF"}'
`

GilDev, 2025-01-29:
Right, well it turns off main light but still not the ambiance light… I don't know what to do, especially with the raw command working, it really seems like there is an issue with the converter…

kianusch, 2025-01-29:
Are you using your own converter?

BTW this is how Zigbee2MQTT sees the device:

Friendly name: AZ.Decke
Description:
Last seen: 51 minutes ago
Availability: [Disabled] (https://www.zigbee2mqtt.io/guide/configuration/device-availability.html#availability-advanced-configuration)
Device type: Router
Zigbee Model: EBF_RGB_Zm
Zigbee Manufacturer: AwoX
Description: ROVITO-Z ceiling light
Support status: Supported

IEEE Address: 0xa4c138aefee9b277
Network address: 0xA69C / 42652
Firmware version: 2.7.5_1446
Manufacturer: [EGLO] (https://www.zigbee2mqtt.io/supported-devices/#v=EGLO)
Model: [900091] (https://www.zigbee2mqtt.io/devices/900091.html#eglo-900091)
Power: 
Interview completed: True

kianusch, 2025-01-29:
Have you tried resetting the light and the connection and then re-repairing the light?

GilDev, 2025-01-29:
No I’m using the internal converter. I get the same info as you:

```
Friendly name: PlafonnierSalleEau
Description:
Last seen: N/A
Availability: Online
Device type: Router
Zigbee Model: EBF_RGB_Zm
Zigbee Manufacturer: AwoX
Description: ROVITO-Z ceiling light
Support status: Supported

IEEE Address: 0xa4c1384f85722159
Network address: 0x9358 / 37720
Firmware version: 2.7.5_1446
Manufacturer: [EGLO](https://www.zigbee2mqtt.io/supported-devices/#v=EGLO)
Model: [900091](https://www.zigbee2mqtt.io/devices/900091.html#eglo-900091)
Power:
Interview completed: True
```

I haven’t tried factory resetting the light, I have to open my switch wall box so I’ll try tomorrow. Thanks for trying to help!

kianusch, 2025-01-29:
And this is my Zigbee2MQTT entry:

  '0xa4c138aefee9b277':
    friendly_name: AZ.Decke
    homeassistant:
      name: AZ.Decke
    optimistic: true

GilDev, 2025-01-29:
Do you please know how to reset the light? It’s not written in the device’s manual nor Z2M’s docs.
I’ll try toggling it 5 times but maybe there’s something else…

kianusch, 2025-02-03:
First I unpair the light from Z2M ...
Then I use the App - first if not paired, I pair it with light ...
Then I unpair the light from the App and start the Pairing in Z2M ...

I see the light blinking a few time ... when pairing.

The app should now tell you, that you have paired the light with a 3rd party gateway.

GilDev, 2025-02-06:
I finally managed to do it! Thanks!
It’s really the unpairing from the app that triggers the Zigbee pairing, the network should be opened just before deleting the device from the app.
It now properly turns off everything, great! It still gets stuck sometimes for a few seconds when doing multiple commands one after the other but it’s mostly good.

kianusch, 2025-02-06:
Great that it works!
As for the "lag" – same here.
All in all, I probably wouldn’t buy this light again. Although it looked promising when I read the description on the box while looking at it in the store.