Koenkk/zigbee2mqtt #20462: Lidl HG06467 christmas lights "firework" effect not working

popy2k14, 2023-12-31:
### What happened?

We have an new year video a few years back (when i was running deconz + conbee II) and saw that the "firework" effect is different in Z2M now. In Z2M it's like the updown effect.

Here is a log on calling the effect:

`Info 2023-12-31 18:35:31MQTT publish: topic 'zigbee2mqtt/Wohnzimmer Weihnachtsbaum', payload '{"brightness":100,"color":{"b":74,"h":240,"s":71},"color_mode":"hs","effect":{"colors":[{"b":249,"g":253,"r":255},{"b":7,"g":0,"r":255},{"b":197,"g":255,"r":127},{"b":255,"g":184,"r":79},{"b":255,"g":81,"r":68},{"b":76,"g":255,"r":210}],"effect":"firework","speed":56},"last_seen":"2023-12-31T18:35:33+01:00","linkquality":80,"state":"ON"}'`  

When calling with effect "updown" it's the same effect happening on the lights.

### What did you expect to happen?

When calling effect firework, that the firework effect is active and not the updown.

### How to reproduce it (minimal and precise)

Here is the Mqtt call from HA (replace all state_attr accordingly):

`service: mqtt.publish
data:
  qos: 0
  topic: zigbee2mqtt/Wohnzimmer Weihnachtsbaum/set
  payload: >-
    {"effect":{"effect":"firework","speed":{{states("input_number.wohnzimmer_weihnachtsbaum_geschwindigkeit")|int(0)}},"colors":[{"r":{{state_attr("light.virtual_wohnzimmer_weihnachtsbaum_farbe_1","rgb_color")[0]}},"g":{{state_attr("light.virtual_wohnzimmer_weihnachtsbaum_farbe_1","rgb_color")[1]}},"b":{{state_attr("light.virtual_wohnzimmer_weihnachtsbaum_farbe_1","rgb_color")[2]}}},{"r":{{state_attr("light.virtual_wohnzimmer_weihnachtsbaum_farbe_2","rgb_color")[0]}},"g":{{state_attr("light.virtual_wohnzimmer_weihnachtsbaum_farbe_2","rgb_color")[1]}},"b":{{state_attr("light.virtual_wohnzimmer_weihnachtsbaum_farbe_2","rgb_color")[2]}}},{"r":{{state_attr("light.virtual_wohnzimmer_weihnachtsbaum_farbe_3","rgb_color")[0]}},"g":{{state_attr("light.virtual_wohnzimmer_weihnachtsbaum_farbe_3","rgb_color")[1]}},"b":{{state_attr("light.virtual_wohnzimmer_weihnachtsbaum_farbe_3","rgb_color")[2]}}},{"r":{{state_attr("light.virtual_wohnzimmer_weihnachtsbaum_farbe_4","rgb_color")[0]}},"g":{{state_attr("light.virtual_wohnzimmer_weihnachtsbaum_farbe_4","rgb_color")[1]}},"b":{{state_attr("light.virtual_wohnzimmer_weihnachtsbaum_farbe_4","rgb_color")[2]}}},{"r":{{state_attr("light.virtual_wohnzimmer_weihnachtsbaum_farbe_5","rgb_color")[0]}},"g":{{state_attr("light.virtual_wohnzimmer_weihnachtsbaum_farbe_5","rgb_color")[1]}},"b":{{state_attr("light.virtual_wohnzimmer_weihnachtsbaum_farbe_5","rgb_color")[2]}}},{"r":{{state_attr("light.virtual_wohnzimmer_weihnachtsbaum_farbe_6","rgb_color")[0]}},"g":{{state_attr("light.virtual_wohnzimmer_weihnachtsbaum_farbe_6","rgb_color")[1]}},"b":{{state_attr("light.virtual_wohnzimmer_weihnachtsbaum_farbe_6","rgb_color")[2]}}}]}}`

### Zigbee2MQTT version

1.34.0

### Adapter firmware version

20230922

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus,Zigbee Gateway TI CC2652P + CP2102N

popy2k14, 2023-12-31:
I think i found the issue in the common.js in herdsman converter:

[image]

See this pull request:

https://github.com/Koenkk/zigbee-herdsman-converters/pull/1799/files

Any chance to get this fixed and make my wife happy? :-)

popy2k14, 2024-01-01:
@Koenkk first, happy new year!

Wanted to make a pull request but not so familiar with the structure of z2m.
Have searched the z2m and herdsman repo for "firework" but did'nt found any match.

Any hint's in which repo the HG06467 code live's :-)

thx

popy2k14, 2024-01-01:
> @Koenkk first, happy new year!
> 
> Wanted to make a pull request but not so familiar with the structure of z2m. Have searched the z2m and herdsman repo for "firework" but did'nt found any match.
> 
> Any hint's in which repo the HG06467 code live's :-)
> 
> thx

Why the heck github is not finding "firework" in the repo!?

[image]

Regardless of this github issue, i have made an pull request.
Sadly i can't test it because i am just running Z2M in an supervised HASS.OS installation:
When i change the code, it reverts on restart.

Any chance to get an test build?

thx

Koenkk, 2024-01-02:
Merged, 
Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

popy2k14, 2024-01-02:
@Koenkk thx for merging.
Tried the "edge" addon in HASS.OS but it's not updated yet, it's at "1.35.0-dev commit: 84387b2" now.

Will try tomorrow again,test the commit and come back here.

thx

popy2k14, 2024-01-03:
@Koenkk i can confirm that the firework effect is working with newest dev (edge) release.

closed 2024-01-03 by popy2k14