Koenkk/zigbee2mqtt #20463: Issue with HomeAssistant Integration "Empty String States" and NodeRED automations

jojolll, 2023-12-31:
### What happened?

When activating the "HomeAssistant" integration in Zigbee2MQTT, the system adds an empty string state with every device state change. This behavior, designed to trigger HomeAssistant events upon each new iteration of a state, now seems to have become the default behavior as of 2023. 

Previously, my HomeAssistant setup never returned empty strings; devices consistently retained their last state as the current state.

Presently, these empty string states replace device states immediately, causing issues, especially with NodeRED actions in HomeAssistant, as it fails to capture all state changes, creating significant problems. I've been trying to find a way to disable this automatic state replacement (and revert to the behavior before 2023 in my installation, it seems to me that this behavior has changed recently), where Zigbee2MQTT preserved device states without replacing them with empty strings. 

### What did you expect to happen?

Keep actel state in "Action" or at least have the option of keeping this state.

Disabling the HomeAssistant integration resolves this issue but leads to other problems and errors in the logs, even though it solves my primary concern.

I seek assistance in understanding why this behavior was enforced (presumably after an update) when the previous functionality was perfectly adequate. Previously, a new state was replacing the old one, allowing my NodeRED automations to trigger correctly even with repeated button presses.

Could you provide steps to revert to the previous behavior without having to redo my automations outside of NodeRED?

### How to reproduce it (minimal and precise)

Use Home Assistant
Use ZigBee2MQTT official HomeAssistant Addon (or a docker container outside the HA system, same behavior)
Use the NodeRED official Addon
Use some Zigbee remote control to activate an action in NodeRED on a state evolution

NodeRED trigger the automation once out of 10 times

### Zigbee2MQTT version

 1.34.0 commit: aae7312

### Adapter firmware version

0x26780700

### Adapter

Conbee II 

### Debug log

N/A

jojolll, 2024-02-01:
When I didn't get an answer, I looked for a solution. Enabling the timestamp and disabling the "HomeAssistant" mode is the best choice for keeping the device status and receiving updates when the button is pressed again, as it only updates the timestamp, keeping the status value intact.

I don't understand why Z2M works any other way, or at least doesn't offer the option of doing so with Home Assistant integration active.

closed 2024-02-01 by jojolll

netopolit, 2024-08-01:
> ### What happened?
> When activating the "HomeAssistant" integration in Zigbee2MQTT, the system adds an empty string state with every device state change. This behavior, designed to trigger [...]

I'm seeing this behavior with regular Home Assistant automations.