Koenkk/zigbee2mqtt #20472: Zigbee.herdsman connection error to SONOFF Dongle Plus

VinvarLaLece, 2024-01-01:
### What happened?

Hi all,

I have troubles getting my SONOFF ZIGBEE 3.0 USB DONGLE PLUS to work. I tried using the containerized version (preferred solution) and by installing it in Linux. For both I followed the [official installation guide](https://www.zigbee2mqtt.io/guide/installation/). I both cases Zigbee.herdsman errors out while starting due to a failure to connect.

### What did you expect to happen?

The installation to succeed. I wonder what I am doing wrong. Would you be able to help, please?

### How to reproduce it (minimal and precise)

The Dongle is connected via a 1,5m extension cord to an INTEL NUC running proxmox 8 (kernel Linux 6.5.11-4-pve). Proxmox is running a LXC in which I tried installing Z2M via Docker and directly on the 'container bare' metal. I tried both installing while plugged into a USB 3 and USB 2 port. Both give me the same errors (see debug log). Eclipse Mosquitto and Docker are the only other applications running in the LXC.

**Preparing the Dongle**
- Flashed it with ncp-uart-sw_6.10.3_115200.gbl & nvm3_initfile.gbl [download here](https://github.com/xsp1989/zigbeeFirmware/tree/master/firmware/Zigbee3.0_Dongle-NoSigned/EZSP) and following this [YouTube video](https://www.youtube.com/watch?v=BU9ha0Uaduc).

**Preparing the host (Proxmox)**
- I created a group to pass permissions: `useradd -u 100000 lxc-root -g lxc-root`
- I created a user to pass permissions: `groupadd -g 100000 lxc-root`
- I added a udev file `SUBSYSTEMS=="usb", ATTRS{idVendor}=="1a86", ATTRS{idProduct}=="55d4", OWNER="lxc-root", GROUP="lxc-root", SYMLINK="ttyACM0"`. I verified the attributes via lsusb.
- I added to the lxc/100.conf: _(USB3=0 as the Dongle is plugged into a USB 2 port currently.)_
```
usb0: host=1a86:55d4,usb3=0
lxc.cgroup2.devices.allow: c 189:* rwm
lxc.mount.entry: /dev/ttyACM0  dev/ttyACM0  none bind,optional,create=file

```
- Rebooted the system.

**In the LXC (Debian 12/bookworm)**
- Verified `/dev/ACCM0` was connected (`ls -l /dev` -> `crw-rw---- root root 189, 4 Jan 1 15:12 ttyACM01`)
- (Tried both with and without) Made sure current user with which I ran the installation was the owner: `sudo chown lxc-admin:lxc-admin /dev/ttyACM0`
- `test -w /dev/ttyACM0 && echo success || echo failure` resulted in 'success'
- Created the configuration: _Below is the Linux install. For Docker attempt I used the same config_
```
mqtt:
  base_topic: zigbee2mqtt
  server: mqtt://retracted_as_is_secret:8883
  user: retracted_as_is_secret
  password: retracted_as_is_secret
  ca: '/etc/ssl/ca.crt'
  key: '/etc/ssl/mqtt-client.key'
  cert: '/etc/ssl/mqtt-client.crt'
serial:
  port: /dev/ttyACM0
  adapter: ezsp
advanced:
  log_level: debug
  pan_id: 13020
  network_key:
    - 15
    - 108
    - 96
    - 197
    - 163
    - 66
    - 253
    - 64
    - 158
    - 233
    - 147
    - 102
    - 85
    - 77
    - 49
    - 154
  channel: 20
```

### Zigbee2MQTT version

1.35.0

### Adapter firmware version

6.10.3

### Adapter

SONOFF DONGLE-E

### Debug log

debug 2024-01-01 14:52:54: Loaded state from file /opt/zigbee2mqtt/data/state.json
info  2024-01-01 14:52:54: Logging to console and directory: '/opt/zigbee2mqtt/data/log/2024-01-01.14-52-54' filename: log.txt
info  2024-01-01 14:52:54: Starting Zigbee2MQTT version 1.35.0 (commit #3d8f0fc)
info  2024-01-01 14:52:54: Starting zigbee-herdsman (0.30.0)
debug 2024-01-01 14:52:54: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null,"delay":null,"disableLED":false},"backupPath":"/opt/zigbee2mqtt/data/coordinator_backup.json","databaseBackupPath":"/opt/zigbee2mqtt/data/dat>error 2024-01-01 14:53:14: Error while starting zigbee-herdsman
error 2024-01-01 14:53:14: Failed to start zigbee
error 2024-01-01 14:53:14: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
error 2024-01-01 14:53:14: Exiting...
error 2024-01-01 14:53:15: Error: Failure to connect
    at Ezsp.connect (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:277:19)
    at Driver.startup (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/driver.ts:139:9)
    at EZSPAdapter.start (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/ezsp/adapter/ezspAdapter.ts:172:16)
    at Controller.start (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/controller.ts:132:29)
    at Zigbee.start (/opt/zigbee2mqtt/lib/zigbee.ts:60:27)
    at Controller.start (/opt/zigbee2mqtt/lib/controller.ts:101:27)
    at start (/opt/zigbee2mqtt/index.js:107:5)

VinvarLaLece, 2024-01-02:
Today I tried it on an VM so I could use the `/dev/serial/by-id`. Same result unfortunately.

Config: _disable_led & rtscts: I tried them without, with one and with both._ 
```
mqtt:
  base_topic: zigbee2mqtt
  server: mqtt://retracted_as_is_secret:8883
  user: retracted_as_is_secret
  password: retracted_as_is_secret
  ca: '/etc/ssl/ca.crt'
  key: '/etc/ssl/mqtt-client.key'
  cert: '/etc/ssl/mqtt-client.crt'
serial:
  port: /dev/serial/by-id/usb-ITEAD_SONOFF_Zigbee_3.0_USB_Dongle_Plus_V2_20230508165132-if00
  adapter: ezsp
  disable_led: true
  rtscts: false
advanced:
  log_level: debug
```

Error log
```
> zigbee2mqtt@1.35.0 start
> node index.js

Zigbee2MQTT:debug 2024-01-02 16:35:12: Can't load state from file /opt/zigbee2mqtt/data/state.json (doesn't exist)
Zigbee2MQTT:info  2024-01-02 16:35:12: Logging to console and directory: '/opt/zigbee2mqtt/data/log/2024-01-02.16-35-12' filename: log.txt
Zigbee2MQTT:info  2024-01-02 16:35:12: Starting Zigbee2MQTT version 1.35.0 (commit #3d8f0fc)
Zigbee2MQTT:info  2024-01-02 16:35:12: Starting zigbee-herdsman (0.30.0)
Zigbee2MQTT:debug 2024-01-02 16:35:12: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null,"delay":null,"disableLED":true},"backupPath":"/opt/zigbee2mqtt/data/coordinator_backup.json","databaseBackupPath":"/opt/zigbee2mqtt/data/database.db.backup","databasePath":"/opt/zigbee2mqtt/data/database.db","network":{"channelList":[11],"extendedPanID":[221,221,221,221,221,221,221,221],"networkKey":"HIDDEN","panID":6754},"serialPort":{"adapter":"ezsp","path":"/dev/serial/by-id/usb-ITEAD_SONOFF_Zigbee_3.0_USB_Dongle_Plus_V2_20230508165132-if00","rtscts":false}}'
Zigbee2MQTT:error 2024-01-02 16:35:32: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2024-01-02 16:35:32: Failed to start zigbee
Zigbee2MQTT:error 2024-01-02 16:35:32: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
Zigbee2MQTT:error 2024-01-02 16:35:32: Exiting...
Zigbee2MQTT:error 2024-01-02 16:35:32: Error: Failure to connect
    at Ezsp.connect (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:277:19)
    at Driver.startup (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/driver.ts:139:9)
    at EZSPAdapter.start (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/ezsp/adapter/ezspAdapter.ts:172:16)
    at Controller.start (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/controller.ts:132:29)
    at Zigbee.start (/opt/zigbee2mqtt/lib/zigbee.ts:60:27)
    at Controller.start (/opt/zigbee2mqtt/lib/controller.ts:101:27)
    at start (/opt/zigbee2mqtt/index.js:107:5)
```

Eggaossi, 2024-01-06:
Zigbee2MQTT:error 2024-01-06 17:19:03: Error: Failure to connect
    at Ezsp.connect (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:277:19)
    at Driver.startup (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/driver.ts:139:9)
    at EZSPAdapter.start (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/adapter/ezspAdapter.ts:172:16)
    at Controller.start (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:132:29)
    at Zigbee.start (/app/lib/zigbee.ts:60:27)
    at Controller.start (/app/lib/controller.ts:101:27)
    at start (/app/index.js:107:5)

VinvarLaLece, 2024-01-07:
@Eggaossi, I struggle to understand your post. Do you meant saying that you encounter the same issue as I have?

> Zigbee2MQTT:error 2024-01-06 17:19:03: Error: Failure to connect at Ezsp.connect (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:277:19) at Driver.startup (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/driver.ts:139:9) at EZSPAdapter.start (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/adapter/ezspAdapter.ts:172:16) at Controller.start (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:132:29) at Zigbee.start (/app/lib/zigbee.ts:60:27) at Controller.start (/app/lib/controller.ts:101:27) at start (/app/index.js:107:5)
> 
> with skyconnect on raspbery pi 4

burntoc, 2024-01-07:
Seeing the same issue here - Skyconnect running the latest firmware.

Eggaossi, 2024-01-07:
The problem is, the skyconnect make only one protokoll: shutt down the
normaly zigbee. The red silicon app

burntoc, 2024-01-07:
Sorry, I don't understand your comment.  I have removed ZHA and both SIlabs addons and this is still the behavior.

Eggaossi, 2024-01-07:
only mqtt may be under settings device integrations, otherwise no other
protocol

Eggaossi, 2024-01-07:
server: mqtt://192.168.178.????:1883
user: mqtt-user
password:mypassword

port: dev/serial/by-id/usb-Nabu_Casa_SkyConnect_v1.0_000000000000000000000ac-if00-port0

here my settings

burntoc, 2024-01-07:
I don't have other device integrations - it isn't that.  Are you running 1.35 Z2M?  And if so, are you running 7.4 Zigbee firmware on your SkyConnect?  Because I am, and those aren't working.

burntoc, 2024-01-09:
Mine is working again using Z2M Edge and the multiprotocol firmware with this as part of the config:

`port: tcp://core-silabs-multiprotocol:9999
adapter: ezsp`

VinvarLaLece, 2024-01-10:
Wonderful to hear yours are working again! I got a step closer to the solution (or so it feels).

I decided to create a bare metal system. So, I removed Proxmox and I installed Debian directly on to the bare metal. With the chain being `SONOFF Zigbee 3.0 USB Dongle Plus - 1,5m extension - Debian 12 @ NUC - Z2M`, I witnessed a successful start up. As is seen in the logs:
```
...
Zigbee2MQTT:info 2024-01-08 10:56:20: Starting zigbee-herdsman (0.30.0)
Zigbee2MQTT:info 2024-01-08 10:56:20: zigbee-herdsman started (reset)
Zigbee2MQTT:info 2024-01-08 10:56:20: Coordinator firmware version: '{"meta":{"maintrel":"3 ","majorrel":"6","minorrel":"10","product":8,"revision":"6.10.3.0 build 297"}, "type":"EZSP v8}'
Zigbee2MQTT:info 2024-01-08 10:56:20: Currently 0 devices are joined:
...
```

So, I deducted the error to be in the combination of Proxmox, unpriv LXC and Z2M. I am confused as I saw the Dongle with the right permissions in the LXC when using Proxmox. The last two days I tried by looking into the [uid/gid mapping](https://pve.proxmox.com/wiki/Unprivileged_LXC_containers) between Proxmox and LXC but without any success so far.

Therefore, all your thoughts, suggestions or solutions are still welcome. :)

MarvinLeeSchroeder, 2024-02-03:
I'm encountering the same "Error while starting zigbee-herdsman" for a few days now with my Sonoff Zigbee Dongle and Z2M LXCs on a Dell Wyse 5070 with Proxmox 8.

I've tried several combinations of [TTeck Z2M LXCs](https://tteck.github.io/Proxmox/) and a bunch of ways to passthrough the dongle:
- As privileged LXC
- By adding it to the LXC's config
- configured lxc.mount.entry
- added is by-id and directly 

I've also tried
- a second dongle and
- plugging in the dongle on different ports and w/o an extension cable
- resetting the dongle
- turn off all existing Zigbee routers

One setup was with complete new hard- and software, so no old stuff that could interfere...

I mostly followed [this excellent guide](https://smarthomescene.com/guides/how-to-separate-zigbee2mqtt-from-home-assistant-in-proxmox).

I can verify that the MQTT LXC is working with Homeassistant.

I'm so frustrated, as I also have no lights in some rooms atm which makes the matter a little more urgent to me 🙈 So any help would be highly appreciated 🙏

MarvinLeeSchroeder, 2024-02-04:
Maybe it's obvious for some, but I think this issue might be related with [this issue](https://github.com/Koenkk/Z-Stack-firmware/discussions/483).

MarvinLeeSchroeder, 2024-02-04:
Ok, I solved my problem but I feel stupid...

TLDR: Use `adapter: zstack` in the `configuration.yaml` when using Sonoff ZDBDongle-P.

I stumbled over the above mentioned issue and was on my way to flash my Dongle as described there. At that point I noticed that the zigbee2mqtt instructions point out, that there are two different dongles, which resulted in me checking which one I have. When I noticed that they have different Serial Interface Protocols (EZSP and ZSTACK) it dawned on me.

Unsure which dongle version I have myself, I just tried ZSTACK and zigbee2mqtt just started. 

Hope it helps someone 🙏

VinvarLaLece, 2024-02-05:
@MarvinLeeSchroeder, wonderful to hear yours is working! Would you happen to know when/why to use 'zstack' as the adapter setting?

As in: I got mine working on bare metal when using 'ezsp' as the adapter setting. It sounds onlogical to me an adapter setting change is needed as it is the same adapter in both scenario's - bare metal vs LXC.

MarvinLeeSchroeder, 2024-02-05:
I would say that depends on the chipset of your ZBDongle:
[image]

If you have the "short" version use `EZSP`, if you have the 'long' version use `ZSTACK`.

VinvarLaLece, 2024-02-05:
Thanks! Reconfirmed that I have an -E. So, EZSP.

> I would say that depends on the chipset of your ZBDongle: 
> ...
> If you have the "short" version use `EZSP`, if you have the 'long' version use `ZSTACK`.

Unfortunately, that means your solution does not solve mine. I keep looking for solution although I am seriously considering getting an old laptop at the dump, refurbish it and add it to my homelab just for running Z2M. (Which would be overkill, but after 2 months of trail-and-error it gets time to move on.)

discoone, 2024-02-06:
> Ok, I solved my problem but I feel stupid...
> 
> TLDR: Use `adapter: zstack` in the `configuration.yaml` when using Sonoff ZDBDongle-P.
> 
> I stumbled over the above mentioned issue and was on my way to flash my Dongle as described there. At that point I noticed that the zigbee2mqtt instructions point out, that there are two different dongles, which resulted in me checking which one I have. When I noticed that they have different Serial Interface Protocols (EZSP and ZSTACK) it dawned on me.
> 
> Unsure which dongle version I have myself, I just tried ZSTACK and zigbee2mqtt just started.
> 
> Hope it helps someone 🙏
> 
> edit:
> 
> here's my full `configuration.yaml`:
> 
> ```
> permit_join: false
> mqtt:
>   base_topic: zigbee2mqtt
>   server: mqtt://192.168.178.8:1883
>   user: marvin
>   password: tPk4ezR8..
> serial:
>   port: /dev/serial/by-id/usb-ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus_2287b6d152beed119e566b2e38a92db5-if00-port0
>   adapter: zstack
> frontend:
>   port: 8080
>   host: 127.0.0.1
> homeassistant: true
> ```

Thank you! I had been struggling this for a while

Nvince76, 2024-08-26:
zstack adapter solved my problem too
many thanks

MightyMeik, 2024-11-03:
> Ok, I solved my problem but I feel stupid...
> 
> TLDR: Use `adapter: zstack` in the `configuration.yaml` when using Sonoff ZDBDongle-P.
> 
> (...)
> 
> Hope it helps someone 🙏
> 
> edit:
> 
> (...)
> ```

Thank you! After hours of searching, your reply saved my day!

closed 2025-02-17 by github-actions[bot]