Koenkk/zigbee2mqtt #20474: MOES MS-104CZ (3 gang) has disappeared from supported devices

jvcsw, 2024-01-01:
### What happened?

The device:

https://github.com/Koenkk/zigbee2mqtt/issues/14632

[image]

now appears as 

https://www.zigbee2mqtt.io/devices/TS0003.html

[image]

### What did you expect to happen?

It appeas as a Moes device as before.

This issue not appears:

https://github.com/Koenkk/zigbee2mqtt/issues/20354
 

### How to reproduce it (minimal and precise)

Update to last versión: 1.35
Show logs.
Show zigbee2mqtt web UI to check the device brand.

### Zigbee2MQTT version

1.35.0

### Adapter firmware version

20211210

### Adapter

zzh

### Debug log

Failed to configure '0xXXXXXXXXXXXXXX', attempt 3 (Error: Read 0xXXXXXXXXXXXXXX/1 genOnOff(["startUpOnOff"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE') at Endpoint.checkStatus (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:321:28) at Endpoint.read (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:463:22) at setupAttributes (/app/node_modules/zigbee-herdsman-converters/src/lib/modernExtend.ts:62:9) at result.configure (/app/node_modules/zigbee-herdsman-converters/src/lib/modernExtend.ts:113:17) at Object.configure (/app/node_modules/zigbee-herdsman-converters/src/index.ts:139:25) at Configure.configure (/app/lib/extension/configure.ts:117:13))

Koenkk, 2024-01-02:
Were you using an external converter for this device? If yes, it likely has become incompatible and needs to be upgraded.

jvcsw, 2024-01-02:
I don't use any external converter.

Everything was working fine (for years) and now the device has changed and the indicated error appears.

Koenkk, 2024-01-02:
Fixed now

closed 2024-01-02 by Koenkk

jvcsw, 2024-01-03:
Thanks @Koenkk