Koenkk/zigbee2mqtt #20475: [Feature request]: Add adapter_delay parameter description to the official documentation

cnaslain, 2024-01-01:
### Is your feature request related to a problem? Please describe

There is no documentation in https://www.zigbee2mqtt.io/guide/configuration/adapter-settings.html regarding the `adapter_delay` parameter.

### Describe the solution you'd like

A detailed description of the purpose of the `adapter_delay` parameter.

### Describe alternatives you've considered

None.

### Additional context

This may resolve `no response received` issues like mentioned in issues [11647](https://github.com/Koenkk/zigbee2mqtt/issues/11647) or [20118](https://github.com/Koenkk/zigbee2mqtt/issues/20118) ?

Koenkk, 2024-01-02:
Done!

closed 2024-01-02 by Koenkk

cnaslain, 2024-01-02:
Thanks a lot :-)