Koenkk/zigbee2mqtt #20476: Errors after upgrade to 1.35.0 for _TZE200_rtrmfadk:  Publish 'set' 'preset' to 'REMOVED' failed: 'Error: No datapoint defined for 'preset''

Da-Hermit, 2024-01-01:
### What happened?

Following update to version 1.35.0 where the TRV _TZE200_rtrmfadk was now being configured as a TRV602 instead of TRV601, I am getting the following errors when trying to change 'preset'.

Error 2024-01-01 15:48:03Publish 'set' 'preset' to 'REMOVED DEVICE NAME' failed: 'Error: No datapoint defined for 'preset''

Devices show up as TS602 and link to correct page/photo. 

### What did you expect to happen?

Expected features available to be corrected

### How to reproduce it (minimal and precise)

Attempt to change Preset value in frontend or from HomeAssistant.

### Zigbee2MQTT version

1.35.0

### Adapter firmware version

7.3.1.0 build 176

### Adapter

EZSP v12

### Debug log

[zblog.txt](https://github.com/Koenkk/zigbee2mqtt/files/13805245/zblog.txt)

ASNNetworks, 2024-01-01:
Same for this radiator valve: TS0601 TZE200_z1tyspqw: https://www.zigbee2mqtt.io/devices/GTZ06.html

I have three of these and none now can set any thing anymore. Presets give the same error as mentioned above. So also can't select manual preset to change temperature.

Tried deleting and pairing device, no luck.

Da-Hermit, 2024-01-01:
I have resorted to using an external converter which fixed the issue but it doesn't include all the features the new release had.

PrzemsiMat, 2024-01-02:
the same issue for [TRV602] and [GTZ06]

Koenkk, 2024-01-02:
Setting the `preset` is not supported indeed, @lompi  could you take a look at this?

lompi, 2024-01-03:
@Koenkk works correctly before extracting part of the code related to `preset` and `system_mode`. So the problem is in the data converter. As I told in my PR, js is not my technology, so I may made mistake on this small refactor.

Koenkk, 2024-01-03:
Could you check if the issue is fixed with the following [external converter](https://gist.github.com/Koenkk/388387067ac18ba01369dfb898b72a2b): 

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m, check if issue is fixed

PrzemsiMat, 2024-01-03:
2024-01-03 10:37:58Publish 'set' 'preset' to 'Głowica Salon' failed: 'Error: Expected one of: 0, 1, 2, 3, got: 'on''
2024-01-03 10:38:06Failed to add 'ZbMoesKnob1' from 'knobmoes'
2024-01-03 10:38:15Publish 'set' 'preset' to 'Głowica Salon' failed: 'Error: Expected one of: 0, 1, 2, 3, got: 'off''
2024-01-03 10:38:22Publish 'set' 'preset' to 'Głowica Salon' failed: 'Error: Expected one of: 0, 1, 2, 3, got: 'on''
2024-01-03 10:38:29Publish 'set' 'preset' to 'Głowica Salon' failed: 'Error: Expected one of: 0, 1, 2, 3, got: 'off''
2024-01-03 10:38:36Publish 'set' 'preset' to 'Głowica Salon' failed: 'Error: Expected one of: 0, 1, 2, 3, got: 'manual''
2024-01-03 10:38:41Publish 'set' 'preset' to 'Głowica Salon' failed: 'Error: Expected one of: 0, 1, 2, 3, got: 'off''

Koenkk, 2024-01-03:
Updated https://gist.github.com/Koenkk/388387067ac18ba01369dfb898b72a2b

PrzemsiMat, 2024-01-03:
Preset change sucsesfull but something i wrong.
Before update when i set off , TRV LCD display OFF (valve set 0% ), when i set on  -  LCD display ON (valve set 100%), and now presets are changed but TRV acts like in Auto mode

lompi, 2024-01-03:
@PrzemsiMat With these changes, the `preset` changes but the `system_mode` does not change. Currently working on a fix.

lompi, 2024-01-03:
I created PR to fix this problem https://github.com/Koenkk/zigbee-herdsman-converters/pull/6819.

Koenkk, 2024-01-03:
Fixed, thanks @lompi 

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) tomorrow

pawol, 2024-01-05:
Probably I reported similar issue there: 
https://github.com/Koenkk/zigbee2mqtt/issues/20598

closed 2024-01-06 by Koenkk