Koenkk/zigbee2mqtt #20477: Zigbee2MQTT will not start after upgrade to 1.35.0-1

keithskillicorn, 2024-01-01:
### What happened?

Upgraded to 1.35.0-1.  No zigbee devices available in homeassistant.  Tired restart - same result.
When tried to visit Zigbee2MQTT page, informed it was not started and asked to restart.  Tried to restart but failed.  Ended with No Gateway page.
Restored backup (1.34) and all was good again.
I have attached snippet of logs from Grafana - can provide more if required.

### How to reproduce it (minimal and precise)

Upgrade.

### Zigbee2MQTT version

1.35.0-1

### Adapter firmware version

20230507

### Adapter

CP2102N_USB_to_UART_Bridge_Controller - Sonoff dongle

### Debug log

`<html>
<body>

2024-01-01 14:59:51.046 | 2024-01-01 14:59:50.966 WARNING (Recorder) [homeassistant.components.recorder.db_schema] State attributes for select.zigbee2mqtt_bridge_log_level exceed maximum size of 16384 bytes. This can cause database performance issues; Attributes will not be stored |  
-- | -- | --
  |   | 2024-01-01 14:59:51.022 | 2024-01-01 14:59:50.965 WARNING (Recorder) [homeassistant.components.recorder.db_schema] State attributes for sensor.zigbee2mqtt_bridge_version exceed maximum size of 16384 bytes. This can cause database performance issues; Attributes will not be stored |  
  |   | 2024-01-01 14:59:50.989 | 2024-01-01 14:59:50.964 WARNING (Recorder) [homeassistant.components.recorder.db_schema] State attributes for sensor.zigbee2mqtt_bridge_permit_join_timeout exceed maximum size of 16384 bytes. This can cause database performance issues; Attributes will not be stored |  
  |   | 2024-01-01 14:59:50.967 | 2024-01-01 14:59:50.963 WARNING (Recorder) [homeassistant.components.recorder.db_schema] State attributes for switch.zigbee2mqtt_bridge_permit_join exceed maximum size of 16384 bytes. This can cause database performance issues; Attributes will not be stored |  
  |   | 2024-01-01 14:59:50.597 | Zigbee2MQTT:info  2024-01-01 14:59:50: zigbee-herdsman started (resumed) |  
  |   | 2024-01-01 14:59:49.991 | Zigbee2MQTT:info  2024-01-01 14:59:49: Starting zigbee-herdsman (0.30.0) |  
  |   | 2024-01-01 14:59:49.957 | Zigbee2MQTT:info  2024-01-01 14:59:49: Logging to console and directory: '/config/zigbee2mqtt/log/2024-01-01.14-59-49' filename: log.txt |  
  |   | 2024-01-01 14:59:45.081 | 24-01-01 14:59:45 INFO (MainThread) [supervisor.docker.addon] Starting Docker add-on zigbee2mqtt/zigbee2mqtt-amd64 with version 1.35.0-1 |  
  |   | 2024-01-01 14:59:44.525 | 24-01-01 14:59:44 INFO (SyncWorker_7) [supervisor.docker.manager] Cleaning addon_45df7312_zigbee2mqtt application |  
  |   | 2024-01-01 14:59:44.041 | at Controller.<anonymous> (/app/lib/zigbee.ts:115:27)) |  
  |   | 2024-01-01 14:59:40.999 | 2024-01-01 14:59:40.941 WARNING (Recorder) [homeassistant.components.recorder.db_schema] State attributes for select.zigbee2mqtt_bridge_log_level exceed maximum size of 16384 bytes. This can cause database performance issues; Attributes will not be stored |  
  |   | 2024-01-01 14:59:40.993 | 2024-01-01 14:59:40.940 WARNING (Recorder) [homeassistant.components.recorder.db_schema] State attributes for sensor.zigbee2mqtt_bridge_version exceed maximum size of 16384 bytes. This can cause database performance issues; Attributes will not be stored |  
  |   | 2024-01-01 14:59:40.986 | 2024-01-01 14:59:40.939 WARNING (Recorder) [homeassistant.components.recorder.db_schema] State attributes for sensor.zigbee2mqtt_bridge_permit_join_timeout exceed maximum size of 16384 bytes. This can cause database performance issues; Attributes will not be stored |  
  |   | 2024-01-01 14:59:40.947 | 2024-01-01 14:59:40.939 WARNING (Recorder) [homeassistant.components.recorder.db_schema] State attributes for switch.zigbee2mqtt_bridge_permit_join exceed maximum size of 16384 bytes. This can cause database performance issues; Attributes will not be stored |  
  |   | 2024-01-01 14:59:40.566 | Zigbee2MQTT:info  2024-01-01 14:59:40: zigbee-herdsman started (resumed) |  
  |   | 2024-01-01 14:59:39.970 | Zigbee2MQTT:info  2024-01-01 14:59:39: Starting zigbee-herdsman (0.30.0) |  
  |   | 2024-01-01 14:59:39.936 | Zigbee2MQTT:info  2024-01-01 14:59:39: Logging to console and directory: '/config/zigbee2mqtt/log/2024-01-01.14-59-39' filename: log.txt |  
  |   | 2024-01-01 14:59:34.991 | 24-01-01 14:59:34 INFO (MainThread) [supervisor.docker.addon] Starting Docker add-on zigbee2mqtt/zigbee2mqtt-amd64 with version 1.35.0-1 |  
  |   | 2024-01-01 14:59:34.404 | 24-01-01 14:59:34 INFO (SyncWorker_7) [supervisor.docker.manager] Cleaning addon_45df7312_zigbee2mqtt application |  
  |   | 2024-01-01 14:59:34.022 | at Controller.<anonymous> (/app/lib/zigbee.ts:115:27)) |  
  |   | 2024-01-01 14:59:31.814 | 2024-01-01 14:59:31.810 WARNING (Recorder) [homeassistant.components.recorder.db_schema] State attributes for select.zigbee2mqtt_bridge_log_level exceed maximum size of 16384 bytes. This can cause database performance issues; Attributes will not be stored |  
  |   | 2024-01-01 14:59:31.814 | 2024-01-01 14:59:31.810 WARNING (Recorder) [homeassistant.components.recorder.db_schema] State attributes for sensor.zigbee2mqtt_bridge_version exceed maximum size of 16384 bytes. This can cause database performance issues; Attributes will not be stored |  
  |   | 2024-01-01 14:59:31.813 | 2024-01-01 14:59:31.809 WARNING (Recorder) [homeassistant.components.recorder.db_schema] State attributes for sensor.zigbee2mqtt_bridge_permit_join_timeout exceed maximum size of 16384 bytes. This can cause database performance issues; Attributes will not be stored |  
  |   | 2024-01-01 14:59:31.813 | 2024-01-01 14:59:31.809 WARNING (Recorder) [homeassistant.components.recorder.db_schema] State attributes for switch.zigbee2mqtt_bridge_permit_join exceed maximum size of 16384 bytes. This can cause database performance issues; Attributes will not be stored |  
  |   | 2024-01-01 14:59:29.877 | Zigbee2MQTT:info  2024-01-01 14:59:29: zigbee-herdsman started (resumed) |  
  |   | 2024-01-01 14:59:29.283 | Zigbee2MQTT:info  2024-01-01 14:59:29: Starting zigbee-herdsman (0.30.0) |  
  |   | 2024-01-01 14:59:29.253 | Zigbee2MQTT:info  2024-01-01 14:59:29: Logging to console and directory: '/config/zigbee2mqtt/log/2024-01-01.14-59-29' filename: log.txt |  
  |   | 2024-01-01 14:59:23.875 | 24-01-01 14:59:23 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/zigbee2mqtt/hassio-zigbee2mqtt repository |  
  |   | 2024-01-01 14:59:23.842 | 24-01-01 14:59:23 INFO (MainThread) [supervisor.docker.addon] Starting Docker add-on zigbee2mqtt/zigbee2mqtt-amd64 with version 1.35.0-1 |  
  |   | 2024-01-01 14:59:22.333 | 24-01-01 14:59:22 INFO (SyncWorker_6) [supervisor.docker.manager] Cleanup images: ['zigbee2mqtt/zigbee2mqtt-amd64:1.34.0-1'] |  
  |   | 2024-01-01 14:59:22.245 | 24-01-01 14:59:22 INFO (MainThread) [supervisor.addons.addon] Add-on '45df7312_zigbee2mqtt' successfully updated |  
  |   | 2024-01-01 14:59:22.206 | 24-01-01 14:59:22 INFO (SyncWorker_0) [supervisor.docker.manager] Cleaning addon_45df7312_zigbee2mqtt application |  
  |   | 2024-01-01 14:59:21.790 | Zigbee2MQTT:info  2024-01-01 14:59:21: Stopped zigbee-herdsman |  
  |   | 2024-01-01 14:59:14.301 | 01-01 14:59:14 INFO Uploading 'addon_45df7312_zigbee2mqtt_1.34.0-1' to Google Drive |  
  |   | 2024-01-01 14:59:12.712 | Zigbee2MQTT:info  2024-01-01 14:59:12: Stopping zigbee-herdsman... |  
  |   | 2024-01-01 14:59:12.676 | 24-01-01 14:59:12 INFO (SyncWorker_0) [supervisor.docker.manager] Stopping addon_45df7312_zigbee2mqtt application |  
  |   | 2024-01-01 14:59:02.740 | 24-01-01 14:59:02 INFO (MainThread) [supervisor.docker.interface] Downloading docker image zigbee2mqtt/zigbee2mqtt-amd64 with tag 1.35.0-1. |  
  |   | 2024-01-01 14:59:02.738 | 24-01-01 14:59:02 INFO (MainThread) [supervisor.docker.addon] Updating image zigbee2mqtt/zigbee2mqtt-amd64:1.34.0-1 to zigbee2mqtt/zigbee2mqtt-amd64:1.35.0-1 |  
  |   | 2024-01-01 14:59:02.715 | 24-01-01 14:59:02 INFO (MainThread) [supervisor.addons.addon] Finish backup for addon 45df7312_zigbee2mqtt |  
  |   | 2024-01-01 14:59:02.709 | 24-01-01 14:59:02 INFO (MainThread) [supervisor.addons.addon] Building backup for add-on 45df7312_zigbee2mqtt

</body>
</html>`

TheHawk1337, 2024-01-02:
I have the same issue. 1.34 works fine, 1.35 doesn't start.

I get these logs:

```
info  2024-01-02 16:50:22: Logging to console and directory: '/app/data/log/2024-01-02.16-50-22' filename: log.txt
info  2024-01-02 16:50:22: Starting Zigbee2MQTT version 1.35.0 (commit #b5a90e9)
info  2024-01-02 16:50:22: Starting zigbee-herdsman (0.30.0)
info  2024-01-02 16:50:52: Accepting joining not in blocklist device '0xa4c138f24689e023'
info  2024-01-02 16:50:52: Accepting joining not in blocklist device '0xa4c138f24689e023'
error 2024-01-02 16:50:53: Failed to start zigbee
error 2024-01-02 16:50:53: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
error 2024-01-02 16:50:53: Exiting...
error 2024-01-02 16:50:53: Error: Read 0x0017880108d038e1/11 lightingColorCtrl(["colorCapabilities"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufact>    at ZStackAdapter.sendZclFrameToEndpointInternal (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:415:23)
    at Queue.executeNext (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:32:32)
```

aizmob, 2024-01-03:
In the same situation with v1.35 I've got success after _full system restart_ (not HA but computer) only. System works fine now with z2m 1.35 and subsequent HA restarts are working OK too.

Koenkk, 2024-01-03:
Please check if this is fixed in the latest [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

keithskillicorn, 2024-01-04:
I stopped Zigbee2MQTT, installed 'Edge' add-on and started that.  The following errors were in the logs and Z2M stopped.

Zigbee2MQTT:info  2024-01-04 20:45:35: MQTT publish: topic 'zigbee2mqtt/Air Quality (Vindstryka)', payload '{"humidity":68,"linkquality":29,"pm25":2,"temperature":16,"update":{"installed_version":16777233,"latest_version":16777233,"state":"idle"},"update_available":null,"voc_index":184}'
Zigbee2MQTT:info  2024-01-04 20:45:35: Zigbee2MQTT started!
Zigbee2MQTT:info  2024-01-04 20:45:36: MQTT publish: topic 'zigbee2mqtt/Develco Electricity Meter', payload '{"battery_low":false,"check_meter":false,"current_summation":0,"energy":3867.744,"interface_mode":"electricity","linkquality":7,"power":782,"pulse_configuration":1000}'
Error: Unhandled error. (TypeError: Cannot destructure property 'vendor' of 'data.device._definition' as it is undefined.
    at /app/data/extension/miboxer-fut089z-controls-exposer.js:151:21
    at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:168:29)
    at EventEmitter.emit (node:events:529:35)
    at EventBus.emitDeviceMessage (/app/lib/eventBus.ts:102:22)
    at Controller.<anonymous> (/app/lib/zigbee.ts:115:27))
    at new NodeError (node:internal/errors:405:5)
    at Controller.emit (node:events:506:17)
    at emitUnhandledRejectionOrErr (node:events:398:10)
    at processTicksAndRejections (node:internal/process/task_queues:84:21)

Koenkk, 2024-01-05:
@keithskillicorn there seems to be an error in an external user extension you are using, try removing that from your configuraiton.yaml

Bartojolao, 2024-01-05:
I had a similar problem with Zigbee2MQTT 1.35.0-1: after the update, it was not possible to start it again. 

As @Koenkk suggested I've removed an external extension (MiBoxer FUT089Z controls-exposer.js). Zigbee2MQTT 1.35.0-1 was able to start and the only problems are that on the Zigbee2MQTT page the 'Last seen' data are missing (I'll see to it later) and the MiBoxer FUT089Z remote is not working properly anymore.

Koenkk, 2024-01-05:
@Bartojolao see https://github.com/Koenkk/zigbee2mqtt-user-extensions/issues/7

keithskillicorn, 2024-01-05:
Yep removed the extension and all is good with edge. Should have shouted that myself. Thanks

Koenkk, 2024-01-06:
great, I've also pushed a fix to make Z2M more robust (will not crash anymore in case of these errors)

closed 2024-01-06 by Koenkk