Koenkk/zigbee2mqtt #20478: Unable to change state of multiple devices after upgrading to v1.35.0

Atriusftw, 2024-01-01:
### What happened?

Multiple devices are unresponsive and throw errors when interacting with them.

### Zigbee2MQTT version

1.35.0

### Adapter firmware version

20210708

### Adapter

SONOFF Dongle-P

### Debug log

`Publish 'set' 'state' to '0x90fd9ffffe86cbb0' failed: 'Error: Command 0x90fd9ffffe86cbb0/1 genOnOff.on({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (SRSP - AF - dataRequest after 6000ms)'`

ASNNetworks, 2024-01-01:
One of my 7 power meters was also stuck on usage and couldn't turn off/on through Z2M. Pressing the power button did work and also updated device state in Z2M. I simply took the plug out of the socket then put it back in. Worked fine again after that, no re-pair needed.

Atriusftw, 2024-01-01:
I just ended up rolling back to 1.34.0. Too many devices failing. Among them are IKEA bulbs, SONOFF SNZB-03, Develco     EMIZB-132.

lameritos, 2024-01-01:
i have all zigbee devices in NA state
and issues 
Error: Command 0x804b50fffe4e5db0/1 manuSpecificTuya.dataRequest({"seq":0,"dpValues":[{"dp":27,"datatype":2,"data":
failed (Data request failed with error: 'MAC channel access failure' (225))'
Sonoff P
z2m 1.35

flowcool, 2024-01-02:
same here since this morning (updated during the night). Nothing was working anymore so I rolled back

johnnykang, 2024-01-15:
> I just ended up rolling back to 1.34.0. Too many devices failing. Among them are IKEA bulbs, SONOFF SNZB-03, Develco EMIZB-132.

hi, how do i do the downgrade if i am using the HA add-on ?

Atriusftw, 2024-01-18:
> > I just ended up rolling back to 1.34.0. Too many devices failing. Among them are IKEA bulbs, SONOFF SNZB-03, Develco EMIZB-132.
> 
> hi, how do i do the downgrade if i am using the HA add-on ?

You will have to restore from a backup if you created one before updating. (Settings -> System -> Backups)

closed 2024-08-17 by github-actions[bot]