Koenkk/zigbee2mqtt #20481: Since update 1.35 I no longer have the last seen information

kwetnico, 2024-01-01:
### What happened?

Since update 1.35 I no longer have the last seen information (I have to move to another frontend menu to see it reappear).
I have a raspberry pi4B with a SONOFF ZIGBEE 3.0 USB DONGLE PLUS-E
[image]

I have already rebooted, it works if I move to other menus (examples: logs, settings...) and I return to the main menu.
I tried on Firefox and Chrome (in private mode)

### What did you expect to happen?

The installation to succeed. 

### How to reproduce it (minimal and precise)

Just install z2m to 1.35 version with a SONOFF ZIGBEE 3.0 USB DONGLE PLUS-E

### Zigbee2MQTT version

1.35

### Adapter firmware version

 6.10.3.0 build 297

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus-E

### Debug log

`2024-01-01 19:09:55MQTT publish: topic 'zigbee2mqtt/Lexman_Prise', payload '{"current":0,"energy":73.75,"last_seen":"2024-01-01T19:10:38+01:00","linkquality":180,"power":5,"power_on_behavior":"off","state":"ON","voltage":0}'
Info 2024-01-01 19:09:59MQTT publish: topic 'zigbee2mqtt/Temperature_Lyne', payload '{"battery":97,"elapsed":352420,"humidity":61.85,"last_seen":"2024-01-01T19:10:43+01:00","linkquality":204,"power_outage_count":43270,"pressure":995.6,"temperature":19.34,"voltage":2995}'
Info 2024-01-01 19:09:59MQTT publish: topic 'zigbee2mqtt/Temperature_Lyne', payload '{"battery":97,"elapsed":10,"humidity":62.02,"last_seen":"2024-01-01T19:10:43+01:00","linkquality":204,"power_outage_count":43270,"pressure":995.6,"temperature":19.34,"voltage":2995}'
Info 2024-01-01 19:09:59MQTT publish: topic 'zigbee2mqtt/Temperature_Lyne', payload '{"battery":97,"elapsed":10,"humidity":62.02,"last_seen":"2024-01-01T19:10:43+01:00","linkquality":204,"power_outage_count":43270,"pressure":995.4,"temperature":19.34,"voltage":2995}'
Info 2024-01-01 19:10:06MQTT publish: topic 'zigbee2mqtt/Sirene2', payload '{"alarm":false,"battery":100,"last_seen":"2024-01-01T19:10:50+01:00","linkquality":164,"voltage":5000,"volume":80}'
Info 2024-01-01 19:10:14MQTT publish: topic 'zigbee2mqtt/Temperature_ChambreAmi', payload '{"battery":100,"elapsed":1255673,"humidity":64.55,"last_seen":"2024-01-01T19:10:58+01:00","linkquality":144,"power_outage_count":51,"pressure":995.6,"temperature":19.34,"voltage":3025}'
Info 2024-01-01 19:10:14MQTT publish: topic 'zigbee2mqtt/Temperature_ChambreAmi', payload '{"battery":100,"elapsed":9,"humidity":65.07,"last_seen":"2024-01-01T19:10:58+01:00","linkquality":144,"power_outage_count":51,"pressure":995.6,"temperature":19.34,"voltage":3025}'
Info 2024-01-01 19:10:14MQTT publish: topic 'zigbee2mqtt/Temperature_ChambreAmi', payload '{"battery":100,"elapsed":12,"humidity":65.07,"last_seen":"2024-01-01T19:10:58+01:00","linkquality":140,"power_outage_count":51,"pressure":995.3,"temperature":19.34,"voltage":3025}'
Info 2024-01-01 19:10:17MQTT publish: topic 'zigbee2mqtt/Temperature_Cellier', payload '{"battery":100,"elapsed":1228129,"humidity":56.49,"last_seen":"2024-01-01T19:11:00+01:00","linkquality":255,"power_outage_count":13,"pressure":996,"temperature":20.33,"voltage":3015}'
Info 2024-01-01 19:10:17MQTT publish: topic 'zigbee2mqtt/Temperature_Cellier', payload '{"battery":100,"elapsed":8,"humidity":56.54,"last_seen":"2024-01-01T19:11:00+01:00","linkquality":255,"power_outage_count":13,"pressure":996,"temperature":20.33,"voltage":3015}'
Info 2024-01-01 19:10:17MQTT publish: topic 'zigbee2mqtt/Temperature_Cellier', payload '{"battery":100,"elapsed":11,"humidity":56.54,"last_seen":"2024-01-01T19:11:00+01:00","linkquality":255,"power_outage_count":13,"pressure":995.8,"temperature":20.33,"voltage":3015}'
Info 2024-01-01 19:10:30MQTT publish: topic 'zigbee2mqtt/Xiaomi_Prise', payload '{"consumption":802.1908569335938,"device_temperature":34,"energy":802.19,"last_seen":"2024-01-01T19:11:13+01:00","linkquality":144,"power":0,"power_outage_count":221,"state":"OFF","update":{"installed_version":92,"latest_version":92,"state":"idle"},"update_available":false}'
Info 2024-01-01 19:10:39MQTT publish: topic 'zigbee2mqtt/Temperature_Exterieur', payload '{"battery":57,"elapsed":988879,"humidity":91.22,"last_seen":"2024-01-01T19:11:23+01:00","linkquality":180,"power_outage_count":65300,"pressure":997.6,"temperature":7.67,"voltage":2935}'
Info 2024-01-01 19:10:39MQTT publish: topic 'zigbee2mqtt/Temperature_Exterieur', payload '{"battery":57,"elapsed":9,"humidity":91.25,"last_seen":"2024-01-01T19:11:23+01:00","linkquality":180,"power_outage_count":65300,"pressure":997.6,"temperature":7.67,"voltage":2935}'
Info 2024-01-01 19:10:39MQTT publish: topic 'zigbee2mqtt/Temperature_Exterieur', payload '{"battery":57,"elapsed":11,"humidity":91.25,"last_seen":"2024-01-01T19:11:23+01:00","linkquality":180,"power_outage_count":65300,"pressure":997.3,"temperature":7.67,"voltage":2935}'
Info 2024-01-01 19:11:20MQTT publish: topic 'zigbee2mqtt/Fenetre_SDB', payload '{"battery":83,"contact":true,"device_temperature":23,"elapsed":3017424,"last_seen":"2024-01-01T19:12:04+01:00","linkquality":144,"power_outage_count":66,"voltage":2975}'
Info 2024-01-01 19:11:27MQTT publish: topic 'zigbee2mqtt/Xiaomi_Prise', payload '{"consumption":802.1908569335938,"device_temperature":34,"energy":802.19,"last_seen":"2024-01-01T19:12:10+01:00","linkquality":140,"power":0,"power_outage_count":221,"state":"OFF","update":{"installed_version":92,"latest_version":92,"state":"idle"},"update_available":false}'
Info 2024-01-01 19:11:44MQTT publish: topic 'zigbee2mqtt/Temperature_Garage', payload '{"battery":83,"elapsed":1141994,"humidity":80.01,"last_seen":"2024-01-01T19:12:28+01:00","linkquality":120,"power_outage_count":6,"pressure":995.8,"temperature":11.57,"voltage":2975}'
Info 2024-01-01 19:11:44MQTT publish: topic 'zigbee2mqtt/Temperature_Garage', payload '{"battery":83,"elapsed":9,"humidity":80.35,"last_seen":"2024-01-01T19:12:28+01:00","linkquality":120,"power_outage_count":6,"pressure":995.8,"temperature":11.57,"voltage":2975}'
Info 2024-01-01 19:11:44MQTT publish: topic 'zigbee2mqtt/Temperature_Garage', payload '{"battery":83,"elapsed":11,"humidity":80.35,"last_seen":"2024-01-01T19:12:28+01:00","linkquality":124,"power_outage_count":6,"pressure":995.6,"temperature":11.57,"voltage":2975}'
Info 2024-01-01 19:12:19MQTT publish: topic 'zigbee2mqtt/Presence_Etage', payload '{"battery":100,"device_temperature":23,"elapsed":2578961,"illuminance":0,"illuminance_lux":0,"last_seen":"2024-01-01T19:13:02+01:00","linkquality":88,"occupancy":false,"power_outage_count":117,"voltage":3015}'
Info 2024-01-01 19:12:21MQTT publish: topic 'zigbee2mqtt/Xiaomi_Prise', payload '{"consumption":802.1908569335938,"device_temperature":34,"energy":802.19,"last_seen":"2024-01-01T19:13:04+01:00","linkquality":88,"power":0,"power_outage_count":221,"state":"OFF","update":{"installed_version":92,"latest_version":92,"state":"idle"},"update_available":false}'
Info 2024-01-01 19:12:51MQTT publish: topic 'zigbee2mqtt/Sirene2', payload '{"alarm":false,"battery":100,"last_seen":"2024-01-01T19:13:35+01:00","linkquality":164,"voltage":5000,"volume":80}'
Info 2024-01-01 19:13:13MQTT publish: topic 'zigbee2mqtt/Xiaomi_Prise', payload '{"consumption":802.1908569335938,"device_temperature":34,"energy":802.19,"last_seen":"2024-01-01T19:13:56+01:00","linkquality":140,"power":0,"power_outage_count":221,"state":"OFF","update":{"installed_version":92,"latest_version":92,"state":"idle"},"update_available":false}'
Info 2024-01-01 19:13:58MQTT publish: topic 'zigbee2mqtt/Temperature_Cellier', payload '{"battery":100,"elapsed":221672,"humidity":56.47,"last_seen":"2024-01-01T19:14:42+01:00","linkquality":255,"power_outage_count":13,"pressure":995.8,"temperature":20.34,"voltage":3015}'
Info 2024-01-01 19:14:05MQTT publish: topic 'zigbee2mqtt/Xiaomi_Prise', payload '{"consumption":802.1908569335938,"device_temperature":34,"energy":802.19,"last_seen":"2024-01-01T19:14:48+01:00","linkquality":140,"power":0,"power_outage_count":221,"state":"OFF","update":{"installed_version":92,"latest_version":92,"state":"idle"},"update_available":false}'
Info 2024-01-01 19:14:17MQTT publish: topic 'zigbee2mqtt/Porte_Moteur_Garage', payload '{"elapsed":299995,"last_seen":"2024-01-01T19:15:01+01:00","linkquality":120,"state":"OFF"}'
Info 2024-01-01 19:14:41Successfully changed options
Info 2024-01-01 19:14:41MQTT publish: topic 'zigbee2mqtt/bridge/response/options', payload '{"data":{"restart_required":false},"status":"ok","transaction":"tbk14-1"}'`

ASNNetworks, 2024-01-01:
I think it's a UI bug. When I open a device and back out to device panel, all last seen times are shown again. Also attributes of all devices are shown in HA entity correctly. So it seems it shows N/A only when first opening Z2M dashboard, but it still works.

kwetnico, 2024-01-01:
> I think it's a UI bug. When I open a device and back out to device panel, all last seen times are shown again. Also attributes of all devices are shown in HA entity correctly. So it seems it shows N/A only when first opening Z2M dashboard, but it still works.

Hello yes I have exactly the same symptoms as you.  This since update 1.35

ASNNetworks, 2024-01-01:
> Hello yes I have exactly the same symptoms as you. This since update 1.35

Yes, so the good thing is that it's just a UI bug and doesn't break the actual function. Good on creating the ticket, but luckily it doesn't cause any issues.

terierul, 2024-01-02:
> I think it's a UI bug. When I open a device and back out to device panel, all last seen times are shown again. Also attributes of all devices are shown in HA entity correctly. So it seems it shows N/A only when first opening Z2M dashboard, but it still works.

for me is the same

Koenkk, 2024-01-02:
@nurikk could you check this?

sammyke007, 2024-01-02:
Exact same issue over here since 1.35.0.
If I click Dashboard and then on Devices again, availability and last seen is back however.
Cleared complete cache and cookies, but doesn't fix it.

TimBroddin, 2024-01-02:
I think this might've been caused by my PR here: https://github.com/nurikk/zigbee2mqtt-frontend/pull/1841

I'll create a revert PR.

closed 2024-01-02 by nurikk

aleksander000, 2024-01-05:
I found other issue. My CC2531 (PTVO) works with BMP pressure doesn't refresh data from it. Very strange because other analog sensors work correctly (digital inputs too).