Koenkk/zigbee2mqtt #20484: Homebridge/Zigbee2MQTT fails to start after update to 1.35.0

SaschaRH, 2024-01-01:
### What happened?

After updating to 1.35.0 (from 1.33.0) Homebridge/Zigbee2MQTT throws the following exception and restarts until the plugin is disabled or Z2M reverted to 1.33.0:

[1.1.2024, 19:20:36] TypeError: Cannot read properties of null (reading 'vendor')
    at isDeviceDefinition (/var/lib/homebridge/node_modules/homebridge-z2m/src/z2mModels.ts:294:73)
    at Zigbee2mqttAccessory.updateDeviceInformation (/var/lib/homebridge/node_modules/homebridge-z2m/src/platformAccessory.ts:290:30)
    at new Zigbee2mqttAccessory (/var/lib/homebridge/node_modules/homebridge-z2m/src/platformAccessory.ts:82:10)
    at Zigbee2mqttPlatform.addAccessory (/var/lib/homebridge/node_modules/homebridge-z2m/src/platform.ts:419:19)
    at Zigbee2mqttPlatform.configureAccessory (/var/lib/homebridge/node_modules/homebridge-z2m/src/platform.ts:319:10)
    at /var/lib/homebridge/node_modules/homebridge/src/bridgeService.ts:328:25
    at Array.filter (<anonymous>)
    at BridgeService.restoreCachedPlatformAccessories (/var/lib/homebridge/node_modules/homebridge/src/bridgeService.ts:288:69)
    at Server.start (/var/lib/homebridge/node_modules/homebridge/src/server.ts:188:24)

### Zigbee2MQTT version

1.35.0

### Adapter firmware version

0x26780700

### Adapter

Conbee2

basalphenaar, 2024-01-01:
I'm having the same problem, I reverted back to v1.34.0.

DevRedOWL, 2024-01-01:
Same, but even revert not helps

noob4722019, 2024-01-01:
Same here, 

[09:07:33] INFO: Preparing to start...
[09:07:33] INFO: Socat not enabled
[09:07:34] INFO: Starting Zigbee2MQTT...

Koenkk, 2024-01-02:
Please create an issue in the homebridge-z2m issue tracker.

jfcabral, 2024-01-02:
The subject is being tracked here: https://github.com/itavero/homebridge-z2m/issues/794

There's already a new version that fixes this issue: https://github.com/itavero/homebridge-z2m/releases/tag/v1.9.3-rc.0