Koenkk/zigbee2mqtt #20485: Problem scene group & individual.

msaphoto06, 2024-01-01:
### What happened?

Hello, sorry for my bad english

- Individual scenes don't show up in Home Assistant.

- My group scenes no longer control the element. -> the potentiometers move but the bulb does not change state (The change of state in individual scenes is OK)

Thank you 

### Zigbee2MQTT version

1.35

### Adapter firmware version

Last

### Adapter

Sonoff P

mundschenk-at, 2024-01-21:
Hi @msaphoto06, I can reproduced your first issue regarding scenes for individual entities/endpoints not getting properly discovered by Home Assistant. I'll look into this.

mundschenk-at, 2024-01-22:
So I discovered that the issue was an incorrect optimization for the discovery updates, preventing the first scene added to either an endpoint or a group from being published. (If an entity or group already had at least one scene present on startup of Z2M, any newly added scenes would trigger the necessary updates.)

This should be fixed in https://github.com/Koenkk/zigbee2mqtt/pull/20952

mundschenk-at, 2024-07-21:
@Koenkk This issue can be closed.

closed 2024-07-21 by Koenkk