Koenkk/zigbee2mqtt #20486: Exception while calling fromZigbee converter: Value '2' is not allowed, expected one of [object Object],[object Object],[object Object]

olili, 2024-01-01:
### What happened?

acc. release notes v1.35 the issue [https://github.com/Koenkk/zigbee2mqtt/issues/16648](Issue 16648) should have been resolved. 

Nevertheless I face still the problem of errors in holiday mode after update. 

Can somebody share this experience?

### What did you expect to happen?

fixed issue

### How to reproduce it (minimal and precise)

log

### Zigbee2MQTT version

latest

### Adapter firmware version

latest

### Adapter

Sonoff cc2562

olili, 2024-01-03:
created an external converter for my TV02s (_TZE200_mudxchsu) providing known bugfix (index 2 instead index 3):

`        meta: {
            tuyaDatapoints: [
                [2, 'preset', tuya.valueConverterBasic.lookup({ 'auto': tuya.enum(0), 'manual': tuya.enum(1), 'holiday': tuya.enum(2)  })],`

[trv_tv02.zip](https://github.com/Koenkk/zigbee2mqtt/files/13818964/trv_tv02.zip)

Maybe someone can incorporate the improvement in the herdsmen-converter-lib, (devices/tuya.js)

Koenkk, 2024-01-04:
Fixed!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

closed 2024-01-04 by Koenkk