Koenkk/zigbee2mqtt #20488: 1.35.0 - Z2M bridge entities causing Home Assistant recorder issues

RobertD502, 2024-01-01:
### What happened?

After updating to 1.35.0, entities related to the Z2M bridge device are raising errors in Home Assistant.

```
Logger: homeassistant.components.recorder.db_schema
Source: components/recorder/db_schema.py:582
Integration: Recorder (documentation, issues)
First occurred: 3:24:36 PM (13 occurrences)
Last logged: 3:27:13 PM

State attributes for select.zigbee2mqtt_bridge_log_level exceed maximum size of 16384 bytes. This can cause database performance issues; Attributes will not be stored
State attributes for sensor.zigbee2mqtt_bridge_permit_join_timeout exceed maximum size of 16384 bytes. This can cause database performance issues; Attributes will not be stored
State attributes for sensor.zigbee2mqtt_bridge_version exceed maximum size of 16384 bytes. This can cause database performance issues; Attributes will not be stored
State attributes for switch.zigbee2mqtt_bridge_permit_join exceed maximum size of 16384 bytes. This can cause database performance issues; Attributes will not be stored
State attributes for binary_sensor.zigbee2mqtt_bridge_restart_required exceed maximum size of 16384 bytes. This can cause database performance issues; Attributes will not be stored
```

### Zigbee2MQTT version

1.35.0

### Adapter firmware version

20200925

### Adapter

Electrolama

mundschenk-at, 2024-01-01:
The errors are the result of having legacy entity attributes enabled.

RobertD502, 2024-01-01:
@mundschenk-at Thanks! That was the root of it.

desosav, 2024-01-02:
Disabling legacy entity attributes is a solution or just a workaround? This issue is marked as completed however since the option to enable legacy entity attributes is still there it shouldn't generate errors in the logs. I am wondering if the support for legacy attr has been dropped...

alexsaas, 2024-01-02:
I installed z2m fresh today and ‘legacy entity attributes’ is deactivated by default. Unfortunately, I still have the error.

RobertD502, 2024-01-02:
> Disabling legacy entity attributes is a solution or just a workaround? This issue is marked as completed however since the option to enable legacy entity attributes is still there it shouldn't generate errors in the logs. I am wondering if the support for legacy attr has been dropped...

It is more of a workaround. Legacy attributes are completely valid for Zigbee device sensors since the attributes aren't extensive. The option becomes an issue with the Bridge device - some of the sensors have extensive attributes such as all of your ZM2 settings, properties, etc - this is where the issue comes up.

I'll reopen the issue in case @Koenkk wants to chime in and give his take on whether this is an issue that needs fixing by avoiding sending lengthy attributes for the bridge sensors in question or if it is time to deprecate the legacy options.

mundschenk-at, 2024-01-02:
@desosav [It was decided](https://github.com/Koenkk/zigbee2mqtt/pull/19994#issuecomment-1863346427) during development that having that flag enabled was not a supported configuration going forward.

alexsaas, 2024-01-02:
> @alexsaas Try reloading the HA MQTT integration (or just restarting HA).

Thank you. But after HA full restart the error exists.

mundschenk-at, 2024-01-02:
Try manually clearing out retained messages in your MQTT broker.

alexsaas, 2024-01-02:
Sorry but i'm newbie. How could i do that?

mundschenk-at, 2024-01-02:
Have a look at https://stackoverflow.com/questions/36729300/how-to-clear-all-retained-mqtt-messages-from-mosquitto for pointers. (Depending on your broker, there might also be a web frontend allowing you to remove retained topics.)

RobertD502, 2024-01-02:
Download [MQTT explorer](http://mqtt-explorer.com). You can clear topics on your broker using it. Best to shut down Z2M before doing so, so that you don't have topics filling up while trying to clear them.

alexsaas, 2024-01-02:
Okay, i'll try it later. My Computer runs an update at the moment. Thanks in advance.

alexsaas, 2024-01-02:
Sorry, it was my mistake! I made an error while configuring the MQTT broker by followed a tutorial. It said that a username and password should be entered under the Add-on configuration at the logins point. I loaded the factory settings and left out the entries. Since then, the error has disappeared after a restart. Please forgive me for taking up your time.

RobertD502, 2024-01-02:
No skin off of my back. Glad you got it figured out.

alexsaas, 2024-01-02:
Me too. Thank you again and sorry again at all. I found out that when creating user data for MQTT, it also needs to be stored in the z2m configuration. Logically. I'm still lesrning. Interesting that the error messages were still the same. But now it’s running fortunately.

david883, 2024-01-03:
Same Error after Upgrade to 1.3.5

2024-01-02 20:54:52.558 WARNING (Recorder) [homeassistant.components.recorder.db_schema] State attributes for select.zigbee2mqtt_bridge_log_level exceed maximum size of 16384 bytes. This can cause database performance issues; Attributes will not be stored
2024-01-02 20:54:52.559 WARNING (Recorder) [homeassistant.components.recorder.db_schema] State attributes for sensor.zigbee2mqtt_bridge_permit_join_timeout exceed maximum size of 16384 bytes. This can cause database performance issues; Attributes will not be stored
2024-01-02 20:54:52.559 WARNING (Recorder) [homeassistant.components.recorder.db_schema] State attributes for sensor.zigbee2mqtt_bridge_version exceed maximum size of 16384 bytes. This can cause database performance issues; Attributes will not be stored
2024-01-02 20:54:52.559 WARNING (Recorder) [homeassistant.components.recorder.db_schema] State attributes for switch.zigbee2mqtt_bridge_permit_join exceed maximum size of 16384 bytes. This can cause database performance issues; Attributes will not be stored
2024-01-02 20:54:52.947 WARNING (Recorder) [homeassistant.components.recorder.db_schema] State attributes for select.zigbee2mqtt_bridge_log_level exceed maximum size of 16384 bytes. This can cause database performance issues; Attributes will not be stored
2024-01-02 20:54:52.948 WARNING (Recorder) [homeassistant.components.recorder.db_schema] State attributes for sensor.zigbee2mqtt_bridge_permit_join_timeout exceed maximum size of 16384 bytes. This can cause database performance issues; Attributes will not be stored
2024-01-02 20:54:52.948 WARNING (Recorder) [homeassistant.components.recorder.db_schema] State attributes for sensor.zigbee2mqtt_bridge_version exceed maximum size of 16384 bytes. This can cause database performance issues; Attributes will not be stored
2024-01-02 20:54:52.948 WARNING (Recorder) [homeassistant.components.recorder.db_schema] State attributes for switch.zigbee2mqtt_bridge_permit_join exceed maximum size of 16384 bytes. This can cause database performance issues; Attributes will not be stored

Boztekau, 2024-01-03:
I uninstalled both Zigbee2MQTT and Mosquitto, rebooted HA doing fresh installs and still get the same error;
State attributes for select.zigbee2mqtt_bridge_log_level exceed maximum size of 16384 bytes. This can cause database performance issues; Attributes will not be stored
State attributes for sensor.zigbee2mqtt_bridge_permit_join_timeout exceed maximum size of 16384 bytes. This can cause database performance issues; Attributes will not be stored
State attributes for sensor.zigbee2mqtt_bridge_version exceed maximum size of 16384 bytes. This can cause database performance issues; Attributes will not be stored
State attributes for switch.zigbee2mqtt_bridge_permit_join exceed maximum size of 16384 bytes. This can cause database performance issues; Attributes will not be stored
State attributes for binary_sensor.zigbee2mqtt_bridge_restart_required exceed maximum size of 16384 bytes. This can cause database performance issues; Attributes will not be stored

mundschenk-at, 2024-01-03:
Which is expected if you don't change your configuration to disable `legacy_entity_attributes`.

Boztekau, 2024-01-03:
Same Error after Upgrade to 1.3.5

2024-01-02 20:54:52.558 WARNING (Recorder) [homeassistant.components.recorder.db_schema] State attributes for select.zigbee2mqtt_bridge_log_level exceed maximum size of 16384 bytes. This can cause database performance issues; Attributes will not be stored
2024-01-02 20:54:52.559 WARNING (Recorder) [homeassistant.components.recorder.db_schema] State attributes for sensor.zigbee2mqtt_bridge_permit_join_timeout exceed maximum size of 16384 bytes. This can cause database performance issues; Attributes will not be stored
2024-01-02 20:54:52.559 WARNING (Recorder) [homeassistant.components.recorder.db_schema] State attributes for sensor.zigbee2mqtt_bridge_version exceed maximum size of 16384 bytes. This can cause database performance issues; Attributes will not be stored
2024-01-02 20:54:52.559 WARNING (Recorder) [homeassistant.components.recorder.db_schema] State attributes for switch.zigbee2mqtt_bridge_permit_join exceed maximum size of 16384 bytes. This can cause database performance issues; Attributes will not be stored
2024-01-02 20:54:52.947 WARNING (Recorder) [homeassistant.components.recorder.db_schema] State attributes for select.zigbee2mqtt_bridge_log_level exceed maximum size of 16384 bytes. This can cause database performance issues; Attributes will not be stored
2024-01-02 20:54:52.948 WARNING (Recorder) [homeassistant.components.recorder.db_schema] State attributes for sensor.zigbee2mqtt_bridge_permit_join_timeout exceed maximum size of 16384 bytes. This can cause database performance issues; Attributes will not be stored
2024-01-02 20:54:52.948 WARNING (Recorder) [homeassistant.components.recorder.db_schema] State attributes for sensor.zigbee2mqtt_bridge_version exceed maximum size of 16384 bytes. This can cause database performance issues; Attributes will not be stored
2024-01-02 20:54:52.948 WARNING (Recorder) [homeassistant.components.recorder.db_schema] State attributes for switch.zigbee2mqtt_bridge_permit_join exceed maximum size of 16384 bytes. This can cause database performance issues; Attributes will not be stored

Boztekau, 2024-01-03:
disabled at config and same error on fresh install. Went back to 1.3.4 and no errors.

tluethi71, 2024-01-03:
same here. I have updated to 1.35. Got the same error... legacy_entity_attributes is false. 

Further I have the following error popping up 
`Template variable error: 'value_json' is undefined when rendering '{{ value_json.state }}'`
`Error parsing value: 'value_json' is undefined (value: offline, template: {{ value_json.state }})`

mundschenk-at, 2024-01-03:
Well, turns out that coexistence with legacy entity attributes is a one-liner after all, see #20548.

Boztekau, 2024-01-04:
Done a fresh install of HA then latest version of Zigbee@MQTT and all is good.  I must of had a bug somewhere in my old Remote HA as the primary HA server had no issue updating to the current version of Zigbee@MQTT.  Thanks for the support!

duczz, 2024-01-04:
i have the same error and a few more:
`homeassistant: true
legacy_entity_attributes: false
mqtt:
  server: mqtt://core-mosquitto:1883
  user: homeassistant
  password: xi1xiegxxeeixxxxxx
  keepalive: 60
  reject_unauthorized: true
  version: 4
  include_device_information: false
serial:
  port: /dev/ttyUSB0
frontend:
  port: 8099
devices:
  "0x00124b001cd59551":
    friendly_name: Wohnzimmer Temperatursensor SNZB-02
  "0x00158d0001dd412e":
    friendly_name: Eingangstür - Türsensor (Aqara)
advanced:
  log_level: info
  log_syslog:
    app_name: Zigbee2MQTT
    eol: /n
    host: localhost
    localhost: localhost
    path: /dev/log
    pid: process.pid
    port: 514
    protocol: udp4
    type: "5424"
  last_seen: ISO_8601_local
`
it starts with:
`Logger: homeassistant.components.mqtt.mixins
Source: components/mqtt/mixins.py:598
Integration: MQTT (documentation, issues)
First occurred: 13:48:44 (1 occurrences)
Last logged: 13:48:44

Erroneous JSON: online`

than
`Logger: homeassistant.helpers.template
Source: helpers/template.py:2318
First occurred: 13:48:44 (1 occurrences)
Last logged: 13:48:44

Template variable error: 'value_json' is undefined when rendering '{{ value_json.state }}'`

than
`Logger: homeassistant.helpers.template
Source: helpers/template.py:749
First occurred: 13:48:44 (1 occurrences)
Last logged: 13:48:44

Error parsing value: 'value_json' is undefined (value: online, template: {{ value_json.state }})`

than
`Logger: homeassistant.components.recorder.db_schema
Source: components/recorder/db_schema.py:582
Integration: Recorder (documentation, issues)
First occurred: 13:48:51 (8 occurrences)
Last logged: 13:48:52

State attributes for switch.zigbee2mqtt_bridge_permit_join exceed maximum size of 16384 bytes. This can cause database performance issues; Attributes will not be stored
State attributes for sensor.zigbee2mqtt_bridge_permit_join_timeout exceed maximum size of 16384 bytes. This can cause database performance issues; Attributes will not be stored
State attributes for sensor.zigbee2mqtt_bridge_version exceed maximum size of 16384 bytes. This can cause database performance issues; Attributes will not be stored
State attributes for select.zigbee2mqtt_bridge_log_level exceed maximum size of 16384 bytes. This can cause database performance issues; Attributes will not be stored`

mundschenk-at, 2024-01-05:
@Boztekau OK, so it turns out you need to delete* the device in HA to get rid of the `recorder` warnings, even if there are no attributes mentioned in the discovery topic (or you can wait until the recorded states are purged, by default 7 days IIRC).

* It will be automatically re-created, but you will have to re-do any entity renaming.

RobertD502, 2024-01-05:
> @Boztekau OK, so it turns out you need to delete* the device in HA to get rid of the `recorder` warnings, even if there are no attributes mentioned in the discovery topic (or you can wait until the [...]

I did not need to delete devices. Simply disabling the setting, restarting Z2M and then restarting HA stopped the recorder warnings. I'm assuming some aren't restarting Z2M for the new setting to take effect and/or have retained messages.

mundschenk-at, 2024-01-05:
I don't know what exactly caused that state, but I have seen the issue with my test installation (after switching between various branches of Z2M-dev). Restarting Z2M did not fix the symptoms in that case, but clearing out the device definition in HA did.

Claudio1L, 2024-01-06:
HI,

not clear to me how to apply the workaround
Someone could clarify?
Thanks

mundschenk-at, 2024-01-06:
Disable the `legacy_entity_attributes` setting, restart Z2M and (only if necessary) remove the device definition in HA via the dots menu in the device view of the bridge in HA.

Claudio1L, 2024-01-06:
Ok thanks

Claudio1L, 2024-01-06:
just to be sure, the device of the coordinator? This one?

[image]

Claudio1L, 2024-01-06:
ok, restarting the entire HA, error logs disappeared
Thanks

mundschenk-at, 2024-01-06:
Oh, yeah, reloading the MQTT integration (which happens when you restart HA) would probably also work. I don't know why I didn't try that when I had the issue on my test system.

mundschenk-at, 2024-01-07:
@Koenkk I think this issue can be closed now that #20548 is merged.

Koenkk, 2024-01-07:
@mundschenk-at and just release in 1.35.1 😄

PR #20548 (2024-01-04): Prevent warnings for bridge entities when legacy_entity_attributes is enabled
Prevents warnings for bridge entities when `legacy_entity_attributes` is enabled. Fixes #20488.
closed 2024-01-07 by Koenkk