Koenkk/zigbee2mqtt #20489: 1.35.0 - Bridge Connection state entity parsing undefined state key

RobertD502, 2024-01-01:
### What happened?

The Zigbee2Mqtt bridge entity `Connection state` is attempting to parse the `state` key, which is not present in the state_topic defined for this entity. As a result, Home Assistant raises two exceptions:

```
Logger: homeassistant.helpers.template
Source: helpers/template.py:2318
First occurred: 5:21:04 PM (1 occurrences)
Last logged: 5:21:04 PM

Template variable error: 'value_json' is undefined when rendering '{{ value_json.state }}'
```

```
Logger: homeassistant.helpers.template
Source: helpers/template.py:749
First occurred: 5:21:04 PM (1 occurrences)
Last logged: 5:21:04 PM

Error parsing value: 'value_json' is undefined (value: online, template: {{ value_json.state }})
```

As seen below, the entity is looking for the `state` key in the topic `zigbee/bridge/state`:

```json4
device:
  hw_version: zStack3x0 20200925
  identifiers:
    - zigbee2mqtt_bridge_0x00124b0021600270
  manufacturer: Zigbee2MQTT
  model: Bridge
  name: Zigbee2MQTT Bridge
  sw_version: 1.35.0
device_class: connectivity
entity_category: diagnostic
name: Connection state
object_id: zigbee2mqtt_bridge_connection_state
origin:
  name: Zigbee2MQTT
  sw_version: 1.35.0
  support_url: https://www.zigbee2mqtt.io
payload_off: offline
payload_on: online
state_topic: zigbee/bridge/state
unique_id: bridge_0x00124b0021600270_connection_state_zigbee
value_template: '{{ value_json.state }}'
platform: mqtt
```

However, Zigbee2MQTT publishes `online` / `offline` directly to the `zigbee/bridge/state` topic instead of within a `state` key at the `zigbee/bridge/state` topic. 

```
Received 5:21:04 PM
QoS: 0, Retained
Payload: online
```

### Zigbee2MQTT version

1.35.0

### Adapter firmware version

20200925

### Adapter

Electrolama

hmmbob, 2024-01-02:
Fix could be as simple as dropping this line, as there's no value_template needed in this case.

https://github.com/Koenkk/zigbee2mqtt/blob/master/lib/extension/homeassistant.ts#L1812

karmazinondrej, 2024-01-02:
@RobertD502 will this cause frequent disconnect of all sensors? Seems that all sensors I have are "reconnecting" frequntly, right after they are "Unavailable".

hmmbob, 2024-01-02:
It shouldn't - the issue is in the code for the bridge specifically, not for the sensors. Looking at my own sensors, all seems fine.

RobertD502, 2024-01-02:
^ correct. In particular this only affects the single entity within the Bridge Device that I mentioned in the title.

mundschenk-at, 2024-01-02:
This is caused by having `legacy_availability_payload` enabled. 

@Koenkk Do you want a workaround or should legacy options be retired?

RobertD502, 2024-01-02:
> This is caused by having `legacy_availability_payload` enabled. 
> 
> 
> 
> @Koenkk Do you want a workaround or should the legacy options be retired?

That indeed fixed it and started introducing the state key at said topic. Can't remember why I had that option enabled since it has been a few years.

hmmbob, 2024-01-02:
Yeah, same here. Can't remember setting this actively

Koenkk, 2024-01-02:
@mundschenk-at is there an easy way to do try/catch logic for the JSON parsing?

mundschenk-at, 2024-01-02:
You can use conditional logic in the `value_template`, but since that is evaluated frequently, it would be preferable to handle this during the generation of the discovery topic.

Koenkk, 2024-01-02:
Then I propose to do a check in `homeassistant.ts` to see if this option is enabled (you can skip adding test case for it, will ignore the coverage part). Would that be doable?

mundschenk-at, 2024-01-02:
Sure.

Koenkk, 2024-01-04:
Fixed, many thanks @mundschenk-at !

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

PR #20547 (2024-01-04): Support legacy_availability_payload for connection_state entity
Adds support for `legacy_availability_payload` setting in `connection_state` bridge entity discovery. Fixes #20489.
closed 2024-01-04 by Koenkk