Koenkk/zigbee2mqtt #20492: Lixee too verbose in the logs (#6791, #6672)

merlinpimpim, 2024-01-02:
### What happened?

In the (add-on) log, references to the Lixee module are constantly recurring (not in debug mode):

> zlinky config: standard, three_phase, false, BASE
zlinky config: standard, three_phase, false, BASE
zlinky config: standard, three_phase, false, BASE
zlinky config: standard, three_phase, false, BASE
zlinky config: standard, three_phase, false, BASE
zlinky config: standard, three_phase, false, BASE
zlinky config: standard, three_phase, false, BASE
zlinky config: standard, three_phase, false, BASE
zlinky config: standard, three_phase, false, BASE
zlinky config: standard, three_phase, false, BASE
zlinky config: standard, three_phase, false, BASE
Failed to read zigbee attributes: Error: Cluster 'liXeePrivate' has no attribute 'relais1'
Failed to read zigbee attributes: Error: Cluster 'liXeePrivate' has no attribute 'relais2'
Failed to read zigbee attributes: Error: Cluster 'liXeePrivate' has no attribute 'relais3'
Failed to read zigbee attributes: Error: Cluster 'liXeePrivate' has no attribute 'relais4'
Failed to read zigbee attributes: Error: Cluster 'liXeePrivate' has no attribute 'relais5'
Failed to read zigbee attributes: Error: Cluster 'liXeePrivate' has no attribute 'relais6'
Failed to read zigbee attributes: Error: Cluster 'liXeePrivate' has no attribute 'relais7'
Failed to read zigbee attributes: Error: Cluster 'liXeePrivate' has no attribute 'relais8'
zlinky config: standard, three_phase, false, BASE
Failed to read zigbee attributes: Error: Cluster 'liXeePrivate' has no attribute 'relais1'
Failed to read zigbee attributes: Error: Cluster 'liXeePrivate' has no attribute 'relais2'
Failed to read zigbee attributes: Error: Cluster 'liXeePrivate' has no attribute 'relais3'
Failed to read zigbee attributes: Error: Cluster 'liXeePrivate' has no attribute 'relais4'
Failed to read zigbee attributes: Error: Cluster 'liXeePrivate' has no attribute 'relais5'
Failed to read zigbee attributes: Error: Cluster 'liXeePrivate' has no attribute 'relais6'
Failed to read zigbee attributes: Error: Cluster 'liXeePrivate' has no attribute 'relais7'
Failed to read zigbee attributes: Error: Cluster 'liXeePrivate' has no attribute 'relais8'
zlinky config: standard, three_phase, false, BASE
Failed to read zigbee attributes: Error: Cluster 'liXeePrivate' has no attribute 'relais1'
Failed to read zigbee attributes: Error: Cluster 'liXeePrivate' has no attribute 'relais2'
Failed to read zigbee attributes: Error: Cluster 'liXeePrivate' has no attribute 'relais3'
Failed to read zigbee attributes: Error: Cluster 'liXeePrivate' has no attribute 'relais4'
Failed to read zigbee attributes: Error: Cluster 'liXeePrivate' has no attribute 'relais5'
Failed to read zigbee attributes: Error: Cluster 'liXeePrivate' has no attribute 'relais6'
Failed to read zigbee attributes: Error: Cluster 'liXeePrivate' has no attribute 'relais7'
Failed to read zigbee attributes: Error: Cluster 'liXeePrivate' has no attribute 'relais8'
zlinky config: standard, three_phase, false, BASE
Failed to read zigbee attributes: Error: Cluster 'liXeePrivate' has no attribute 'relais1'
Failed to read zigbee attributes: Error: Cluster 'liXeePrivate' has no attribute 'relais2'
Failed to read zigbee attributes: Error: Cluster 'liXeePrivate' has no attribute 'relais3'
Failed to read zigbee attributes: Error: Cluster 'liXeePrivate' has no attribute 'relais4'
Failed to read zigbee attributes: Error: Cluster 'liXeePrivate' has no attribute 'relais5'
Failed to read zigbee attributes: Error: Cluster 'liXeePrivate' has no attribute 'relais6'
Failed to read zigbee attributes: Error: Cluster 'liXeePrivate' has no attribute 'relais7'
Failed to read zigbee attributes: Error: Cluster 'liXeePrivate' has no attribute 'relais8'
zlinky config: standard, three_phase, false, BASE
Failed to read zigbee attributes: Error: Cluster 'liXeePrivate' has no attribute 'relais1'
Failed to read zigbee attributes: Error: Cluster 'liXeePrivate' has no attribute 'relais2'
Failed to read zigbee attributes: Error: Cluster 'liXeePrivate' has no attribute 'relais3'
Failed to read zigbee attributes: Error: Cluster 'liXeePrivate' has no attribute 'relais4'
Failed to read zigbee attributes: Error: Cluster 'liXeePrivate' has no attribute 'relais5'
Failed to read zigbee attributes: Error: Cluster 'liXeePrivate' has no attribute 'relais6'
Failed to read zigbee attributes: Error: Cluster 'liXeePrivate' has no attribute 'relais7'
Failed to read zigbee attributes: Error: Cluster 'liXeePrivate' has no attribute 'relais8'
zlinky config: standard, three_phase, false, BASE
Failed to read zigbee attributes: Error: Cluster 'liXeePrivate' has no attribute 'relais1'
Failed to read zigbee attributes: Error: Cluster 'liXeePrivate' has no attribute 'relais2'
Failed to read zigbee attributes: Error: Cluster 'liXeePrivate' has no attribute 'relais3'
Failed to read zigbee attributes: Error: Cluster 'liXeePrivate' has no attribute 'relais4'
Failed to read zigbee attributes: Error: Cluster 'liXeePrivate' has no attribute 'relais5'
Failed to read zigbee attributes: Error: Cluster 'liXeePrivate' has no attribute 'relais6'
Failed to read zigbee attributes: Error: Cluster 'liXeePrivate' has no attribute 'relais7'
Failed to read zigbee attributes: Error: Cluster 'liXeePrivate' has no attribute 'relais8'
zlinky config: standard, three_phase, false, BASE
Failed to read zigbee attributes: Error: Cluster 'liXeePrivate' has no attribute 'relais1'
Failed to read zigbee attributes: Error: Cluster 'liXeePrivate' has no attribute 'relais2'
Failed to read zigbee attributes: Error: Cluster 'liXeePrivate' has no attribute 'relais3'
Failed to read zigbee attributes: Error: Cluster 'liXeePrivate' has no attribute 'relais4'
Failed to read zigbee attributes: Error: Cluster 'liXeePrivate' has no attribute 'relais5'
Failed to read zigbee attributes: Error: Cluster 'liXeePrivate' has no attribute 'relais6'
Failed to read zigbee attributes: Error: Cluster 'liXeePrivate' has no attribute 'relais7'
Failed to read zigbee attributes: Error: Cluster 'liXeePrivate' has no attribute 'relais8'
zlinky config: standard, three_phase, false, BASE
Failed to read zigbee attributes: Error: Cluster 'liXeePrivate' has no attribute 'relais1'
Failed to read zigbee attributes: Error: Cluster 'liXeePrivate' has no attribute 'relais2'
Failed to read zigbee attributes: Error: Cluster 'liXeePrivate' has no attribute 'relais3'
Failed to read zigbee attributes: Error: Cluster 'liXeePrivate' has no attribute 'relais4'
Failed to read zigbee attributes: Error: Cluster 'liXeePrivate' has no attribute 'relais5'
Failed to read zigbee attributes: Error: Cluster 'liXeePrivate' has no attribute 'relais6'
Failed to read zigbee attributes: Error: Cluster 'liXeePrivate' has no attribute 'relais7'
Failed to read zigbee attributes: Error: Cluster 'liXeePrivate' has no attribute 'relais8'
zlinky config: standard, three_phase, false, BASE
Failed to read zigbee attributes: Error: Cluster 'liXeePrivate' has no attribute 'relais1'
Failed to read zigbee attributes: Error: Cluster 'liXeePrivate' has no attribute 'relais2'
Failed to read zigbee attributes: Error: Cluster 'liXeePrivate' has no attribute 'relais3'
Failed to read zigbee attributes: Error: Cluster 'liXeePrivate' has no attribute 'relais4'
Failed to read zigbee attributes: Error: Cluster 'liXeePrivate' has no attribute 'relais5'
Failed to read zigbee attributes: Error: Cluster 'liXeePrivate' has no attribute 'relais6'
Failed to read zigbee attributes: Error: Cluster 'liXeePrivate' has no attribute 'relais7'
Failed to read zigbee attributes: Error: Cluster 'liXeePrivate' has no attribute 'relais8'
zlinky config: standard, three_phase, false, BASE
Failed to read zigbee attributes: Error: Cluster 'liXeePrivate' has no attribute 'relais1'
Failed to read zigbee attributes: Error: Cluster 'liXeePrivate' has no attribute 'relais2'
Failed to read zigbee attributes: Error: Cluster 'liXeePrivate' has no attribute 'relais3'
Failed to read zigbee attributes: Error: Cluster 'liXeePrivate' has no attribute 'relais4'
Failed to read zigbee attributes: Error: Cluster 'liXeePrivate' has no attribute 'relais5'
Failed to read zigbee attributes: Error: Cluster 'liXeePrivate' has no attribute 'relais6'
Failed to read zigbee attributes: Error: Cluster 'liXeePrivate' has no attribute 'relais7'
Failed to read zigbee attributes: Error: Cluster 'liXeePrivate' has no attribute 'relais8'``

### What did you expect to happen?

No error, no log ? :)

### How to reproduce it (minimal and precise)

Nothing special

### Zigbee2MQTT version

1.35.0-1

### Adapter firmware version

7.3.1.0 build 0

### Adapter

Skyconnect

mildis, 2024-01-02:
HI @merlinpimpim, sorry for the inconvenience.
This is part of https://github.com/Koenkk/zigbee-herdsman-converters/pull/6791 adding the virtual relay values.
@Koenkk : as these are computed values extracted from a real unique value retrieved from Zigbee, should they really be exposed ? (see lines 599-606) There is no call to the zigbee source to be made for these ones.

Koenkk, 2024-01-02:
@mildis I think these shouldn't be exposed indeed, removed them.

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

closed 2024-01-02 by Koenkk

steve41200, 2024-01-27:
I've exactly the same log in 1.35.1-dev commit: [97eac16](https://github.com/Koenkk/zigbee2mqtt/commit/97eac16)
`zlinky config: historique, single_phase, false, BASE
zlinky config: historique, single_phase, false, BASE
zlinky config: historique, single_phase, false, BASE
zlinky config: historique, single_phase, false, BASE
zlinky config: historique, single_phase, false, BASE
Failed to read zigbee attributes: Error: Cluster 'liXeePrivate' has no attribute 'motDEtat'
zlinky config: historique, single_phase, false, BASE
zlinky config: historique, single_phase, false, BASE
zlinky config: historique, single_phase, false, BASE
zlinky config: historique, single_phase, false, BASE
zlinky config: historique, single_phase, false, BASE
zlinky config: historique, single_phase, false, BASE
zlinky config: historique, single_phase, false, BASE
zlinky config: historique, single_phase, false, BASE
zlinky config: historique, single_phase, false, BASE
zlinky config: historique, single_phase, false, BASE
Failed to read zigbee attributes: Error: Cluster 'liXeePrivate' has no attribute 'motDEtat'
zlinky config: historique, single_phase, false, BASE
zlinky config: historique, single_phase, false, BASE
zlinky config: historique, single_phase, false, BASE
zlinky config: historique, single_phase, false, BASE
zlinky config: historique, single_phase, false, BASE
zlinky config: historique, single_phase, false, BASE
zlinky config: historique, single_phase, false, BASE
zlinky config: historique, single_phase, false, BASE
zlinky config: historique, single_phase, false, BASE
Failed to read zigbee attributes: Error: Cluster 'liXeePrivate' has no attribute 'motDEtat'`

When the log are in warn mode.
Before I was not in this mode but in normal mode with a lot of info in log.
I don't know if this is new
.

The failed to read is :

MOTDETAT
Meter Status Word
Sans valeur

mildis, 2024-01-27:
Hi @steve41200,
This issue was initially related to a patch regarding the relais values.
It's been solved and closed.
I think you should open a new issue for your case.