Koenkk/zigbee2mqtt #20494: Since upgrading to 1.35.0-1 in Home Assistant I get no attibute/invlaid power_on_behavior errors in my HA log

heyitsyang, 2024-01-02:
### What happened?

Since upgrading to Zigbee2MQTT 1.35.0-1 in Home Assistant I get the following errors in my Home Assistant log:

2024-01-01 10:04:49.710 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'power_on_behavior' when rendering '{{ value_json.power_on_behavior }}'
2024-01-01 10:04:49.711 ERROR (MainThread) [homeassistant.components.mqtt.select] Invalid option for select.master_closet_lights_power_on_behavior: '' (valid options: ['off', 'on', 'toggle', 'previous'])
2024-01-01 10:04:49.745 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'power_on_behavior' when rendering '{{ value_json.power_on_behavior }}'
2024-01-01 10:04:49.746 ERROR (MainThread) [homeassistant.components.mqtt.select] Invalid option for select.island_barstool_outlet_power_on_behavior: '' (valid options: ['off', 'on', 'toggle', 'previous'])

### What did you expect to happen?

These warnings and errors only appeared after the 1.3.5.0-1 upgrade.  The devices seem to work otherwise. Please fix the underlying cause of the errors .

### How to reproduce it (minimal and precise)

In Home Assistant, any action involving the devices will cause the error.  They are both GE Enbrighten Zigbee devices.

### Zigbee2MQTT version

1.35.0.-1

### Adapter firmware version

    6.10.3.0 build 297

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus-E

Hewelius, 2024-01-03:
I have the same issue. In some devices appears new attributes, not present before update to 1.35. And z2m reporting errors "unsupported attribute"

msbc42, 2024-01-09:
I am also seeing this issue - my adapter is SkyConnect.

avvalex, 2024-01-14:
Me too, with Vimar 14592.0

FuturamaPhill, 2024-03-01:
Also having this issue, is there not an option to disable the unsupported feature per device?

closed 2024-09-28 by github-actions[bot]