Koenkk/zigbee2mqtt #20495: 1.35.0 - Custom Zigbee sensor not longer supported (ptvo.info)

misterobotique, 2024-01-02:
### What happened?

After upgrading to 1.35.0, my custom Zigbee device appears "unsupported" in [Devices] panel despite the presence of an external converter.
My custom device is based on ptvo.info configurable firmware and is configured to send and receive UART messages that are JSON formatted.

My converter is set as follow:
```
const zigbeeHerdsmanConverters = require('zigbee-herdsman-converters');

const exposes = zigbeeHerdsmanConverters.exposes;
const ea = exposes.access;
const e = exposes.presets;
const fz = zigbeeHerdsmanConverters.fromZigbeeConverters;
const tz = zigbeeHerdsmanConverters.toZigbeeConverters;

const ptvo_switch = zigbeeHerdsmanConverters.findByDevice({modelID: 'ptvo.switch'});
fz.legacy = ptvo_switch.meta.tuyaThermostatPreset;

const device = {
    zigbeeModel: ['RAS-LJ22Y_ADPT'],
    model: 'RAS-LJ22Y_ADPT',
    vendor: 'Custom devices (DiY)',
    description: 'Zigbee adapter for Hitachi air conditioner',
    fromZigbee: [fz.ignore_basic_report, fz.ptvo_switch_uart],
    toZigbee: [tz.ptvo_switch_trigger, tz.ptvo_switch_uart],
    exposes: [exposes.text('action', ea.STATE_SET).withDescription('button clicks or data from/to UART'), e.linkquality()],
    meta: {
        multiEndpoint: true,

    },
    endpoint: (device) => {
        return {
            l1: 1, action: 1,
        };
    },
};

module.exports = device;
```

Zigbee2MQTT reports the following when receiving a message from my custom device ([Logs] panel in Debug mode):
 _-> see Debug log_

I presume because my device is unsupported, the received message is recognized as a string of decimals rather than ASCII; content that can be seen in below _Debug log_ gives the following when converted to ASCII:
`{"MODE":"heat","FAN":"auto","TMP":21,"IN_TMP":23}`

I tried to follow the tutorial [Support new devices](https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html) and created a new converter but the problem remains.

### What did you expect to happen?

Converter to still works with latest version and no particular problem with my custom Zigbee device.

### Zigbee2MQTT version

1.35.0

### Adapter firmware version

20210120

### Adapter

Electrolama zig-a-zig-ah! (zzh!)

### Debug log

```
Debug 2024-01-02 10:58:13Received Zigbee message from 'living_room_A/C', type 'attributeReport', cluster 'genMultistateValue', data '{"stateText":{"data":[123,34,77,79,68,69,34,58,34,104,101,97,116,34,44,34,70,65,78,34,58,34,97,117,116,111,34,44,34,84,77,80,34,58,50,49,44,34,73,78,95,84,77,80,34,58,50,51,125],"type":"Buffer"}}' from endpoint 1 with groupID 0
Debug 2024-01-02 10:58:13No converter available for 'RAS-LJ22Y_ADPT' with cluster 'genMultistateValue' and type 'attributeReport' and data '{"stateText":{"data":[123,34,77,79,68,69,34,58,34,104,101,97,116,34,44,34,70,65,78,34,58,34,97,117,116,111,34,44,34,84,77,80,34,58,50,49,44,34,73,78,95,84,77,80,34,58,50,51,125],"type":"Buffer"}}'
```

smarecki, 2024-01-02:
Hello Happy New Year!
I have the same problem with external converters. Z2M cannot load converters. I have many devices, from temperature measurements to boiler PWN modulation.

Mar1usW3, 2024-01-02:
maybe this will help you https://ptvo.info/zigbee2mqtt-1-35-failed-to-start-1368/

smarecki, 2024-01-02:
Tested Option#2 - same problem. I will try option1, but Imust find HEX.

Flavien, 2024-01-02:
This worked for me:

```javascript
const zigbeeHerdsmanConverters = require('zigbee-herdsman-converters');
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
```

customstudioyandexru, 2024-01-02:
I also have a problem with the custom sensor after the update. (((

bachy, 2024-01-02:
+1 
2 custom pzem with ptvo firmware working very good for months lost all their entities with z2m 1.35.0 in Home assistant.

Koenkk, 2024-01-02:
3 changes have to be made to make the external converter compatible again, I've listed them [here](https://gist.github.com/Koenkk/59447e70454ac1e976e39ee65cfabb78)

closed 2024-01-02 by Koenkk