Koenkk/zigbee2mqtt #20497: Aqara Zigbee 3.0 LED strip T1 no RGB-colors configurable

CrazyThunderbird, 2024-01-02:
### What happened?

I don't know exactly when it happened but i assum either the last z2m update or the ota broke something.
Because since then:
I'm no longer able to define any RGB colors!
I can just switch it on/off and define the white color temperature and brightness.

What did i do:
Completely removed the device from z2m and completely rebootet the system and then paired it again, didn't work
Unpowered it for approx an hour, also didn't fix anything
I switched to z2m edge for testing, also bummer :(

[image]
[image]

### Zigbee2MQTT version

1.35.0

### Adapter firmware version

20230507

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus,Zigbee Gateway TI CC2652P + CP2102N

KevinBalkoski, 2024-01-02:
Best I can tell, it may have been from these changes: https://github.com/Koenkk/zigbee-herdsman-converters/compare/v15.131.0...v18.3.0#diff-413b42ed3dc2bb32fbcfbc4db8b0242f76ff72d0e313986424e2f8a319baeb2bL979-L1060

zigbee-herdsman-converters got upgraded from 15.131.0 to 18.3.0 for zigbee2mqtt 1.35.0

Koenkk, 2024-01-02:
Will be fixed in https://github.com/Koenkk/zigbee-herdsman-converters/pull/6812

CrazyThunderbird, 2024-01-02:
When is the fix expected to drop? :)

Glad for this fast response

Koenkk, 2024-01-02:
Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

Additionally I will probably make a hotfix release somewhere in the coming week.

closed 2024-01-02 by Koenkk

CrazyThunderbird, 2024-01-02:
oh i'l give it a try in "edge" then and will report back :)

CrazyThunderbird, 2024-01-03:
Tested the Edge branch:
It's fixed
[image]

Ot4eVash, 2024-01-17:
[image]

I have the Edge. But it's not help((

cablesandcoffee, 2024-10-26:
Any news on this? I have the same problem. Still does not work running z2m 1.39.1

MadSpindel, 2024-11-24:
How do you put the LED Strip in pairing mode?