Koenkk/zigbee2mqtt #20499: Exception while calling fromZigbee converter: Cannot set properties of undefined (setting 'power')} since update to 1.35

JCinside13, 2024-01-02:
### What happened?

Since the update to 1.35, I have the error "Exception while calling fromZigbee converter: Cannot set properties of undefined (setting 'power')}" more than every second.

### What did you expect to happen?

I did not have these error before

### How to reproduce it (minimal and precise)

I did not something else than starting Zigbee2MQTT after I updated it.

### Zigbee2MQTT version

1.35.0 commit: 3d8f0fc3

### Adapter firmware version

0x26580700

### Adapter

ConBee2 

### Debug log

Error 2024-01-02 10:04:48Exception while calling fromZigbee converter: Cannot set properties of undefined (setting 'power')}
Error 2024-01-02 10:04:49Exception while calling fromZigbee converter: Cannot set properties of undefined (setting 'power')}
Error 2024-01-02 10:04:49Exception while calling fromZigbee converter: Cannot set properties of undefined (setting 'power')}
Error 2024-01-02 10:04:49Exception while calling fromZigbee converter: Cannot set properties of undefined (setting 'power')}
Error 2024-01-02 10:04:49Exception while calling fromZigbee converter: Cannot set properties of undefined (setting 'power')}
Error 2024-01-02 10:04:50Exception while calling fromZigbee converter: Cannot set properties of undefined (setting 'power')}
Error 2024-01-02 10:04:50Exception while calling fromZigbee converter: Cannot set properties of undefined (setting 'power')}
Error 2024-01-02 10:04:50Exception while calling fromZigbee converter: Cannot set properties of undefined (setting 'power')}
Error 2024-01-02 10:04:50Exception while calling fromZigbee converter: Cannot set properties of undefined (setting 'power')}
Error 2024-01-02 10:04:51Exception while calling fromZigbee converter: Cannot set properties of undefined (setting 'power')}
Error 2024-01-02 10:04:51Exception while calling fromZigbee converter: Cannot set properties of undefined (setting 'power')}
Error 2024-01-02 10:04:51Exception while calling fromZigbee converter: Cannot set properties of undefined (setting 'power')}
Error 2024-01-02 10:04:51Exception while calling fromZigbee converter: Cannot set properties of undefined (setting 'power')}
Error 2024-01-02 10:04:51Exception while calling fromZigbee converter: Cannot set properties of undefined (setting 'power')}
Error 2024-01-02 10:04:52Exception while calling fromZigbee converter: Cannot set properties of undefined (setting 'power')}
Error 2024-01-02 10:04:52Exception while calling fromZigbee converter: Cannot set properties of undefined (setting 'power')}
Error 2024-01-02 10:04:52Exception while calling fromZigbee converter: Cannot set properties of undefined (setting 'power')}

JCinside13, 2024-01-02:
I restarted Z2M ans I have no more error...

PSMGoossens, 2024-01-02:
I've the same issue since today. Reboot didn't help :(

djsyl, 2024-01-02:
same issue since 1.35.0

PSMGoossens, 2024-01-02:
Some relevant logging
`info  2024-01-02 19:12:33: MQTT publish: topic 'zigbee2mqtt/Stekker PC', payload '{"child_lock":"UNLOCK","current":0,"device":{"applicationVersion":192,"dateCode":"","friendlyName":"Stekker PC","hardwareVersion":1,"ieeeAddr":"0xa4c138d359a38661","manufacturerID":4417,"manufacturerName":"_TZ3000_zloso4jk","model":"TS011F_plug_1","networkAddress":55101,"powerSource":"Mains (single phase)","stackVersion":0,"type":"Router","zclVersion":3},"energy":89.75,"indicator_mode":"off","last_seen":"2024-01-02T18:12:33.780Z","linkquality":174,"power":0,"power_outage_memory":"restore","state":"OFF","update":{"installed_version":192,"latest_version":192,"state":"idle"},"update_available":false,"voltage":235}'
debug 2024-01-02 19:12:33: Received Zigbee message from 'Stekker PC', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":0,"rmsCurrent":0,"rmsVoltage":235}' from endpoint 1 with groupID 0
error 2024-01-02 19:12:33: Exception while calling fromZigbee converter: Cannot set properties of undefined (setting 'power')}
debug 2024-01-02 19:12:33: TypeError: Cannot set properties of undefined (setting 'power')
    at Object.convert (/app/node_modules/zigbee-herdsman-converters/src/devices/tuya.ts:477:29)
    at processTicksAndRejections (node:internal/process/task_queues:95:5)
    at Receive.onDeviceMessage (/app/lib/extension/receive.ts:147:35)
info  2024-01-02 19:12:33: MQTT publish: topic 'zigbee2mqtt/Stekker PC', payload '{"child_lock":"UNLOCK","current":0,"device":{"applicationVersion":192,"dateCode":"","friendlyName":"Stekker PC","hardwareVersion":1,"ieeeAddr":"0xa4c138d359a38661","manufacturerID":4417,"manufacturerName":"_TZ3000_zloso4jk","model":"TS011F_plug_1","networkAddress":55101,"powerSource":"Mains (single phase)","stackVersion":0,"type":"Router","zclVersion":3},"energy":89.75,"indicator_mode":"off","last_seen":"2024-01-02T18:12:33.983Z","linkquality":178,"power":0,"power_outage_memory":"restore","state":"OFF","update":{"installed_version":192,"latest_version":192,"state":"idle"},"update_available":false,"voltage":235}'
debug 2024-01-02 19:12:34: Received Zigbee message from 'Stekker PC', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":0,"rmsCurrent":0,"rmsVoltage":235}' from endpoint 1 with groupID 0
error 2024-01-02 19:12:34: Exception while calling fromZigbee converter: Cannot set properties of undefined (setting 'power')}
debug 2024-01-02 19:12:34: TypeError: Cannot set properties of undefined (setting 'power')
    at Object.convert (/app/node_modules/zigbee-herdsman-converters/src/devices/tuya.ts:477:29)
    at processTicksAndRejections (node:internal/process/task_queues:95:5)
    at Receive.onDeviceMessage (/app/lib/extension/receive.ts:147:35)
`

JCinside13, 2024-01-02:
With debug  :
 
2024-01-02 19:45:48Received Zigbee message from 'Prise TV Constance', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":0,"rmsCurrent":0,"rmsVoltage":222}' from endpoint 1 with groupID null
2024-01-02 19:45:48TypeError: Cannot set properties of undefined (setting 'power') at Object.convert (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/src/devices/tuya.ts:477:29) at Receive.onDeviceMessage (/opt/zigbee2mqtt/lib/extension/receive.ts:147:35)
2024-01-02 19:45:48Received Zigbee message from 'Prise Serveur video surveillance', type 'attributeReport', cluster 'genBasic', data '{"65506":54,"65508":0,"appVersion":192}' from endpoint 1 with groupID null
2024-01-02 19:45:48Received Zigbee message from 'Prise TV Constance', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":0,"rmsCurrent":0,"rmsVoltage":222}' from endpoint 1 with groupID null
2024-01-02 19:45:48TypeError: Cannot set properties of undefined (setting 'power') at Object.convert (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/src/devices/tuya.ts:477:29) at Receive.onDeviceMessage (/opt/zigbee2mqtt/lib/extension/receive.ts:147:35)
2024-01-02 19:45:48Received Zigbee message from 'Prise TV Constance', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":0,"rmsCurrent":0,"rmsVoltage":222}' from endpoint 1 with groupID null
2024-01-02 19:45:48TypeError: Cannot set properties of undefined (setting 'power') at Object.convert (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/src/devices/tuya.ts:477:29) at Receive.onDeviceMessage (/opt/zigbee2mqtt/lib/extension/receive.ts:147:35)
2024-01-02 19:45:48Received Zigbee message from 'Prise TV Constance', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":0,"rmsCurrent":0,"rmsVoltage":222}' from endpoint 1 with groupID null
2024-01-02 19:45:48TypeError: Cannot set properties of undefined (setting 'power') at Object.convert (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/src/devices/tuya.ts:477:29) at Receive.onDeviceMessage (/opt/zigbee2mqtt/lib/extension/receive.ts:147:35)
2024-01-02 19:45:49Received Zigbee message from 'Prise TV Constance', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":0,"rmsCurrent":0,"rmsVoltage":222}' from endpoint 1 with groupID null
2024-01-02 19:45:49TypeError: Cannot set properties of undefined (setting 'power') at Object.convert (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/src/devices/tuya.ts:477:29) at Receive.onDeviceMessage (/opt/zigbee2mqtt/lib/extension/receive.ts:147:35)
2024-01-02 19:45:49Received Zigbee message from 'Prise TV Constance', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":0,"rmsCurrent":0,"rmsVoltage":222}' from endpoint 1 with groupID null
2024-01-02 19:45:49TypeError: Cannot set properties of undefined (setting 'power') at Object.convert (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/src/devices/tuya.ts:477:29) at Receive.onDeviceMessage (/opt/zigbee2mqtt/lib/extension/receive.ts:147:35)
2024-01-02 19:45:49Received Zigbee message from 'Prise TV Constance', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":0,"rmsCurrent":0,"rmsVoltage":222}' from endpoint 1 with groupID null
2024-01-02 19:45:49TypeError: Cannot set properties of undefined (setting 'power') at Object.convert (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/src/devices/tuya.ts:477:29) at Receive.onDeviceMessage (/opt/zigbee2mqtt/lib/extension/receive.ts:147:35)
2024-01-02 19:45:49Received Zigbee message from 'Prise TV Constance', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":0,"rmsCurrent":0,"rmsVoltage":222}' from endpoint 1 with groupID null
2024-01-02 19:45:49TypeError: Cannot set properties of undefined (setting 'power') at Object.convert (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/src/devices/tuya.ts:477:29) at Receive.onDeviceMessage (/opt/zigbee2mqtt/lib/extension/receive.ts:147:35)
2024-01-02 19:45:49Received Zigbee message from 'Prise TV Constance', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":0,"rmsCurrent":0,"rmsVoltage":222}' from endpoint 1 with groupID null
2024-01-02 19:45:49TypeError: Cannot set properties of undefined (setting 'power') at Object.convert (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/src/devices/tuya.ts:477:29) at Receive.onDeviceMessage (/opt/zigbee2mqtt/lib/extension/receive.ts:147:35)
2024-01-02 19:45:50Received Zigbee message from 'Prise TV Constance', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":0,"rmsCurrent":0,"rmsVoltage":222}' from endpoint 1 with groupID null
2024-01-02 19:45:50TypeError: Cannot set properties of undefined (setting 'power') at Object.convert (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/src/devices/tuya.ts:477:29) at Receive.onDeviceMessage (/opt/zigbee2mqtt/lib/extension/receive.ts:147:35)
2024-01-02 19:45:50Received Zigbee message from 'Prise TV Constance', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":0,"rmsCurrent":0,"rmsVoltage":222}' from endpoint 1 with groupID null
2024-01-02 19:45:50TypeError: Cannot set properties of undefined (setting 'power') at Object.convert (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/src/devices/tuya.ts:477:29) at Receive.onDevic [... cut ...]
2024-01-02 19:45:54Received Zigbee message from 'Prise TV Constance', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":0,"rmsCurrent":0,"rmsVoltage":222}' from endpoint 1 with groupID null
2024-01-02 19:45:54TypeError: Cannot set properties of undefined (setting 'power') at Object.convert (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/src/devices/tuya.ts:477:29) at Receive.onDeviceMessage (/opt/zigbee2mqtt/lib/extension/receive.ts:147:35)
2024-01-02 19:45:54Received Zigbee message from 'Prise TV salon', type 'attributeReport', cluster 'seMetering', data '{"currentSummDelivered":[0,1688]}' from endpoint 1 with groupID null

Koenkk, 2024-01-02:
fixed! 

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

DutchTaurus, 2024-01-03:
> fixed!
> 
> Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

Hi Koen,

Because of this issue I tried to switch to the dev branch following your instructions but can't. For me this fails after: git checkout dev -> error: pathspec 'dev' did not match any file(s) known to git. (I (re-)installed zigbee2mqtt per the latest linux install instructions on your webpage.)

Following advise on the web I get after executing 'git show-ref':
3d8f0fc3a8710409e241ef74bea79a2dc61495f7 refs/heads/master
3d8f0fc3a8710409e241ef74bea79a2dc61495f7 refs/remotes/origin/HEAD
3d8f0fc3a8710409e241ef74bea79a2dc61495f7 refs/remotes/origin/master

To me this means 'dev' is missing but I'm by far a git expert :-), so could you help me out please? Thx!

pawol, 2024-01-03:
I think 1.35 should be rolled back. TRV GTZ06 also has 50% of features not working after update

chrizzlibaer, 2024-01-03:
For me, 1.35 is a huge win overall. Had many strange issues that seem to be just gone after upgrading besides the error regarding this issue.
Hope your feature set can be fixed quickly @pawol 🙂

marazmarci, 2024-01-03:
@DutchTaurus try running `git fetch` first 😉

DutchTaurus, 2024-01-04:
> @DutchTaurus try running `git fetch` first 😉

@marazmarci Thank you for your reply. I followed the link @Koenkk provided to switch to the dev branch and followed those instructions to the letter. It lists a 'git fetch origin dev' command before the 'git checkout dev', I assume you are referring to that fetch operation? The 'fetch' command succeeds/doesn't throw an error, the 'checkout' throws the error I mentioned. After this my system is in a non-working state.

JCinside13, 2024-01-04:
I updated to the dev branch but the problem IS still here.
My version : Zigbee2MQTT 1.35.0 commit: [3d8f0fc3]

Some  log :
Exception while calling fromZigbee converter: Cannot set properties of undefined (setting 'power')}
Exception while calling fromZigbee converter: Cannot set properties of undefined (setting 'power')}
Exception while calling fromZigbee converter: Cannot set properties of undefined (setting 'power')}
Exception while calling fromZigbee converter: Cannot set properties of undefined (setting 'power')}
Exception while calling fromZigbee converter: Cannot set properties of undefined (setting 'power')}
Exception while calling fromZigbee converter: Cannot set properties of undefined (setting 'power')}
Exception while calling fromZigbee converter: Cannot set properties of undefined (setting 'power')}
Exception while calling fromZigbee converter: Cannot set properties of undefined (setting 'power')}
Exception while calling fromZigbee converter: Cannot set properties of undefined (setting 'power')}
Exception while calling fromZigbee converter: Cannot set properties of undefined (setting 'power')}
Exception while calling fromZigbee converter: Cannot set properties of undefined (setting 'power')}
Exception while calling fromZigbee converter: Cannot set properties of undefined (setting 'power')}
Exception while calling fromZigbee converter: Cannot set properties of undefined (setting 'power')}
Exception while calling fromZigbee converter: Cannot set properties of undefined (setting 'power')}
Exception while calling fromZigbee converter: Cannot set properties of undefined (setting 'power')}
Exception while calling fromZigbee converter: Cannot set properties of undefined (setting 'power')}
Exception while calling fromZigbee converter: Cannot set properties of undefined (setting 'power')}
Exception while calling fromZigbee converter: Cannot set properties of undefined (setting 'power')}
Exception while calling fromZigbee converter: Cannot set properties of undefined (setting 'power')}
Exception while calling fromZigbee converter: Cannot set properties of undefined (setting 'power')}
Exception while calling fromZigbee converter: Cannot set properties of undefined (setting 'power')}
Exception while calling fromZigbee converter: Cannot set properties of undefined (setting 'power')}
Exception while calling fromZigbee converter: Cannot set properties of undefined (setting 'power')}
Exception while calling fromZigbee converter: Cannot set properties of undefined (setting 'power')}
Exception while calling fromZigbee converter: Cannot set properties of undefined (setting 'power')}
Exception while calling fromZigbee converter: Cannot set properties of undefined (setting 'power')}
Exception while calling fromZigbee converter: Cannot set properties of undefined (setting 'power')}
Exception while calling fromZigbee converter: Cannot set properties of undefined (setting 'power')}
Exception while calling fromZigbee converter: Cannot set properties of undefined (setting 'power')}
Exception while calling fromZigbee converter: Cannot set properties of undefined (setting 'power')}
Exception while calling fromZigbee converter: Cannot set properties of undefined (setting 'power')}
Exception while calling fromZigbee converter: Cannot set properties of undefined (setting 'power')}
Exception while calling fromZigbee converter: Cannot set properties of undefined (setting 'power')}
Exception while calling fromZigbee converter: Cannot set properties of undefined (setting 'power')}
Exception while calling fromZigbee converter: Cannot set properties of undefined (setting 'power')}
Exception while calling fromZigbee converter: Cannot set properties of undefined (setting 'power')}
Exception while calling fromZigbee converter: Cannot set properties of undefined (setting 'power')}
Exception while calling fromZigbee converter: Cannot set properties of undefined (setting 'power')}
Exception while calling fromZigbee converter: Cannot set properties of undefined (setting 'power')}
Exception while calling fromZigbee converter: Cannot set properties of undefined (setting 'power')}
Exception while calling fromZigbee converter: Cannot set properties of undefined (setting 'power')}
Exception while calling fromZigbee converter: Cannot set properties of undefined (setting 'power')}
Exception while calling fromZigbee converter: Cannot set properties of undefined (setting 'power')}
Exception while calling fromZigbee converter: Cannot set properties of undefined (setting 'power')}
Exception while calling fromZigbee converter: Cannot set properties of undefined (setting 'power')}
Exception while calling fromZigbee converter: Cannot set properties of undefined (setting 'power')}
Exception while calling fromZigbee converter: Cannot set properties of undefined (setting 'power')}
Exception while calling fromZigbee converter: Cannot set properties of undefined (setting 'power')}
Exception while calling fromZigbee converter: Cannot set properties of undefined (setting 'power')}
 
2024-01-04 21:07:44Received Zigbee message from 'Prise TV Constance', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":0,"rmsCurrent":0,"rmsVoltage":227}' from endpoint 1 with groupID null
2024-01-04 21:07:44TypeError: Cannot set properties of undefined (setting 'power') at Object.convert (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/src/devices/tuya.ts:477:29) at Receive.onDeviceMessage (/opt/zigbee2mqtt/lib/extension/receive.t [... cut ...] th groupID null
2024-01-04 21:07:51TypeError: Cannot set properties of undefined (setting 'power') at Object.convert (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/src/devices/tuya.ts:477:29) at Receive.onDeviceMessage (/opt/zigbee2mqtt/lib/extension/receive.ts:147:35)
2024-01-04 21:07:51Received Zigbee message from 'Prise TV Constance', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":0,"rmsCurrent":0,"rmsVoltage":227}' from endpoint 1 with groupID null

Koenkk, 2024-01-05:
@JCinside13 you are not using the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

IvanSivric, 2024-01-05:
> fixed!
> 
> Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

It works. Thanks

closed 2024-01-05 by Koenkk

aperezva, 2024-01-06:
Hi  

I have the same problem, when wil you fix it in a final version?. I don´t want to switc to dev branch @Koenkk ?

Thanks

Koenkk, 2024-01-07:
I will push a hotfix release in the coming days.

aperezva, 2024-01-07:
> I will push a hotfix release in the coming days.

Thanks a lot

JCinside13, 2024-01-07:
> I will push a hotfix release in the coming days.

Many thanks for your work