Koenkk/zigbee2mqtt #20500: Smart Zigbee energy meter 80A din rail (DAC2161C) by MatSee Plus

MBRetrofit, 2024-01-02:
### What happened?

After latest z2m update this device refresh data after 12 or more minutes. On previous version the refresh was a 2,5 minute and I was hoping for a solution to speed up this process.

### What did you expect to happen?

As this product is quite nice, a new converter or OTA firmware update it will be much appreciated 

### How to reproduce it (minimal and precise)

I think that from latest z2m update this delay is based on device parameters change, as if the Power is almost constant the z2m doesn't listen and doesn't update the status.

### Zigbee2MQTT version

1.35.0-1

### Adapter firmware version

2023.12.4

### Adapter

MatSee Plus DAC2161C

jabss, 2024-01-18:
I also have a similar issue, I thought it was the device loosing connection, but now it seems it was due to an update on ZB2MQTT. :(
My device does an update less than once a day. Becomes useless.
Tried to force availability on it, but the behavior remains the same:

`'0xXXXXXXc946XXXXXX':
    friendly_name: Power Meter
    voltage_calibration: ''
    power_calibration: ''
    current_calibration: ''
    energy_calibration: ''
    availability: true
`

jabss, 2024-01-22:
I have 2 errors every other minute in the logs and it seems it is always after the Power Meter sent his update. Not 100% sure, but it should be related:

> **debug** 2024-01-22 18:30:33Received Zigbee message from 'Power Meter', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[8,226,0,6,157,0,0,202],"type":"Buffer"},"datatype":0,"dp":6}],"seq":53015}' from endpoint 1 with groupID 0

> **error** 2024-01-22 18:30:33Exception while calling fromZigbee converter: tuya.getDataValue is not a function}

> **debug** 2024-01-22 18:30:33TypeError: tuya.getDataValue is not a function at Object.convert (/app/data/extension/externally-loaded.js:18:36) at Receive.onDeviceMessage (/app/lib/extension/receive.ts:147:51) at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:167:23) at EventEmitter.emit (node:events:529:35) at EventBus.emitDeviceMessage (/app/lib/eventBus.ts:101:22) at Controller.<anonymous> (/app/lib/zigbee.ts:115:27)

> **debug** 2024-01-22 18:30:33Received Zigbee message from 'Power Meter', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,6,2,124],"type":"Buffer"},"datatype":2,"dp":1}],"seq":53271}' from endpoint 1 with groupID 0

> **error** 2024-01-22 18:30:33Exception while calling fromZigbee converter: tuya.getDataValue is not a function}

> **debug** 2024-01-22 18:30:33TypeError: tuya.getDataValue is not a function at Object.convert (/app/data/extension/externally-loaded.js:18:36) at Receive.onDeviceMessage (/app/lib/extension/receive.ts:147:51) at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:167:23) at EventEmitter.emit (node:events:529:35) at EventBus.emitDeviceMessage (/app/lib/eventBus.ts:101:22) at Controller.<anonymous> (/app/lib/zigbee.ts:115:27)

Not sure what to do with this, simply sharing it here so someone more knowledgeable could find it useful.
Any way to import this specific adapter as it was previously without breaking anything else?

Thanks

jabss, 2024-03-16:
Any workaround that could be used while this issue isn't fixed?

jabss, 2024-09-13:
I've installed ZB2MQTT [1.39.1 commit: [e132316] and it started working again. 
Still some errors (see below example) but I can get the power consumption and store it in FluxDB.

`zhc: Failed to apply calibration to 'current': 'current_calibration' is not a number, got string ()`

closed 2025-01-09 by github-actions[bot]