Koenkk/zigbee2mqtt #20501: SSM-U01 Aqara T1 with neutral not reporting power consumption

vlebourl, 2024-01-02:
### What happened?

I have an Aqara T1s that does not report any power or energy consumption. I have tried several versions of z2m, up to latest 1.35.0-dev commit: [84387b2](https://github.com/Koenkk/zigbee2mqtt/commit/84387b2). I have had this device for a while (a year or so) and it never reported power usage...

How can I find whether the device is actually sending the info? What can I do to get the power measurement feature?
Any help would be very very welcomed!

Here are some more details while the device plugged to the T1 is active and drawing about 800W:
[image]
[image]
[image]
[image]

here are some extra logs:
```
Zigbee2MQTT:debug 2023-12-20 11:18:17: Received Zigbee message from 'Fil Pilote Lenaic', type 'attributeReport', cluster 'genDeviceTempCfg', data '{"currentTemperature":21}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2023-12-20 11:18:17: MQTT publish: topic 'zigbee2mqtt_casa/Fil Pilote Lenaic', payload '{"consumption":0,"current":0,"device_temperature":21,"energy":0,"linkquality":91,"power":0,"power_outage_count":0,"power_outage_memory":true,"state":"OFF","switch_type":"toggle","update":{"installed_version":23,"latest_version":23,"state":"idle"},"update_available":null,"voltage":236.49}'
Zigbee2MQTT:debug 2023-12-20 11:18:23: Received Zigbee message from 'Fil Pilote Lenaic', type 'attributeReport', cluster 'aqaraOpple', data '{"247":{"data":[100,16,0,3,40,21,152,57,0,0,0,0,149,57,0,0,0,0,150,57,0,58,19,69,151,57,0,0,0,0,5,33,1,0,154,32,0,11,32,0,8,33,23,1,9,33,0,1],"type":"Buffer"}}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2023-12-20 11:18:23: MQTT publish: topic 'zigbee2mqtt_casa/Fil Pilote Lenaic', payload '{"consumption":0,"current":0,"device_temperature":21,"energy":0,"linkquality":91,"power":0,"power_outage_count":0,"power_outage_memory":true,"state":"OFF","switch_type":"toggle","update":{"installed_version":23,"latest_version":23,"state":"idle"},"update_available":null,"voltage":235.56}'
```

### Zigbee2MQTT version

1.35.0-dev commit: 84387b2

### Adapter firmware version

20230507

### Adapter

zStack3x0 (SONOFF ZigBee 3.0 USB Dongle Plus, TI CC2652P)

### Debug log

```
2024-01-02 11:18:51Received Zigbee message from 'Fil Pilote Bureau', type 'attributeReport', cluster 'aqaraOpple', data '{"247":{"data":[100,16,1,3,40,24,152,57,0,0,0,0,149,57,0,0,0,0,150,57,122,109,20,69,151,57,0,0,0,0,5,33,1,0,154,32,0,11,32,0,8,33,23,1,9,33,0,5],"type":"Buffer"}}' from endpoint 1 with groupID 0
Info 2024-01-02 11:18:51MQTT publish: topic 'zigbee2mqtt_casa/Fil Pilote Bureau', payload '{"consumption":0,"current":0,"device_temperature":24,"energy":0,"linkquality":131,"power":0,"power_outage_count":0,"power_outage_memory":null,"state":"ON","switch_type":null,"update":{"installed_version":23,"latest_version":23,"state":"idle"},"update_available":null,"voltage":237.48}'
Info 2024-01-02 11:19:49MQTT publish: topic 'zigbee2mqtt_casa/Fil Pilote Bureau', payload '{"consumption":0,"current":0,"device_temperature":24,"energy":0,"linkquality":131,"power":0,"power_outage_count":0,"power_outage_memory":null,"state":"ON","switch_type":null,"update":{"installed_version":23,"latest_version":23,"state":"idle"},"update_available":null,"voltage":237.48}'
Debug 2024-01-02 11:23:02Received Zigbee message from 'Fil Pilote Bureau', type 'attributeReport', cluster 'genDeviceTempCfg', data '{"currentTemperature":25}' from endpoint 1 with groupID 0
Info 2024-01-02 11:23:02MQTT publish: topic 'zigbee2mqtt_casa/Fil Pilote Bureau', payload '{"consumption":0,"current":0,"device_temperature":25,"energy":0,"linkquality":127,"power":0,"power_outage_count":0,"power_outage_memory":null,"state":"ON","switch_type":null,"update":{"installed_version":23,"latest_version":23,"state":"idle"},"update_available":null,"voltage":237.48}'
Debug 2024-01-02 11:24:43Received Zigbee message from 'Fil Pilote Bureau', type 'attributeReport', cluster 'aqaraOpple', data '{"247":{"data":[100,16,1,3,40,25,152,57,0,0,0,0,149,57,0,0,0,0,150,57,0,89,20,69,151,57,0,0,0,0,5,33,1,0,154,32,0,11,32,0,8,33,23,1,9,33,0,5],"type":"Buffer"}}' from endpoint 1 with groupID 0
Info 2024-01-02 11:24:43MQTT publish: topic 'zigbee2mqtt_casa/Fil Pilote Bureau', payload '{"consumption":0,"current":0,"device_temperature":25,"energy":0,"linkquality":127,"power":0,"power_outage_count":0,"power_outage_memory":null,"state":"ON","switch_type":null,"update":{"installed_version":23,"latest_version":23,"state":"idle"},"update_available":null,"voltage":237.36}'
Debug 2024-01-02 11:28:01Received Zigbee message from 'Fil Pilote Bureau', type 'attributeReport', cluster 'genDeviceTempCfg', data '{"currentTemperature":26}' from endpoint 1 with groupID 0
Info 2024-01-02 11:28:01MQTT publish: topic 'zigbee2mqtt_casa/Fil Pilote Bureau', payload '{"consumption":0,"current":0,"device_temperature":26,"energy":0,"linkquality":131,"power":0,"power_outage_count":0,"power_outage_memory":null,"state":"ON","switch_type":null,"update":{"installed_version":23,"latest_version":23,"state":"idle"},"update_available":null,"voltage":237.36}'
Debug 2024-01-02 11:30:35Received Zigbee message from 'Fil Pilote Bureau', type 'attributeReport', cluster 'aqaraOpple', data '{"247":{"data":[100,16,1,3,40,25,152,57,0,0,0,0,149,57,0,0,0,0,150,57,247,231,20,69,151,57,0,0,0,0,5,33,1,0,154,32,0,11,32,0,8,33,23,1,9,33,0,5],"type":"Buffer"}}' from endpoint 1 with groupID 0
Info 2024-01-02 11:30:35MQTT publish: topic 'zigbee2mqtt_casa/Fil Pilote Bureau', payload '{"consumption":0,"current":0,"device_temperature":25,"energy":0,"linkquality":123,"power":0,"power_outage_count":0,"power_outage_memory":null,"state":"ON","switch_type":null,"update":{"installed_version":23,"latest_version":23,"state":"idle"},"update_available":null,"voltage":238.25}'
Debug 2024-01-02 11:33:00Received Zigbee message from 'Fil Pilote Bureau', type 'attributeReport', cluster 'genDeviceTempCfg', data '{"currentTemperature":25}' from endpoint 1 with groupID 0
Info 2024-01-02 11:33:01MQTT publish: topic 'zigbee2mqtt_casa/Fil Pilote Bureau', payload '{"consumption":0,"current":0,"device_temperature":25,"energy":0,"linkquality":131,"power":0,"power_outage_count":0,"power_outage_memory":null,"state":"ON","switch_type":null,"update":{"installed_version":23,"latest_version":23,"state":"idle"},"update_available":null,"voltage":238.25}'
```

procaktomasz, 2024-05-02:
I have the same problem, after re-pairing, the energy is shown for a while, after some time it only shows zero.
Interestingly, if I use the console, the value is available.
I use Home Assistant SkyConnect to connect.

Zigbee2MQTT version: 1.37.0-1 

[image]
[image]

vlebourl, 2024-05-02:
in my case, I can't click the read nor write button in the dev console... Dunno what I did wrong...
Plus haElectricalMeasurement isn't listed in the clusters in the "Clusters" tab of the device...

vlebourl, 2024-10-30:
Still no energy shown...

vlebourl, 2025-01-01:
Still an issue

201505409, 2025-01-09:
I also have this problem. Just stopped all of a sudden.
[image]

hexer4u, 2025-03-05:
Still an issue for me. I also see a correct value for activePower in haElectricalMeasurement in the console.
Maybe there is a way to regularly poll instead of report?

vlebourl, 2025-05-05:
still an issue, help to resolve it would be much appreciated :)

vlebourl, 2025-07-06:
still an issue.

closed 2025-09-14 by github-actions[bot]