Koenkk/zigbee2mqtt #20503: ZNQBKG25LM not pssing any action events

utegental, 2024-01-02:
### What happened?

At versions `1.33.*` of zigbee2mqtt,  I created converter for ZNQBKG25LM and it worked perfectly, but at version `1.34.0` it became broken, so I reverted to `1.33.2-1`. The issue was that any clicks on buttons not producing any actions.
Now I'm trying to adapt it. As I see, someone diв added support, but made a mistake with `buttonLookup` in `xiaomi_multistate_action` of `fromZigbee.ts`.
I've adapted converter to latest changes - some files was relocated, so I just need to change value in `require`, but for some reasons, that now not giving me a fix for events now.

### What did you expect to happen?

single_left, single_right and other actions to happen. when I'm clicking buttons.

### How to reproduce it (minimal and precise)

Click any button and see that nothing is happening

### Zigbee2MQTT version

1.35.0

### Adapter firmware version

7.3.1.0 build 176

### Adapter

Sonoff ZBDongle-E

utegental, 2024-01-02:
Original converter, which is working on versions prior `1.33.2-1`
`const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const preventReset = async (type, data, device) => {
    if (
        // options.allow_reset ||
        type !== 'message' ||
        data.type !== 'attributeReport' ||
        data.cluster !== 'genBasic' ||
        !data.data[0xfff0] ||
        // eg: [0xaa, 0x10, 0x05, 0x41, 0x87, 0x01, 0x01, 0x10, 0x00]
        !data.data[0xFFF0].slice(0, 5).equals(Buffer.from([0xaa, 0x10, 0x05, 0x41, 0x87]))
    ) {
        return;
    }
    const options = {manufacturerCode: 0x115f};
    const payload = {[0xfff0]: {
        value: [0xaa, 0x10, 0x05, 0x41, 0x47, 0x01, 0x01, 0x10, 0x01],
        type: 0x41,
    }};
    await device.getEndpoint(1).write('genBasic', payload, options);
};

const transactionStore = {};

function hasAlreadyProcessedMessage(msg, model, ID=null, key=null) {
    if (model.meta && model.meta.publishDuplicateTransaction) return false;
    const currentID = ID !== null ? ID : msg.meta.zclTransactionSequenceNumber;
    key = key || msg.device.ieeeAddr;
    if (transactionStore[key] === currentID) return true;
    transactionStore[key] = currentID;
    return false;
}

const fzlocalConverters = {
    xiaomi_multistate_action: {
        cluster: 'genMultistateInput',
        type: ['attributeReport'],
        convert: (model, msg, publish, options, meta) => {
            if (hasAlreadyProcessedMessage(msg, model)) return;
            let actionLookup = {0: 'hold', 1: 'single', 2: 'double', 3: 'triple', 255: 'release'};
            if (model.model === 'WXKG12LM') {
                actionLookup = {...actionLookup, 16: 'hold', 17: 'release', 18: 'shake'};
            }

            let buttonLookup = null;
            if (['WXKG02LM_rev2', 'WXKG07LM', 'WXKG15LM', 'WXKG17LM'].includes(model.model)) {
                buttonLookup = {1: 'left', 2: 'right', 3: 'both'};
            }
            if (['QBKG12LM', 'QBKG24LM'].includes(model.model)) buttonLookup = {5: 'left', 6: 'right', 7: 'both'};
            if (['QBKG39LM', 'QBKG41LM', 'WS-EUK02', 'WS-EUK04', 'QBKG20LM', 'QBKG31LM', 'ZNQBKG25LM'].includes(model.model)) {
                buttonLookup = {41: 'left', 42: 'right', 51: 'both'};
            }
            if (['QBKG25LM', 'QBKG26LM', 'QBKG34LM'].includes(model.model)) {
                buttonLookup = {
                    41: 'left', 42: 'center', 43: 'right',
                    51: 'left_center', 52: 'left_right', 53: 'center_right',
                    61: 'all',
                };
            }

            const action = actionLookup[msg.data['presentValue']];
            if (buttonLookup) {
                const button = buttonLookup[msg.endpoint.ID];
                if (button) {
                    return {action: `${action}_${button}`};
                }
            } else {
                return {action};
            }
        },
    },
}

const definition = {
    zigbeeModel: ['lumi.switch.acn030'],
    model: 'ZNQBKG25LM',
    vendor: 'Xiaomi',
    description: 'Aqara smart wall switch H1M EU (with neutral, double rocker)',
    fromZigbee: [fz.on_off, fz.xiaomi_power, fzlocalConverters.xiaomi_multistate_action, fz.aqara_opple],
    toZigbee: [tz.on_off, tz.xiaomi_power, tz.xiaomi_switch_operation_mode_opple, tz.xiaomi_flip_indicator_light],
    meta: {multiEndpoint: true},
    endpoint: (device) => {
            return {'left': 1, 'right': 2};
    },
    exposes: [e.switch().withEndpoint('left'), e.switch().withEndpoint('right'), e.power().withAccess(ea.STATE_GET), e.energy(),
            exposes.enum('operation_mode', ea.ALL, ['control_relay', 'decoupled']).withDescription('Decoupled mode for left button')
                .withEndpoint('left'),
            exposes.enum('operation_mode', ea.ALL, ['control_relay', 'decoupled']).withDescription('Decoupled mode for right button')
                .withEndpoint('right'),
            e.action(['single_left', 'double_left', 'single_right', 'double_right', 'single_both', 'double_both']),
            e.device_temperature().withAccess(ea.STATE), e.flip_indicator_light()],
    onEvent: preventReset,
    configure: async (device, coordinatorEndpoint, logger) => {
            const endpoint1 = device.getEndpoint(1);
            // set "event" mode
            await endpoint1.write('aqaraOpple', {'mode': 1}, {manufacturerCode: 0x115f, disableResponse: true});
        },
};

module.exports = definition;`

utegental, 2024-01-02:
@Koenkk, I wanna take this ticket and do it myself, but looks like it's restricted to change assignee field for me even to myself.

Rumlik, 2024-01-02:
Hi, I think we're all struggling with the same problem.
Ticket: https://github.com/Koenkk/zigbee2mqtt/issues/18483 and my https://github.com/Koenkk/zigbee2mqtt/issues/20114

So far, it doesn't look like anyone's going to be able to help :(

Koenkk, 2024-01-02:
Integrated your fix, thanks @utegental !

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

closed 2024-01-02 by Koenkk

Rumlik, 2024-01-04:
I tried the dev branch of Z2M and it works fine now. **It just can't recognize HOLD** on both right and left or both. @utegental Does this work for you?

utegental, 2024-01-04:
@Rumlik, I'll try it out tomorrow

utegental, 2024-01-05:
@Rumlik, Yeah clicks working well. Hold / release not working and is expected from what I see in code:
[image]

We probably need to find close model with already implemented logic, or debug and implement it with our own hands :-) 

P.S. I found why adapting converters to a newer version was unsuccessful - I had issues with zigbee mesh after unexpectedly smooth dongle migration to ZBDongle-E. Turns out, it just didn't showed any significant issues, but after enableing debug, I saw them. I had only option to repair all devices again and everything became stable.