Koenkk/zigbee2mqtt #20504: 1.35.0 crashes the homebridge-z2m plugin

fiftysomethingguy, 2024-01-02:
### What happened?

1.35.0 causes the homebridge-z2m V1.9.2 plugin by @itavero to crash. This also bring down the whole of homebrige if not installed as a child bridge. Zigbee2MQTT itself is working fine and sees all my devices that I can control from the frontend. However I guess 1.35 has changes the way it arranges the MQTT entries which is causing the plugin to fail when it reads them. Reverted to 1.34.0 and it works fine again.

### What did you expect to happen?

To work like 1.34.0 did

### How to reproduce it (minimal and precise)

Happens every time the plugin tries to start.

### Zigbee2MQTT version

1.35.0

### Adapter firmware version

20210708

### Adapter

SONOFF TI CC2652P

itavero, 2024-01-02:
See itavero/homebridge-z2m#794.

It seems that some devices (like the `Coordinator`)  now have `supported` set to `true` but the `definition` is set to `null` (and apparently me from a couple of years ago assumed that the definition would always be present if it was supported).
It shouldn't be too difficult to fix, but I do have to find some time to do so.

As a workaround (as reported in the issue mentioned above), you should be able to first add the coordinator as an excluded device to homebridge-z2m before upgrading, and then it should probably work without any issues.

fiftysomethingguy, 2024-01-02:
> See [itavero/homebridge-z2m#794](https://github.com/itavero/homebridge-z2m/issues/794).
>
> It seems that some devices (like the `Coordinator`) now have `supported` set to `true` but the `definition` [...]

Many thanks. Yes putting "Coordinator" in excluded devices allows me to run 1.35.0 without issues. Many thanks. Hopefully a permanent fix will prevent issues if new devices are added that are similarly labelled supported==true but definition==null given "generic" devices can now be enabled in 1.35.0+ from their zigbee cluster alone without a profile.

closed 2024-01-02 by Koenkk

guyeinan, 2024-01-03:
Hi sorry for the lack of knowledge but can you please elaborate on the workaround?