Koenkk/zigbee2mqtt #20505: 1.35.0 crashing after upgrade.

djrm05, 2024-01-02:
### What happened?

Unable to start zigbee2mqtt after upgrade the image to latest (1.35.0). I see the following error in the log. Changing the image to 1.34.0, app start correctly without issues.```
TypeError: Cannot read properties of undefined (reading 'light_onoff_brightness')
    at Object.<anonymous> (/app/lib/extension/publish.ts:20:18)
    at Module._compile (node:internal/modules/cjs/loader:1256:14)
    at Object.Module._extensions..js (node:internal/modules/cjs/loader:1310:10)
    at Module.load (node:internal/modules/cjs/loader:1119:32)
    at Function.Module._load (node:internal/modules/cjs/loader:960:12)
    at Module.require (node:internal/modules/cjs/loader:1143:19)
    at require (node:internal/modules/cjs/helpers:119:18)
    at Object.<anonymous> (/app/lib/controller.ts:14:1)
    at Module._compile (node:internal/modules/cjs/loader:1256:14)
    at Object.Module._extensions..js (node:internal/modules/cjs/loader:1310:10)
```

### Zigbee2MQTT version

1.35.0

### Adapter firmware version

6.7.10.0 build 423

### Adapter

EZSP v8

TheHawk1337, 2024-01-02:
I have the same issue. 1.34 works fine, 1.35 doesn't start.

I get these logs:

```
info  2024-01-02 16:50:22: Logging to console and directory: '/app/data/log/2024-01-02.16-50-22' filename: log.txt
info  2024-01-02 16:50:22: Starting Zigbee2MQTT version 1.35.0 (commit #b5a90e9)
info  2024-01-02 16:50:22: Starting zigbee-herdsman (0.30.0)
info  2024-01-02 16:50:52: Accepting joining not in blocklist device '0xa4c138f24689e023'
info  2024-01-02 16:50:52: Accepting joining not in blocklist device '0xa4c138f24689e023'
error 2024-01-02 16:50:53: Failed to start zigbee
error 2024-01-02 16:50:53: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
error 2024-01-02 16:50:53: Exiting...
error 2024-01-02 16:50:53: Error: Read 0x0017880108d038e1/11 lightingColorCtrl(["colorCapabilities"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufact>    at ZStackAdapter.sendZclFrameToEndpointInternal (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:415:23)
    at Queue.executeNext (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:32:32)
```

Koenkk, 2024-01-02:
@djrm05 how are you running z2m?

@TheHawk1337 could you provide the debug log?

See [this](https://www.zigbee2mqtt.io/guide/usage/debug.html) on how to enable debug logging.

djrm05, 2024-01-02:
@Koenkk  docker managed through portainer

TheHawk1337, 2024-01-02:
@Koenkk hope this helps. let me know if I can do something else

```
Using '/app/data' as data directory
Zigbee2MQTT:debug 2024-01-03 00:26:28: Loaded state from file /app/data/state.json
Zigbee2MQTT:info  2024-01-03 00:26:28: Logging to console and directory: '/app/data/log/2024-01-03.00-26-28' filename: log.txt
Zigbee2MQTT:debug 2024-01-03 00:26:28: Removing old log directory '/app/data/log/2024-01-02.16-50-22'
Zigbee2MQTT:info  2024-01-03 00:26:28: Starting Zigbee2MQTT version 1.35.0 (commit #b5a90e9)
Zigbee2MQTT:info  2024-01-03 00:26:28: Starting zigbee-herdsman (0.30.0)
Zigbee2MQTT:debug 2024-01-03 00:26:28: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null,"delay":null,"disableLED":false},"backupPath":"/app/data/coordinator_backup.json","databaseBackupPath":"/app/data/database.db.backup","databasePath":"/app/data/database.db","network":{"channelList":[11],"extendedPanID":[221,221,221,221,221,221,221,221],"networkKey":"HIDDEN","panID":6755},"serialPort":{"path":"/dev/ttyUSB0"}}'
Zigbee2MQTT:error 2024-01-03 00:27:00: Failed to start zigbee
Zigbee2MQTT:error 2024-01-03 00:27:00: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
Zigbee2MQTT:error 2024-01-03 00:27:00: Exiting...
Zigbee2MQTT:error 2024-01-03 00:27:00: Error: Read 0x0017880108d038e1/11 lightingColorCtrl(["colorCapabilities"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'No network route' (205))
    at ZStackAdapter.sendZclFrameToEndpointInternal (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:415:23)
    at Queue.executeNext (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:32:32)
```

Koenkk, 2024-01-03:
@djrm05 are you overriding zigbee-herdsman-converters through a volume somehow?

@TheHawk1337 pushed a possible fix, please check with the latest-dev branch.

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

TheHawk1337, 2024-01-03:
@Koenkk ok nice, will test asap (later tonight or tomorrow morning)!

djrm05, 2024-01-03:
@Koenkk I mapped  the converters folder to a volume few weeks back in portainer, but then I removed the volume and thought this was OK. However after your comment, I just recreated the container and it started succesfully, so obviously something unexpected happened when I created and removed the volume for the previous container.

So thanks! for me this was fixed

TheHawk1337, 2024-01-03:
@Koenkk No difference I think:
```

Using '/app/data' as data directory

Zigbee2MQTT:debug 2024-01-03 17:10:43: Loaded state from file /app/data/state.json

Zigbee2MQTT:info  2024-01-03 17:10:43: Logging to console and directory: '/app/data/log/2024-01-03.17-10-43' filename: log.txt

Zigbee2MQTT:debug 2024-01-03 17:10:43: Removing old log directory '/app/data/log/2024-01-02.17-04-29'

Zigbee2MQTT:info  2024-01-03 17:10:43: Starting Zigbee2MQTT version 1.35.0-dev (commit #e9aee4c)

Zigbee2MQTT:info  2024-01-03 17:10:43: Starting zigbee-herdsman (0.30.0)

Zigbee2MQTT:debug 2024-01-03 17:10:43: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null,"delay":null,"disableLED":false},"backupPath":"/app/data/coordinator_backup.json","databaseBackupPath":"/app/data/database.db.backup","databasePath":"/app/data/database.db","network":{"channelList":[11],"extendedPanID":[221,221,221,221,221,221,221,221],"networkKey":"HIDDEN","panID":6755},"serialPort":{"path":"/dev/ttyUSB0"}}'

Zigbee2MQTT:error 2024-01-03 17:11:14: Failed to start zigbee

Zigbee2MQTT:error 2024-01-03 17:11:14: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions

Zigbee2MQTT:error 2024-01-03 17:11:14: Exiting...

Zigbee2MQTT:error 2024-01-03 17:11:14: Error: Read 0x0017880108d038e1/11 lightingColorCtrl(["colorCapabilities"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'No network route' (205))

    at ZStackAdapter.sendZclFrameToEndpointInternal (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:415:23)

    at Queue.executeNext (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:32:32)
```

However, I don't think you should spend too much time on this, it appears I had some corruption on my system (portainer also had some issues) which might explain why zigbee2mqtt behaves like it does (or not, you probably know best).

Koenkk, 2024-01-03:
@TheHawk1337 could you provide the herdsman debug log?

See [this](https://www.zigbee2mqtt.io/guide/usage/debug.html) on how to enable the herdsman debug logging. Note that this is only logged to STDOUT and not to log files.

TheHawk1337, 2024-01-04:
@Koenkk See the attachments, bit too long to paste here.
[log_1_35_0.txt](https://github.com/Koenkk/zigbee2mqtt/files/13824677/log_1_35_0.txt)
[log_latest_dev.txt](https://github.com/Koenkk/zigbee2mqtt/files/13824678/log_latest_dev.txt)
[log_1_34_0.txt](https://github.com/Koenkk/zigbee2mqtt/files/13824679/log_1_34_0.txt)

Can move this to DM if you'd prefer, not sure if this clutters the thread here. Inexperienced at this, just want to help out :)

Koenkk, 2024-01-04:
I found the issue! Fixed it

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

TheHawk1337, 2024-01-05:
@Koenkk it still fails to start, see log:

[log_latest_dev_bugfix.txt](https://github.com/Koenkk/zigbee2mqtt/files/13841090/log_latest_dev_bugfix.txt)

Koenkk, 2024-01-05:
You are not running the latest dev branch, make sure to update to the latest.

closed 2024-01-06 by Koenkk

Alexisdelyon, 2024-01-07:
@Koenkk Same issue here.

**work fine with 1-33-1:**
debug 2024-01-07 16:57:22: Loaded state from file /app/data/state.json
info  2024-01-07 16:57:22: Logging to console and directory: '/app/data/log/2024-01-07.16-57-22' filename: log.txt
debug 2024-01-07 16:57:22: Removing old log directory '/app/data/log/2024-01-07.13-07-16'
info  2024-01-07 16:57:22: Starting Zigbee2MQTT version 1.33.2 (commit #9996c93)
info  2024-01-07 16:57:22: Starting zigbee-herdsman (0.21.0)
debug 2024-01-07 16:57:22: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null,"delay":null,"disableLED":false},"backupPath":"/app/data/coordinator_backup.json","databaseBackupPath":"/app/data/database.db.backup","databasePath":"/app/data/database.db","network":{"channelList":[11],"extendedPanID":[189,194,134,52,64,247,58,30],"networkKey":"HIDDEN","panID":30606},"serialPort":{"adapter":"ezsp","path":"/dev/ttyACM0"}}'
info  2024-01-07 16:57:28: zigbee-herdsman started (resumed)
info  2024-01-07 16:57:28: Coordinator firmware version: '{"meta":{"maintrel":"3 ","majorrel":"6","minorrel":"10","product":8,"revision":"6.10.3.0 build 297"},"type":"EZSP v8"}'
debug 2024-01-07 16:57:28: Zigbee network parameters: {"channel":11,"extendedPanID":189,"panID":30606}
info  2024-01-07 16:57:28: Currently 5 devices are joined:
info  2024-01-07 16:57:28: Contact Porte (0x00124b00292c5f3a): SNZB-04 - SONOFF Contact sensor (EndDevice)
info  2024-01-07 16:57:28: Lumiere Porche (0x9035eafffe0febda): ZBMINIL2 - SONOFF Zigbee smart switch (no neutral) (EndDevice)
info  2024-01-07 16:57:28: Temperature1 (0x3c2ef5fffebc2c6d): SNZB-02D - SONOFF Temperature and humidity sensor with screen (EndDevice)
info  2024-01-07 16:57:28: Temperature2 (0x3c2ef5fffeb9ae2c): SNZB-02D - SONOFF Temperature and humidity sensor with screen (EndDevice)
info  2024-01-07 16:57:28: 0x0c4314fffe086b50 (0x0c4314fffe086b50): E1603/E1702/E1708 - IKEA TRADFRI control outlet (Router)
warn  2024-01-07 16:57:28: `permit_join` set to  `true` in configuration.yaml.
warn  2024-01-07 16:57:28: Allowing new devices to join.
warn  2024-01-07 16:57:28: Set `permit_join` to `false` once you joined all devices.
info  2024-01-07 16:57:28: Zigbee: allowing new devices to join.
info  2024-01-07 16:57:28: Connecting to MQTT server at mqtt://172.17.0.1:1883
debug 2024-01-07 16:57:28: Using MQTT anonymous login
info  2024-01-07 16:57:28: Connected to MQTT server
info  2024-01-07 16:57:28: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload '{"state":"online"}'
info  2024-01-07 16:57:28: Started frontend on port 0.0.0.0:8080
info  2024-01-07 16:57:28: MQTT publish: topic 'zigbee2mqtt/Contact Porte', payload '{"battery":99,"battery_low":false,"contact":true,"linkquality":176,"tamper":false,"voltage":2900}'
info  2024-01-07 16:57:28: MQTT publish: topic 'zigbee2mqtt/Lumiere Porche', payload '{"linkquality":104,"state":"OFF","update":{"installed_version":4110,"latest_version":4110,"state":"idle"}}'
info  2024-01-07 16:57:28: MQTT publish: topic 'zigbee2mqtt/Temperature1', payload '{"battery":100,"humidity":51.5,"linkquality":172,"temperature":18}'
info  2024-01-07 16:57:28: MQTT publish: topic 'zigbee2mqtt/Temperature2', payload '{"battery":100,"humidity":59.4,"linkquality":80,"temperature":19.4}'
info  2024-01-07 16:57:28: MQTT publish: topic 'zigbee2mqtt/0x0c4314fffe086b50', payload '{"linkquality":172,"state":"OFF","update":{"installed_version":587765297,"latest_version":587765297,"state":"idle"}}'
info  2024-01-07 16:57:28: Zigbee2MQTT started!
debug 2024-01-07 16:57:44: Saving state to file /app/data/state.json
info  2024-01-07 16:57:44: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload '{"state":"offline"}'
info  2024-01-07 16:57:44: Disconnecting from MQTT server
info  2024-01-07 16:57:44: Stopping zigbee-herdsman...
info  2024-01-07 16:57:44: Stopped zigbee-herdsman
info  2024-01-07 16:57:44: Stopped Zigbee2MQTT

**But don't start with 1-34 or 1-35:**
debug 2024-01-07 17:00:50: Loaded state from file /app/data/state.json
info  2024-01-07 17:00:50: Logging to console and directory: '/app/data/log/2024-01-07.17-00-50' filename: log.txt
info  2024-01-07 17:00:50: Starting Zigbee2MQTT version 1.35.1-dev (commit #ef84ff8)
info  2024-01-07 17:00:50: Starting zigbee-herdsman (0.31.0)
debug 2024-01-07 17:00:50: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null,"delay":null,"disableLED":false},"backupPath":"/app/data/coordinator_backup.json","databaseBackupPath":"/app/data/database.db.backup","databasePath":"/app/data/database.db","network":{"channelList":[11],"extendedPanID":[189,194,134,52,64,247,58,30],"networkKey":"HIDDEN","panID":30606},"serialPort":{"adapter":"ezsp","path":"/dev/ttyACM0"}}'
error 2024-01-07 17:01:10: Error while starting zigbee-herdsman
error 2024-01-07 17:01:10: Failed to start zigbee
error 2024-01-07 17:01:10: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
error 2024-01-07 17:01:10: Exiting...
error 2024-01-07 17:01:10: Error: Failure to connect
    at Ezsp.connect (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:279:19)
    at Driver.startup (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/driver.ts:139:9)
    at EZSPAdapter.start (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/adapter/ezspAdapter.ts:172:16)
    at Controller.start (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:132:29)
    at Zigbee.start (/app/lib/zigbee.ts:60:27)
    at Controller.start (/app/lib/controller.ts:98:27)
    at start (/app/index.js:107:5)

Koenkk, 2024-01-08:
Could you provide the herdsman debug log of this?

See [this](https://www.zigbee2mqtt.io/guide/usage/debug.html) on how to enable the herdsman debug logging. Note that this is only logged to STDOUT and not to log files.

Alexisdelyon, 2024-01-08:
> Could you provide the herdsman debug log of this?
> 
> See [this](https://www.zigbee2mqtt.io/guide/usage/debug.html) on how to enable the herdsman debug logging. Note that this is only logged to STDOUT and not to log files.

Hi Koenkk,
Thank you for your help.
I understood where the error was coming from: I didn't set the privileges high enough for the container Docker.
All is working fine. Thank you very much.