Koenkk/zigbee2mqtt #20512: Exception while calling fromZigbee converter: Cannot set properties of undefined (setting 'power')

merlinpimpim, 2024-01-02:
### What happened?

Since Z2M 1.35 I sometimes get these logs.
I don't know which of my 25 devices is causing this problem.

> Zigbee2MQTT:error 2024-02-01 11:13:02: Exception while calling fromZigbee converter: Cannot set properties of undefined (setting 'power')}
Zigbee2MQTT:error 2024-02-01 11:13:03: Exception while calling fromZigbee converter: Cannot set properties of undefined (setting 'power')}
Zigbee2MQTT:error 2024-02-01 11:13:03: Exception while calling fromZigbee converter: Cannot set properties of undefined (setting 'power')}
Zigbee2MQTT:error 2024-02-01 11:13:03: Exception while calling fromZigbee converter: Cannot set properties of undefined (setting 'power')}
Zigbee2MQTT:error 2024-02-01 11:13:03: Exception while calling fromZigbee converter: Cannot set properties of undefined (setting 'power')}
Zigbee2MQTT:error 2024-02-01 11:13:03: Exception while calling fromZigbee converter: Cannot set properties of undefined (setting 'power')}
Zigbee2MQTT:error 2024-02-01 11:13:04: Exception while calling fromZigbee converter: Cannot set properties of undefined (setting 'power')}
Zigbee2MQTT:error 2024-02-01 11:13:04: Exception while calling fromZigbee converter: Cannot set properties of undefined (setting 'power')}
Zigbee2MQTT:error 2024-02-01 11:13:04: Exception while calling fromZigbee converter: Cannot set properties of undefined (setting 'power')}
Zigbee2MQTT:error 2024-02-01 11:13:04: Exception while calling fromZigbee converter: Cannot set properties of undefined (setting 'power')}
Zigbee2MQTT:error 2024-02-01 11:13:04: Exception while calling fromZigbee converter: Cannot set properties of undefined (setting 'power')}
Zigbee2MQTT:error 2024-02-01 11:13:05: Exception while calling fromZigbee converter: Cannot set properties of undefined (setting 'power')}
Zigbee2MQTT:error 2024-02-01 11:13:05: Exception while calling fromZigbee converter: Cannot set properties of undefined (setting 'power')}
Zigbee2MQTT:error 2024-02-01 11:13:05: Exception while calling fromZigbee converter: Cannot set properties of undefined (setting 

> 'power')}

### What did you expect to happen?

Understand what the problem is and stop getting these error logs.

### How to reproduce it (minimal and precise)

Nothing special.
Just be patient and read the logs

### Zigbee2MQTT version

1.35.0

### Adapter firmware version

 7.3.1.0 build 0

### Adapter

Skyconnect multi pan

PSMGoossens, 2024-01-02:
@merlinpimpim Enable debug logging. Then you get something like this:
`error 2024-01-02 19:12:33: Exception while calling fromZigbee converter: Cannot set properties of undefined (setting 'power')}
debug 2024-01-02 19:12:33: TypeError: Cannot set properties of undefined (setting 'power')
    at Object.convert (/app/node_modules/zigbee-herdsman-converters/src/devices/tuya.ts:477:29)
    at processTicksAndRejections (node:internal/process/task_queues:95:5)
    at Receive.onDeviceMessage (/app/lib/extension/receive.ts:147:35)
info  2024-01-02 19:12:33: MQTT publish: topic 'zigbee2mqtt/Stekker PC', payload '{"child_lock":"UNLOCK","current":0,"device":{"applicationVersion":192,"dateCode":"","friendlyName":"Stekker PC","hardwareVersion":1,"ieeeAddr":"0xa4c138d359a38661","manufacturerID":4417,"manufacturerName":"_TZ3000_zloso4jk","model":"TS011F_plug_1","networkAddress":55101,"powerSource":"Mains (single phase)","stackVersion":0,"type":"Router","zclVersion":3},"energy":89.75,"indicator_mode":"off","last_seen":"2024-01-02T18:12:33.983Z","linkquality":178,"power":0,"power_outage_memory":"restore","state":"OFF","update":{"installed_version":192,"latest_version":192,"state":"idle"},"update_available":false,"voltage":235}'
debug 2024-01-02 19:12:34: Received Zigbee message from 'Stekker PC', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":0,"rmsCurrent":0,"rmsVoltage":235}' from endpoint 1 with groupID 0
error 2024-01-02 19:12:34: Exception while calling fromZigbee converter: Cannot set properties of undefined (setting 'power')}
debug 2024-01-02 19:12:34: TypeError: Cannot set properties of undefined (setting 'power')
    at Object.convert (/app/node_modules/zigbee-herdsman-converters/src/devices/tuya.ts:477:29)
    at processTicksAndRejections (node:internal/process/task_queues:95:5)
    at Receive.onDeviceMessage (/app/lib/extension/receive.ts:147:35)
`

Koenkk, 2024-01-02:
fixed in the dev branch

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

closed 2024-07-31 by github-actions[bot]