Koenkk/zigbee2mqtt #20515: Power on behavoiur "UNSUPPORTED_ATTRIBUTE" since last Z2M update for KS-SM001 devices

ALaDoffe, 2024-01-02:
### What happened?

Hello,

Since the last Z2M, i got an error for my devices KS-SM001.
https://www.zigbee2mqtt.io/devices/KS-SM001.html

Failed to configure '0x0xxxxxxxxxxxxx', attempt 1 (Error: Read x0xxxxxxxxxxxxx/11 genOnOff(["startUpOnOff"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')

It  seems to be related to power on behaviour feature that is not supported for this device.

How to disable this feature ?
Thx

### Zigbee2MQTT version

1.35.0-dev commit: 84387b27

### Adapter firmware version

20230507

### Adapter

CC2652P

Cossid, 2024-01-02:
Tried removing my incomplete custom external component handler to see how the `Generate definition` feature would do.

Getting a similar `UNSUPPORTED_ATTRIBUTE` when removing my custom external component handler, which causes the entire process to crash.

```
lightingColorCtrl(["colorCapabilities"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')
```

Koenkk, 2024-01-02:
Fixed!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

closed 2024-01-02 by Koenkk