Koenkk/zigbee2mqtt #20516: Multiple TRADFRI Signal Repeater E1746 dropping connection all at once

Kugelfang666, 2024-01-02:
### What happened?

I‘m operating a large ZigBee network and use TRADFRI Signal Repeater model E1746 to spread the network. In total I‘m using 4 repeater across my home. Since a few weeks these repeaters drop from my network and become unavailable randomly. Interestingly they all drop a the same time and powercyling them bring them back immediately. Sometimes i have no issue for a few days, sometimes they drop 2-3h after a power cycle. 

I was not able to diagnose the root cause for this behavior. 

This setup was working for ca 1 year without any issue, problems started around 4-6 weeks ago

### Zigbee2MQTT version

1.35.0

### Adapter firmware version

20230507

### Adapter

TubesZB Ethernet based coordinator

### Debug log

2024-01-02 16:47:55 Active device 'Repeater - IKEA - Wohnzimmer' was last seen '289.93' minutes ago

Warning 2024-01-02 16:52:23Failed to ping 'Repeater - IKEA - Wohnzimmer' (attempt 1/1, Read XXXXXREDACTEDXXXXX/1 genBasic(["zclVersion"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 17560 - 1 - 156 - 0 - 1 after 10000ms))

chris-1243, 2024-01-02:
Have you changed something on your network ? Firmware update ?

Regarding the coordinator, firmware 20230507 is unfortunately known to cause some problems [#478](https://github.com/Koenkk/Z-Stack-firmware/discussions/478), [#483](https://github.com/Koenkk/Z-Stack-firmware/discussions/483)
Lots of work is done by Koenkk to find a patch on this specific topic but it will take some time.
It may not be the cause at all but I suggest to have a look on this topic.
I personnaly move from 20230507 (which was working pretty fine) to 20220922 after some test and I have a really stable network without ping warnings nor errors.

Kugelfang666, 2024-01-03:
oh i See, thanks for the tip, will check the other issues out and maybe give one of the experimental FWs a try

closed 2024-08-01 by github-actions[bot]