Koenkk/zigbee2mqtt #20519: 1.35.0 Zigbee2MQTT stops responding after update

larry404, 2024-01-02:
### What happened?

On New Years Eve with 30 people in my home all my zigbee devices stopped responding. Not the best time to put out an upgrade. When I attempt to restart the add-on it will work for about 10 minutes or so and then it stops. I have never had any major issues with this addon in the past.

### What did you expect to happen?

Not push out upgrades during major holidays. Have some sort of beta program to make sure the upgrade will work?

### How to reproduce it (minimal and precise)

Restart the add-on and about 10 minutes later it stops responding

### Zigbee2MQTT version

1.35.0

### Adapter firmware version

"meta":{"maintrel":1,"majorrel":2,"minorrel":7,"product":1,"revision":20210708,"transportrev":2

### Adapter

Sonoff Zigbee USB 3.0 dongle

### Debug log

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/13812799/log.txt)

julianfs, 2024-01-02:
Are you using Home Assistant? If so you can turn off auto-update.

larry404, 2024-01-02:
Yes, I am using Home Assistant. In retro-spec that would have been a good idea. However, since I never had issues before I never gave that a second thought. Thanks.

BandBxx, 2024-01-03:
+1,  same problem. After update to v. 1.35  my z2m  going to freeze  after couple minutes of work.  Very bad unstable update!

Hadatko, 2024-01-03:
+1
and in HA community i see more users with same issue

Koenkk, 2024-01-03:
Pushed a fix, please try with the latest-dev.

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

If not, provide the debug log again.

See [this](https://www.zigbee2mqtt.io/guide/usage/debug.html) on how to enable debug logging.

sdotter, 2024-01-03:
@Koenkk let me know if i can test something 👍👌

Koenkk, 2024-01-03:
Could you provide the herdsman debug log of the crash?

See [this](https://www.zigbee2mqtt.io/guide/usage/debug.html) on how to enable the herdsman debug logging. Note that this is only logged to STDOUT and not to log files.

sdotter, 2024-01-03:
I've added  zigbee_herdsman_debug: true to my config but don't see anything new in docker logs... I think I'm doing something wrong to view herdsman logs?

sdotter, 2024-01-03:
Docker logs are stdout also right?

sdotter, 2024-01-03:
`Error: {"address":40606,"clusterId":32817,"sequence":24} after 10000ms
    at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35)
    at listOnTimeout (node:internal/timers:569:17)
    at processTimers (node:internal/timers:512:7)`

merlinpimpim, 2024-01-03:
Version 1.35.0 is completely unreliable.
I have to restart Z2M every 30 minutes to get my 25 devices to work more or less correctly.
An urgent production patch needs to arrive soon.

sdotter, 2024-01-03:
I think we both experience same like issues...
But I think it will arrive soon haha look at the development that is going on for z2m... Its amazing piece of freeware man 😂👍🎉

MiAutomations, 2024-01-03:
I think that my problem is related also ...

[https://github.com/Koenkk/zigbee2mqtt/issues/20506](url)

Once after the update I'm unable to start the addon, but if I go back with a restore all is working good

sdotter, 2024-01-03:
#20506

Koenkk, 2024-01-03:
@sdotter did you add it in the correct place (HA addon config)?

sdotter, 2024-01-03:
yeah i think so:

/usr/share/hassio/homeassistant/zigbee2mqtt/configuration.yaml

`homeassistant:
  legacy_entity_attributes: true
mqtt:
  server: mqtt://192.168.1.230:1883
  user: root
  password: '12345'
  keepalive: 60
  reject_unauthorized: true
  version: 4
serial:
  port: /dev/ttyACM0
  adapter: ezsp
frontend:
  port: 8099
advanced:
  homeassistant_legacy_entity_attributes: false
  legacy_api: false
  legacy_availability_payload: false
  log_level: debug
  channel: 15
device_options:
  legacy: false
permit_join: false
zigbee_herdsman_debug: true
devices:
...`

sdotter, 2024-01-03:
For now i tested and downgraded to: 1.33.2
And everything seems to work a lot faster / more stable (z2m also crashed but lot less) and generating map was much faster

I'll update when i know more... 

Zigbee2MQTT versie
[1.33.2](https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.33.2) commit: [unknown](https://github.com/Koenkk/zigbee2mqtt/commit/unknown)
Coördinator type
EZSP v11
Coordinator revisie
7.2.2.0 build 190
Coordinator IEEE adres
0xe0798dfffef8a9a3
Versie frontend
0.6.142
Zigbee-herdsman-converters versie
15.106.0
Zigbee-herdsman versie
0.21.0

svarogmaliby, 2024-01-04:
Same problem, rolling back to version 1.33.2-1 completely solves it.
The problem with the add-on crash starts with version 1.34.0-1 and higher.

chris-1243, 2024-01-04:
@sdotter 

If your instance of zigbee2mqtt is on docker, add this line in your docker-compose.yaml in the environment section

DEBUG: zigbee-herdsman* 

Restart your container and try to reproduce the error. As soon as you were able to do it, execute: (sudo) docker compose logs zigbee2mqtt  > log.txt 2>&1.

For the addon, I can't help at all...

sdotter, 2024-01-04:
@chris-1243 yeah thanks ... I'm running addon. But I think I'm gonna clone repo and build it myself and run it standalone like so I can debug more. Thanks for thinking along!! 🙏🤛👍

sdotter, 2024-01-04:
> Same problem, rolling back to version 1.33.2-1 completely solves it. The problem with the add-on crash starts with version 1.34.0-1 and higher. ![ff76fd14-f0df-495f-9ef7-ac18d780ecba](https://private-user-images.githubusercontent.com/128695668/294172269-29e7d0a3-60bc-4237-8df3-dc5b53db5e62.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDQzNzI0NzMsIm5iZiI6MTcwNDM3MjE3MywicGF0aCI6Ii8xMjg2OTU2NjgvMjk0MTcyMjY5LTI5ZTdkMGEzLTYwYmMtNDIzNy04ZGYzLWRjNWI1M2RiNWU2Mi5qcGc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMTA0JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDEwNFQxMjQyNTNaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0zZDExNGU2MzIwNjc3OWQxZTFjOWMzMGZkODBmZTkyZjc3MmRjMDc4MGIzMjg4OWFkYTUxZDI3N2E2ZjJkNjZkJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.AT36PS4cD4IzJdMCk9cksGapNwRA-JFvb3qkndevq0w)

Yeah that also my experience

Koenkk, 2024-01-04:
@sdotter this looks like the z2m `configuration.yaml` file, it should be the addon config editor (via HA)

pidzi92, 2024-01-04:
Is there a way to rollback to older version in home assistant without need to reconfigure everything?

sdotter, 2024-01-06:
> @sdotter this looks like the z2m `configuration.yaml` file, it should be the addon config editor (via HA)

thanks man! ive found it 😂
haha... or just edit it as yaml and then add the debug herdsman property thing... 

[image]

But today i tried rcp multipan stuff but there just was an update for Silicon Labs Multiprotocol breaking stuff... so i flashed my stick again with "ncp-uart-sw_7.2.2.0_115200.gbl" and repaired all devices... (not that much work) and everything was working fast as f$#k again... i dont understand it really but yeah...
Learned a lot last two days.... NCP RCP (still dont know wich i should prefer?)

@Koenkk can you give some hints how to contribute (and get familiar first) with the z2m and the herdsman?

pidzi92, 2024-01-06:
> Is there a way to rollback to older version in home assistant without need to reconfigure everything?

Well, I've found a workaround that works for me. I just had to reset all problematic devices.

DO NOT REMOVE IT FROM HA OR Z2M.
Reset device.
It should rejoin to its network automatically and it should work, without any reconfiguration. All your automation and entities should stay the same.
If it does not rejoin, go into Z2M add-on and permit join.

sdotter, 2024-01-06:
> set device.
> It should rejoin to its network automatic

Nice man! thanks i will keep that in mind... yeah for me it started with an really problematic device, an awox device, started to screw things up.... dont use it! haha 😂👍

Koenkk, 2024-01-06:
@sdotter are you issues solved now? what part do you want to contribute to?

sdotter, 2024-01-06:
@Koenkk i dont know for sure...
Long story short, i had a lot of weird issues with my zigbee network and z2m 1.35... a lot less with 1.33.x
ive ended up trying multiprotocol matter and zigbee firmware but that stilll too early to be stable like... so i flashed exactly same  fw (ncp-uart-sw_7.3.1.0_115200.gbl) to my zbdonglee and now everything on 1.35 is a lot faster / z2m hasnt crashed since 11 hours ago...

Koenkk, 2024-01-06:
For developing the frontend the documentation can be found [here](https://github.com/nurikk/zigbee2mqtt-frontend?tab=readme-ov-file#develop)

larry404, 2024-01-06:
Hi for what it’s worth my zigbee network is working again as one week after it started having issues. I have made no changes to the zigbee config except to updating to the latest HA builds. All devices are working as normal. I’ll keep on monitoring and report back if any issues

flowcool, 2024-01-08:
Dear @Koenkk , I changed to dev branch yesterday and until now, no issue. So I assume it's good like this ;)

svarogmaliby, 2024-01-08:
Installed update 1.35.1-1, the addon continues to crash on this version.
[image]

Koenkk, 2024-01-08:
@svarogmaliby looks like you are using an invalid `debounce` value for one of the devices in your z2m `configuration.yaml`.

svarogmaliby, 2024-01-08:
> @svarogmaliby looks like you are using an invalid `debounce` value for one of the devices in your z2m `configuration.yaml`.

No values have been changed; on version 1.33.2-1 the addon works stably

Koenkk, 2024-01-09:
The checks seems to be more strict now (previously an invalid config was ignored)

closed 2024-08-07 by github-actions[bot]