Koenkk/zigbee2mqtt #20525: Xiaomi JY-GZ-01AQ smoke detector missing exposes property "Linkage alarm state" in all message payloads

SteveBrackenbury, 2024-01-02:
### What happened?

Z2M supported device Xiaomi JY-GZ-01AQ smoke detector **is missing attribute/property "Linkage alarm state"** in all message payloads, even though Z2M device documentation (https://www.zigbee2mqtt.io/devices/JY-GZ-01AQ.html) includes this attribute/property.

From the Z2M frontend Devices -> Xiaomi JY-GZ-01AQ smoke detector instance:
 
Device -> Exposes tab shows "Linkage alarm state" value as "N/A" 

Device -> State is missing the linkage_alarm_state attribute/property:
{
    "battery": 100,
    "buzzer_manual_alarm": false,
    "buzzer_manual_mute": false,
    "heartbeat_indicator": false,
    "last_seen": "2024-01-02T15:19:50-05:00",
    "linkage_alarm": "true",
    "linkquality": 127,
    "power_outage_count": 13,
    "smoke": false,
    "smoke_density": 0,
    "smoke_density_dbm": 0,
    "test": false,
    "voltage": 3043
}

### What did you expect to happen?

Device -> Exposes tab should show "Linkage alarm state" with a default value of "false" 

Device -> State should include the linkage_alarm_state attribute/property with a default value of "false":
{
    "battery": 100,
    "buzzer_manual_alarm": false,
    "buzzer_manual_mute": false,
    "heartbeat_indicator": false,
    "last_seen": "2024-01-02T15:19:50-05:00",
    "linkage_alarm": "true",
    **"linkage_alarm_state": "false"**,
    "linkquality": 127,
    "power_outage_count": 13,
    "smoke": false,
    "smoke_density": 0,
    "smoke_density_dbm": 0,
    "test": false,
    "voltage": 3043
} 

### How to reproduce it (minimal and precise)

Should be able to recreate by verifying Device Exposes and Device State in Z2M frontend for Aqara JY-GZ-01AQ Device.

Here are my platform, OS USB coordinator adapter, Zigbee device and associated software/firmware version details:

Hardware Platform:
Raspberry Pi 3 Model B Plus Rev 1.3

OS Version:
6.1.0-rpi7-rpi-v8 #1 SMP PREEMPT Debian 1:6.1.63-1+rpt1 (2023-11-24) aarch64 GNU/Linux

Adapter/Coordinator details:
Adapter Type: CC2652
By-id: /dev/serial/by-id/usb-Silicon_Labs_CP2102N_USB_to_UART_Bridge_Controller_a095bb8111c4ec119af079480af30733-if00-port0
(https://www.amazon.com/dp/B0C7GMVQKR)
Coordinator Type: zStack3x0 
Coordinator firmware revision: 20221226

### Device Model details: 
Manufacturer: Aqara
Model: **JY-GZ-01AQ**
Zigbee Model: **lumi.sensor_smoke.acn03**
Firmware version: **2019www.**

### Zigbee2MQTT version

 1.35.0 commit: 3d8f0fc3

### Adapter firmware version

 20221226

### Adapter

Adapter Type: CC2652 USB Silicon_Labs_CP2102N_USB_to_UART_Bridge_Controller(https://www.amazon.com/dp/B0C7GMVQKR) 

### Debug log

Debug 2024-01-02 12:48:53Received Zigbee message from 'smoke-detector-bsmnt-new-workshop', type 'attributeReport', cluster 'aqaraOpple', data '{"247":{"data":[1,33,225,11,3,40,25,4,33,168,67,5,33,14,0,6,36,1,0,0,0,0,8,33,17,1,10,33,0,0,12,32,1,102,32,3,103,32,1,104,33,168,0,160,33,0,0,161,32,0,162,32,0,163,32,0,164,32,0,165,32,1],"type":"Buffer"}}' from endpoint 1 with groupID 0
Debug 2024-01-02 12:48:53JY-GZ-01AQ: Processed buffer into data {"1":3041,"3":25,"4":17320,"5":14,"6":1,"8":273,"10":0,"12":1,"102":3,"103":1,"104":168,"160":0,"161":0,"162":0,"163":0,"164":0,"165":1}
**Debug 2024-01-02 12:48:53JY-GZ-01AQ: unknown key 6 with value 1
Debug 2024-01-02 12:48:53JY-GZ-01AQ: unknown key 104 with value 168**
Debug 2024-01-02 12:48:53JY-GZ-01AQ: Processed data into payload {"voltage":3041,"battery":100,"power_outage_count":13,"smoke":false,"smoke_density":0,"smoke_density_dbm":0,"test":false,"buzzer_manual_mute":false,"heartbeat_indicator":false,"linkage_alarm":true}
Debug 2024-01-02 12:48:53JY-GZ-01AQ: Processed data into payload {"voltage":3041,"battery":100,"power_outage_count":13,"smoke":false,"smoke_density":0,"smoke_density_dbm":0,"test":false,"buzzer_manual_mute":false,"heartbeat_indicator":false,"linkage_alarm":true}
Info 2024-01-02 12:48:53MQTT publish: topic 'zigbee2mqtt/smoke-detector-bsmnt-new-workshop', payload '{"battery":100,"buzzer_manual_alarm":false,"buzzer_manual_mute":false,"heartbeat_indicator":false,"last_seen":"2024-01-02T12:48:5305:00","linkage_alarm":true,"linkquality":105,"power_outage_count":13,"smoke":false,"smoke_density":0,"smoke_density_dbm":0,"test":false,"voltage":3041}'

SteveBrackenbury, 2024-01-03:
To rule out coordinator firmware version, I flashed my CC2652 Silicon Labs USB stick with the latest coordinator revision:  20230507

https://github.com/Koenkk/Z-Stack-firmware/releases/tag/Z-Stack_3.x.0_coordinator_20230507

After re-pairing all my XIAOMI smoke detectors, the issue persists.

Let me know if you would like to conduct any other tests or provide more details.

SteveBrackenbury, 2024-02-19:
@Koenkk I realize you are very busy.  Would you please take a quick look at this issue and advise on how I can progress it.

Currently I am stuck and need guidance on how to resolve.  I am happy to support any diagnostics or testing that is required.  

With the missing/non functional attribute/property "Linkage alarm state" these smoke detectors cannot be integrated programattically by triggering their alarms to sound in unison should any one of the smoke detectors but put into a real alarm state.

Thanks for your guidance and support!

...Steve

Otnow, 2024-02-21:
As far as I remember, the value of `linkage_alarm_state` cannot be obtained upon request, so it will take a specific value only when the device reports it on its own, and until then after pairing the value will be undefined (N/A).

@Koenkk, is it possible during the device pairing process to set a default value for an attribute whose value cannot be obtained upon request?

https://github.com/Koenkk/zigbee-herdsman-converters/blob/a570fccd89bee9a4b49bdea5134cfe37b8180079/src/devices/lumi.ts#L1789

Koenkk, 2024-02-21:
@Otnow yes, but we dont know the value right? Then we can better have unknown

Otnow, 2024-02-22:
@Koenkk, the default value is `false`, so it would be good to set it immediately when pairing the device, which is what the author of this issues asks for.

SteveBrackenbury, 2024-02-22:
Thanks @Otnow and @Koenkk  for the dialogue on this issue.

I was not aware that the linkage_alarm_state cannot be obtained upon request and that a specific value will only be set/available once the device reports it on its own.  I assumed the state was boolean and therefore should be false or true.  I had not considered that a state of "unknown" (NULL) is considered valid.  It appeared to me that linkage_alarm_state was broken/non-functional in zigbee2mqtt.

It seems @Koenkk is indicating he believes it better to leave the state of linkage_alarm_state in it's "unknown" (NULL) state rather than to presume it false and set it thus.  I can understand that perspective, however,  I'm not sure there is harm is setting the value to false in the absence of a value.  Having it set to false would be more intuitive to the end-user of the device.  If left at "unknown" it would be useful to have additional documentation indicating linkage_alarm_state will be unknown until reported by the device (which will presumably only happen when smoke is detected and the alarm is triggered by the device itself).

I will spend some time over the next few days and will test one of my smoke detectors (physical smoke test) to confirm whether or not I can cause the smoke detector to set linkage_alarm_state to true/false.

I will also test some HA automations to see if I can use the linkage_alarm_state (true) to turn the buzzer on the other smoke detectors. 

I'll get back to confirm the test results.

SteveBrackenbury, 2024-02-24:
@Koenkk & @Otnow 

I tested one of my Xiaomi JY-GZ-01AQ smoke detectors by creating "physical smoke" and forcing smoke detection which then resulted in the smoke detector going into an alarm state (smoke = "true" and the alarm buzzer sounding)  

While performing the "physical smoke" test I monitored the zigbee2mqtt smoke detector device "exposes" tab in the zigbee2mqtt web app.

As soon as the smoke detector detected smoke, the device's smoke property changed from "false" to "true" and, (with the linkage_alarm property set to "true"), the linkage_alarm_state property went from NULL to "true".

At the same time, the Home Assistant (HA) integration I had configured to detect a "linkage_alarm_state" of "true" on other smoke detectors and set the buzzer to "alarm", worked and other smoke detectors sounded their buzzer alarm.

Once the smoke detector I tested no longer detected smoke, (smoke = "false" and the alarm buzzer stopped sounding), the linkage_alarm_state property on the "physical smoke" tested smoke detector went from"true" to "false".

Now the linkage_alarm_state exists on that "physical smoke" tested smoke detector.  However, the linkage_alarm_state remains at NULL (unknown) on all other smoke detectors for which I have not yet conducted a "physical smoke" test.

**### Findings Summary:**

1) The linkage_alarm_state will always be NULL (unknown) until a REAL "physical smoke" test is conducted for the device.
2) The linkage_alarm_state NULL (unknown) state will move to "true" once a "physical smoke" test is conducted and will then move to "false" once the device no longer detects smoke.
3) HomeAssistant integration to manually trigger the "alarm" (buzzer) state on a Xiaomi JY-GZ-01AQ smoke detector when a linkage_alarm_state of "true" is detected on another smoke detector functions as desired. (Note: that the linkage_alarm property must be set to "true" on all of the smoke detectors in order for this functionality to work as desired).

**So the Xiaomi JY-GZ-01AQ smoke detector is working as desired in zigbee2mqtt and Home Assistant (HA).

However, personally, I would like to see the "linkage_alarm_state" set to "false" when the device is first added to zigbee2mqtt.  I think this would be an improvement. Clearly, the "linkage_alarm_state" is a proxy for the "smoke" and "smoke density" states, so if the smoke property is "false" it is safe to set the "linkage_alarm_state" property to "false" in the absence of being able to directly poll the device for the "linkage_alarm_state" property's state.

I find it confusing to have the "linkage_alarm_state" property missing from the "state" tab and to have a value of NULL (unknown) in the "exposes" tab of the zigbee2mqtt web app.  I also don't like that I need to conduct a "physical smoke" test on each and every one of my smoke detectors in order to force a "linkage_alarm_state" value.**

Please consider this enhancement request.  I am happy to support testing the enhancement and/or contributing to the documentation.

closed 2024-09-22 by github-actions[bot]