Koenkk/zigbee2mqtt #20526: A LOT of traffic and errors after 1.35 upgrade - "No network route (205)" and timeouts

nefh, 2024-01-02:
### What happened?

After upgrade to 1.35.0 there is really A LOT of traffic and errors on z2m and mqtt protocol. The UI error messages keeps spamming constantly. Eventually, it will reach 100% CPU thread utilization and automation stops responding. I've upgraded to newest firmware on CC1352P2, the issue persist. 

I've cut 5 minutes of logs on 1.34 and 1.35 for comparison, 1.35 is twice as long in the same time on info level. 

Here is just initial 10 lines containing 'error':

```
Line  47: Zigbee2MQTT:info  2024-01-02 23:49:30: MQTT publish: topic 'homeassistant/select/1221051039810110150109113116116_0x00124b001ca1b865/log_level/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"availability_mode":"all","command_template":"{\"options\": {\"advanced\": {\"log_level\": \"{{ value }}\" } } }","command_topic":"zigbee2mqtt/bridge/request/options","device":{"hw_version":"zStack3x0 20230507","identifiers":["zigbee2mqtt_bridge_0x00124b001ca1b865"],"manufacturer":"Zigbee2MQTT","model":"Bridge","name":"Zigbee2MQTT Bridge","sw_version":"1.35.0"},"entity_category":"config","json_attributes_topic":"zigbee2mqtt/bridge/info","name":"Log level","object_id":"zigbee2mqtt_bridge_log_level","options":["info","warn","error","debug"],"origin":{"name":"Zigbee2MQTT","sw":"1.35.0","url":"https://www.zigbee2mqtt.io"},"state_topic":"zigbee2mqtt/bridge/info","unique_id":"bridge_0x00124b001ca1b865_log_level_zigbee2mqtt","value_template":"{{ value_json.log_level | lower }}"}'
Line 310: Zigbee2MQTT:error 2024-01-02 23:49:40: Failed to configure 'ToaletaLampa', attempt 1 (Error: Bind 0x60a423fffe94f82a/1 genOnOff from '0x00124b001ca1b865/1' failed (AREQ - ZDO - bindRsp after 10000ms)
Line 314: Zigbee2MQTT:error 2024-01-02 23:49:54: Publish 'get' 'brightness' to 'WejscieLampa' failed: 'Error: Read 0x94deb8fffe5750d2/1 genLevelCtrl(["currentLevel"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'No network route' (205))'
Line 315: Zigbee2MQTT:info  2024-01-02 23:49:54: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'get' 'brightness' to 'WejscieLampa' failed: 'Error: Read 0x94deb8fffe5750d2/1 genLevelCtrl([\"currentLevel\"], {\"sendWhen\":\"immediate\",\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":true,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (Data request failed with error: 'No network route' (205))'","meta":{"friendly_name":"WejscieLampa"},"type":"zigbee_publish_error"}'
Line 316: Zigbee2MQTT:error 2024-01-02 23:49:54: Publish 'get' 'brightness' to 'KorytarzSufitLampa' failed: 'Error: Read 0x287681fffef82842/1 genLevelCtrl(["currentLevel"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'No network route' (205))'
Line 317: Zigbee2MQTT:info  2024-01-02 23:49:54: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'get' 'brightness' to 'KorytarzSufitLampa' failed: 'Error: Read 0x287681fffef82842/1 genLevelCtrl([\"currentLevel\"], {\"sendWhen\":\"immediate\",\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":true,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (Data request failed with error: 'No network route' (205))'","meta":{"friendly_name":"KorytarzSufitLampa"},"type":"zigbee_publish_error"}'
Line 318: Zigbee2MQTT:error 2024-01-02 23:49:55: Publish 'get' 'brightness' to 'MamySufitLampa' failed: 'Error: Read 0x003c84fffe2a74d3/1 genLevelCtrl(["currentLevel"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'No network route' (205))'
Line 319: Zigbee2MQTT:info  2024-01-02 23:49:55: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'get' 'brightness' to 'MamySufitLampa' failed: 'Error: Read 0x003c84fffe2a74d3/1 genLevelCtrl([\"currentLevel\"], {\"sendWhen\":\"immediate\",\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":true,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (Data request failed with error: 'No network route' (205))'","meta":{"friendly_name":"MamySufitLampa"},"type":"zigbee_publish_error"}'
Line 320: Zigbee2MQTT:error 2024-01-02 23:49:55: Publish 'get' 'illuminance_lux' to 'SwiatlaCzujnik' failed: 'Error: Read 0x04cf8cdf3c7d1fa4/1 msIlluminanceMeasurement(["measuredValue"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'No network route' (205))'
Line 321: Zigbee2MQTT:info  2024-01-02 23:49:55: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'get' 'illuminance_lux' to 'SwiatlaCzujnik' failed: 'Error: Read 0x04cf8cdf3c7d1fa4/1 msIlluminanceMeasurement([\"measuredValue\"], {\"sendWhen\":\"immediate\",\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":true,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (Data request failed with error: 'No network route' (205))'","meta":{"friendly_name":"SwiatlaCzujnik"},"type":"zigbee_publish_error"}'
```

### What did you expect to happen?

Work as usual.

### How to reproduce it (minimal and precise)

Run 1.35.0 on CC1352P2.

### Zigbee2MQTT version

1.35.0 commit: 3d8f0fc3

### Adapter firmware version

20230507

### Adapter

CC1352P2

### Debug log

[zigbee2mqtt_daemon_1.34.log](https://github.com/Koenkk/zigbee2mqtt/files/13814598/zigbee2mqtt_daemon_1.34.log)
[zigbee2mqtt_daemon_1.35.log](https://github.com/Koenkk/zigbee2mqtt/files/13814599/zigbee2mqtt_daemon_1.35.log)

Koenkk, 2024-01-03:
Pushed a fix, let me know if it works!

If not, provide the debug log.

See [this](https://www.zigbee2mqtt.io/guide/usage/debug.html) on how to enable debug logging.

FaBRiK74, 2024-01-03:
Same here. I just switched to Z2M Edge 10 minutes ago to solve the problem but the problem still exists. E.g. the covers are not responding, or just after 2-5 minutes. And I get a lot of red boxes in Z2M like:
Failed to configure 'xxx', attempt 1 (Error: Read 0xa4c1xxx genBasic(["manufacturerName","zclVersion","appVersion","modelId","powerSource",65534], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 60110 - 1 - 241 - 0 - 1 after 10000ms) at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35) at listOnTimeout (node:internal/timers:569:17) at processTimers (node:internal/timers:512:7))

nefh, 2024-01-03:
> If not, provide the debug log.

@Koenkk Thanks for the feedbacku, unfortunately the fix did not help. Here is 5min debug log:
[zigbee2mqtt_daemon_dev.log](https://github.com/Koenkk/zigbee2mqtt/files/13823906/zigbee2mqtt_daemon_dev.log)

FaBRiK74, 2024-01-03:
Here are some other error logs:
Error 2024-01-03 23:40:40Publish 'set' 'position' to 'Dolgozószoba ablakredőny 2' failed: 'Error: Command 0xa4c1382e6e535346/1 closuresWindowCovering.goToLiftPercentage({"percentageliftvalue":36}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'No network route' (205))'
Error 2024-01-03 23:40:43Publish 'set' 'position' to 'Dolgozószoba ablakredőny 1' failed: 'Error: Command 0xa4c1383338e8b2cd/1 closuresWindowCovering.goToLiftPercentage({"percentageliftvalue":32}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'No network route' (205))'

trizmark, 2024-01-04:
Same issues with 1.34 - network completely down 😭 
Using Electrollama's zzh as a coordinator with fw 20230507

Sample log entries
```
Zigbee2MQTT:error 2024-01-04 14:12:30: Failed to configure 'Ikea_Outlet_1', attempt 1 (Error: ConfigureReporting 0xcc86ecfffe8dc049/1 genOnOff([{"attribute":"onOff","minimumReportInterval":0,"maximumReportInterval":3600,"reportableChange":0}], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'undefined' (25))
    at ZStackAdapter.sendZclFrameToEndpointInternal (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:415:23)
    at Queue.executeNext (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:32:32))
Zigbee2MQTT:warn  2024-01-04 14:12:32: Failed to ping 'Ikea_E27_WS_7' (attempt 1/1, Read 0x84fd27fffeb3a94d/1 genBasic(["zclVersion"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'No network route' (205)))
Zigbee2MQTT:warn  2024-01-04 14:12:37: Failed to ping 'Ikea_Outlet_1' (attempt 1/1, Read 0xcc86ecfffe8dc049/1 genBasic(["zclVersion"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'undefined' (25)))
Zigbee2MQTT:warn  2024-01-04 14:12:42: Failed to ping 'Ikea_E27_FC_02' (attempt 1/1, Read 0x94deb8fffea88c93/1 genBasic(["zclVersion"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'undefined' (25)))
Zigbee2MQTT:error 2024-01-04 14:12:49: Failed to read state of 'Ikea_B22_WS_01' after reconnect (Read 0x003c84fffe333674/1 genOnOff(["onOff"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'undefined' (25)))
```

Koenkk, 2024-01-04:
@nefh can you try the [`20221226`](https://github.com/Koenkk/Z-Stack-firmware/tree/Z-Stack_3.x.0_coordinator_20221226/coordinator/Z-Stack_3.x.0) firmware?

nefh, 2024-01-04:
> @nefh can you try the [`20221226`](https://github.com/Koenkk/Z-Stack-firmware/tree/Z-Stack_3.x.0_coordinator_20221226/coordinator/Z-Stack_3.x.0) firmware?

@Koenkk Unfortunately, the issue is still here with newest commits and 20221226 firmware. Here is the log:

[zigbee2mqtt_daemon_dev_20221226.log](https://github.com/Koenkk/zigbee2mqtt/files/13832458/zigbee2mqtt_daemon_dev_20221226.log)

BTW, before this issue I had firmware 20210120 - so it occurs also with that version.

Koenkk, 2024-01-04:
@nefh can you try changing the `adapter_concurrent` option to `4`? ([docs](https://www.zigbee2mqtt.io/guide/configuration/adapter-settings.html#advanced-configuration))

nefh, 2024-01-04:
@Koenkk Funny thing, initially it has not start with `adapter_concurent: 4`:

```
debug 2024-01-04 20:35:03: Loaded state from file /opt/zigbee2mqtt/data/state.json
info  2024-01-04 20:35:03: Logging to console and directory: '/opt/zigbee2mqtt/data/log/2024-01-04.20-35-03' filename: log.txt
debug 2024-01-04 20:35:03: Removing old log directory '/opt/zigbee2mqtt/data/log/2024-01-02.15-59-23'
info  2024-01-04 20:35:03: Starting Zigbee2MQTT version 1.42.0 (commit #bcf528b2)
info  2024-01-04 20:35:03: Starting zigbee-herdsman (0.30.0)
debug 2024-01-04 20:35:03: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":4,"delay":null,"disableLED":false},"backupPath":"/opt/zigbee2mqtt/data/coordinator_backup.json","databaseBackupPath":"/opt/zigbee2mqtt/data/database.db.backup","databasePath":"/opt/zigbee2mqtt/data/database.db","network":{"channelList":[11],"extendedPanID":[221,221,221,221,221,221,221,221],"networkKey":"HIDDEN","panID":6755},"serialPort":{"path":"/dev/ttyU0"}}'
error 2024-01-04 20:35:23: Error while starting zigbee-herdsman
error 2024-01-04 20:35:23: Failed to start zigbee
error 2024-01-04 20:35:23: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
error 2024-01-04 20:35:23: Exiting...
error 2024-01-04 20:35:23: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
    at ZStackAdapter.start (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:103:27)
    at Controller.start (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/controller.ts:132:29)
    at Zigbee.start (/opt/zigbee2mqtt/lib/zigbee.ts:60:27)
    at Controller.start (/opt/zigbee2mqtt/lib/controller.ts:101:27)
    at start (/opt/zigbee2mqtt/index.js:107:5)
```

However, it started after restart, but the issue still persist. Below is the log:
[log_concurrent_4.txt](https://github.com/Koenkk/zigbee2mqtt/files/13834780/log_concurrent_4.txt)

galvheim, 2024-01-04:
I want to add myself to this thread after getting alot.of the problems described in here lately. Normal HAOS with Z2M and Mosquitto MQTT. Collapses easily and with no warning. No indication other than zwitches.and sensors just hanging in their state. Only a full reboot will resolve things for a day or so befor things creep to a hault again.

Koenkk, 2024-01-05:
@nefh I see you are sending a lot of messages like ` Received MQTT message on 'zigbee2mqtt/0x003c84fffe2a74d3/get' with data '{"state":0,"brightness":0,"color_temp":0}'`, could you try disabling those?

nefh, 2024-01-05:
@Koenkk These messages seems to coming from homebridge zigbee2mqtt integration. After turning of homebridge entirely, the log is much better. However, homebridge integration was working fine before, it would be nice if it could still work. :) Of course, homebridge and integration were on the newest versions now.

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/13842717/log.txt)

nefh, 2024-01-05:
@Koenkk Intresting, I've cleared the cache on homebridge, and on first glance it looked ok. 
However, eventually automations with zigbee2mqtt became again unresponsive and timeouts started appearing in logs. The good news was lack of `No network route (205)` errors.

```
error 2024-01-05 16:02:29: Publish 'set' 'state' to 'KorytarzSufitWlacznik' failed: 'Error: Command 0x680ae2fffe096e3d/1 genOnOff.on({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 59985 - 1 - 45 - 6 - 11 after 10000ms)'
debug 2024-01-05 16:02:29: Error: Command 0x680ae2fffe096e3d/1 genOnOff.on({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 59985 - 1 - 45 - 6 - 11 after 10000ms)
    at Timeout._onTimeout (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35)
    at listOnTimeout (node:internal/timers:573:17)
    at processTimers (node:internal/timers:514:7)
info  2024-01-05 16:02:29: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'state' to 'KorytarzSufitWlacznik' failed: 'Error: Command 0x680ae2fffe096e3d/1 genOnOff.on({}, {\"sendWhen\":\"immediate\",\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":false,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (Timeout - 59985 - 1 - 45 - 6 - 11 after 10000ms)'","meta":{"friendly_name":"KorytarzSufitWlacznik"},"type":"zigbee_publish_error"}'
error 2024-01-05 16:02:29: Publish 'set' 'brightness' to 'KorytarzSufitLampa' failed: 'Error: Command 0x287681fffef82842/1 genLevelCtrl.moveToLevelWithOnOff({"level":254,"transtime":0}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 41752 - 1 - 46 - 8 - 11 after 10000ms)'
debug 2024-01-05 16:02:29: Error: Command 0x287681fffef82842/1 genLevelCtrl.moveToLevelWithOnOff({"level":254,"transtime":0}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 41752 - 1 - 46 - 8 - 11 after 10000ms)
    at Timeout._onTimeout (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35)
    at listOnTimeout (node:internal/timers:573:17)
    at processTimers (node:internal/timers:514:7)
info  2024-01-05 16:02:29: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'brightness' to 'KorytarzSufitLampa' failed: 'Error: Command 0x287681fffef82842/1 genLevelCtrl.moveToLevelWithOnOff({\"level\":254,\"transtime\":0}, {\"sendWhen\":\"immediate\",\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":false,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (Timeout - 41752 - 1 - 46 - 8 - 11 after 10000ms)'","meta":{"friendly_name":"KorytarzSufitLampa"},"type":"zigbee_publish_error"}'
```

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/13843614/log.txt)

Koenkk, 2024-01-06:
@nefh could you provide the herdsman debug log on 1.34 (working) and 1.35 (crashing)?

See [this](https://www.zigbee2mqtt.io/guide/usage/debug.html) on how to enable the herdsman debug logging. Note that this is only logged to STDOUT and not to log files.

nefh, 2024-01-06:
@Koenkk I have a problem with running debug:

```
# DEBUG=zigbee-herdsman* npm start
DEBUG=zigbee-herdsman*: No match.
```

nefh, 2024-01-06:
@Koenkk This time, I had to run dev version much longer to have any timeouts appear and automation after even that time are still responsive. I will run it longer to see if eventually will it freeze. In meantime, logs from ~10min with 1.34 and over an hour of dev version.

[log_herdsman_1.34.txt](https://github.com/Koenkk/zigbee2mqtt/files/13849084/log_herdsman_1.34.txt)
[log_herdsman_dev.txt](https://github.com/Koenkk/zigbee2mqtt/files/13849085/log_herdsman_dev.txt)
[log_1.34.txt](https://github.com/Koenkk/zigbee2mqtt/files/13849086/log_1.34.txt)
[log_dev.txt](https://github.com/Koenkk/zigbee2mqtt/files/13849087/log_dev.txt)

Koenkk, 2024-01-06:
What devices are `ToaletaLampa` and `SwiatlaCzujnik`? Are these reachable? After updating to 1.35.0 a lot of devices require a reconfigure (due to a big internal z2m change). You can see these in the log as `Configuring`, once `Successfully configured` they will not be configured again on next startup (unless you switch back to 1.34.0 and back to 1.35.0).

nefh, 2024-01-06:
ToaletaLampa: [Lonsonho QS-Zigbee-S05-LN](https://www.zigbee2mqtt.io/devices/QS-Zigbee-S05-LN.html#lonsonho-qs-zigbee-s05-ln)
SwiatlaCzujnik: [Xiaomi GZCGQ01LM](https://www.zigbee2mqtt.io/devices/GZCGQ01LM.html)

Both of them are reachable and visible in the UI:
[image]
[image]

Koenkk, 2024-01-06:
- Disabled the configure for QS-Zigbee-S05-LN (apparently not needed)
-  Regarding the [GZCGQ01LM](https://www.zigbee2mqtt.io/devices/GZCGQ01LM.html), try configuring it right after pressing the button on the device (you can use the. yellow refresh icon from your screenshot)

ilbarone87, 2024-01-09:
My network seems affected as well. Network completely down.
Mostly Ikea, Philips and Sonoff devices. Rebooted container, pi and HA instance several times.
I use Slaesh's CC2652RB stick + latest firmware. 

```MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'state' to 'Studio_Switch' failed: 'Error: Command 0x60a423fffed99a28/1 genOnOff.off({}, {\"sendWhen\":\"immediate\",\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":false,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (Data request failed with error: 'No network route' (205))'","meta":{"friendly_name":"Studio_Switch"},"type":"zigbee_publish_error"}'```

If I try to configure any of the device then i get this

`Zigbee2MQTT:info  2024-01-09 00:53:02: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/configure', payload '{"data":{"id":"Garden_Main"},"error":"Failed to configure (Bind 0x804b50fffea7f8bb/1 genOnOff from '0x00124b00214f21f3/1' failed (AREQ - ZDO - bindRsp after 10000ms))","status":"error","transaction":"yvlhn-3"}'`

rccoleman, 2024-01-09:
Weird.  I'm seeing these same messages for random devices that are bound to Inovelli Blue switches, but everything in my mesh is working fine (including the devices the switches and devices participating in the binding from the error message).

nefh, 2024-01-09:
@Koenkk I've run the dev build long run. Good news is it hasn't freezed at all. However, there are few more errors in the logs. Below I paste list of error messages filtered from duplicates and unused devices.

```
Failed to read state of 'KuchniaScianaLampa' after reconnect (Read 0x84fd27fffeb312c3/1 genLevelCtrl(["currentLevel"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 9721 - 1 - 47 - 8 - 1 after 10000ms))
Failed to read state of 'KuchniaSufitLampa' after reconnect (Read 0x84b4dbfffe6766a7/1 genLevelCtrl(["currentLevel"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 13991 - 1 - 46 - 8 - 1 after 10000ms))
Failed to read state of 'MamySufitLampa' after reconnect (Read 0x003c84fffe2a74d3/1 genLevelCtrl(["currentLevel"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'MAC no ack' (233)))
Failed to read state of 'PrzedpokojSufitLampa' after reconnect (Read 0xbc33acfffe94c72f/1 genLevelCtrl(["currentLevel"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'MAC no ack' (233)))
Failed to read state of 'SalonZyrandolPojedynczaLampa' after reconnect (Read 0x842e14fffe4135e8/1 genLevelCtrl(["currentLevel"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'MAC no ack' (233)))
Failed to read state of 'SchodyScianaLampa' after reconnect (Read 0x84fd27fffeaac683/1 genLevelCtrl(["currentLevel"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'MAC no ack' (233)))
Publish 'get' 'battery' to 'SalonBocznaLampaPilot' failed: 'Error: Read 0x588e81fffe523420/1 genPowerCfg(["batteryPercentageRemaining"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 12661 - 1 - 238 - 1 - 1 after 10000ms)'
Publish 'get' 'battery' to 'SalonRoletaPilot' failed: 'Error: Read 0x680ae2fffe6fcc0a/1 genPowerCfg(["batteryPercentageRemaining"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 43835 - 1 - 237 - 1 - 1 after 10000ms)'
Publish 'get' 'brightness' to 'KuchniaScianaLampa' failed: 'Error: Read 0x84fd27fffeb312c3/1 genLevelCtrl(["currentLevel"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 9721 - 1 - 245 - 8 - 1 after 10000ms)'
Publish 'get' 'brightness' to 'KuchniaSufitLampa' failed: 'Error: Read 0x84b4dbfffe6766a7/1 genLevelCtrl(["currentLevel"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 13991 - 1 - 244 - 8 - 1 after 10000ms)'
Publish 'get' 'brightness' to 'MamySufitLampa' failed: 'Error: Read 0x003c84fffe2a74d3/1 genLevelCtrl(["currentLevel"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'MAC channel access failure' (225))'
Publish 'get' 'brightness' to 'PrzedpokojSufitLampa' failed: 'Error: Read 0xbc33acfffe94c72f/1 genLevelCtrl(["currentLevel"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'MAC no ack' (233))'
[... cut ...]
[Lonsonho QS-Zigbee-S04-2C-LN](https://www.zigbee2mqtt.io/devices/QS-Zigbee-S04-2C-LN.html#lonsonho-qs-zigbee-s04-2c-ln)	'KorytarzSufitWlacznik'
[Lonsonho QS-Zigbee-S05-LN](https://www.zigbee2mqtt.io/devices/QS-Zigbee-S05-LN.html#lonsonho-qs-zigbee-s05-ln)	'ToaletaLampa'
[Xiaomi GZCGQ01LM](https://www.zigbee2mqtt.io/devices/GZCGQ01LM.html#xiaomi-gzcgq01lm)	'SwiatlaCzujnik'

Actual logs:
[log_dev_long_run.zip](https://github.com/Koenkk/zigbee2mqtt/files/13871769/log_dev_long_run.zip)

Hope it helps!

trizmark, 2024-01-09:
@ilbarone87 I spent most part of a day trying to get my network back online. Unfortunately, the only way forward was to roll back to 1.34 and revert the coordinator firmware (20221226 on zzh) as well. Once that was done, I deleted the whole device database and re-paired all 100+ devices. 95% IKEA (lights, switches), a couple of Sonoff relays and a couple of tuya curtain motors.

Koenkk, 2024-01-09:
@nefh these errors look like general network issues, I do not expect that 1.35.1 has an effect on it (only the configure part changed here), are you sure these errors didn't occur with 1.34.0?

WarleyGabriel, 2024-01-13:
Hey guys, I’m facing the same issue. 

I have 5 of this device in my network:
[image]

I started seem some errors like this one:
```
Zigbee2MQTT:error 2024-01-13 01:47:03: Publish 'set' 'state' to 'Persiana suíte principal - secundária' failed: 'Error: Command 0xa4c1380bff148d89/1 closuresWindowCovering.downClose({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'No network route' (205))'
Zigbee2MQTT:error 2024-01-13 01:47:19: Publish 'set' 'state' to 'Persiana suíte principal - secundária' failed: 'Error: Command 0xa4c1380bff148d89/1 closuresWindowCovering.stop({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'No network route' (205))'
Zigbee2MQTT:error 2024-01-13 01:55:45: Publish 'set' 'position' to 'Persiana suíte principal - secundária' failed: 'Error: Command 0xa4c1380bff148d89/1 closuresWindowCovering.goToLiftPercentage({"percentageliftvalue":0}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'No network route' (205))'
```

If I remove the device and pair again it starts working and after while it just stop working. 

I’m using zigbee2mqtt version: 1.35.1-1

johnnykang, 2024-01-15:
i am in the same boat. 1.35 has been causing lots of issues.

How do I downgrade back to 1.34 ? i am using the HA add-on.

Spooktram, 2024-01-18:
Hi all.
Same here and also running 1.35 here. All the network seems to be lost, except that I’m still able to toggle my light via Home Assistent with a massive delay. Thanks for any help! 🙏🏻

Clem-, 2024-01-24:
Hi, I also experienced a big network blackout but recovered almost all my devices after a hard reboot.

Two of my devices are still "no network route". Those devices are very similar (TS0502B and YSR-MINI-01_dimmer).
I'm on Z2M 1.35.1

Is there anything I can do to help here?

dangnguyen0906, 2024-02-16:
Comment here to see solution, got this issues in both fw 1.34 and 1.35

Trzinka, 2024-02-24:
# Hi! I have an even bigger problem!!!

[image]

[image]

## Is there anyone solving this problem?

spattinson, 2024-03-16:
I found this thread because I was having similar issues. A bunch of my devices go offline for extended periods. Removing them and trying to pair didn't work - they wouldn't pair. I rebooted my rock pi, it had power issues on boot, a usb drive kept clicking on and off. Changed power supply, no dice. Unplugged a google coral device that wasn't even in use and everything started working again. The coral was in a usb3 port next to the usb2 port my coordinator was plugged into. I suppose i should plug the coral back in and see if it causes an issue again, corals have two power/clock settings so I could try it on the lower one. I dont need it atm as I'm using the NPU onboard for frigate.

lucko71, 2024-03-20:
I am running **1.36** Z2M - I integrated a Sonoff **ZBMINIL2** and get lots of errors when I give commands either on Z2M web GUI or from H.A. Sometimes the command is executed after few minutes, sometimes never. The device is "disabled" in Z2N dashboard.

_2024-03-20 15:50:43Error: Command 0x84ba20fffecf16d5/1 genOnOff.on({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 30569 - 1 - 24 - 6 - 11 after 10000ms) at Object.start (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:63:23) at EZSPAdapter.sendZclFrameToEndpointInternal (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/adapter/ezspAdapter.ts:492:47) at processTicksAndRejections (node:internal/process/task_queues:95:5) at Queue.execute (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:35:20) at Request.send (/app/node_modules/zigbee-herdsman/src/controller/helpers/request.ts:79:20) at Endpoint.command (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:746:28) at Object.convertSet (/app/node_modules/zigbee-herdsman-converters/src/converters/toZigbee.ts:46:17) at Publish.onMQTTMessage (/app/lib/extension/publish.ts:259:36) at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:174:17)_

anordnes, 2024-04-04:
Mine is also like this arter updating to latest z2m  . Does not work to roll back to earlier backup. Did somebody found a fix for this?

tott, 2024-04-04:
Same here - happens for me with Edge as well as 1.36.1-1

anordnes, 2024-04-04:
I disconnected all electronics near my raspberry and coordinator. Moved my ssd drive away and restarted it and seems ok again.

evanfoster, 2024-04-10:
Hey there!

I need to try downgrading my Z2m from 1.36.1 to 1.34.X so I can generate comparable logs, but here's debug logs (including herdsman) that show the problem: https://gist.github.com/evanfoster/7de176651bffba5df5b4d725d6f01d15

At 18:55:28, I double pressed `Main Bedroom Button`. In Home Assistant, I have an automation that reacts to this automation and turns off the `Bedroom Plant Light`, which is a Third Reality 3RSP02028BZ plug (I'm doing this through HA instead of Z2M for Reasons :tm: I know that the more correct approach would be to bind the two devices together in Z2M, but this issue still happens with bound devices). Later on in that same second (18:55:28), the `OFF` message comes in from Home Assistant over MQTT. It appears that Z2M transmits to the Bedroom Plant Light, but the light is never turned off. I repeated the double click at ~5 second intervals, which works out to 18:55:33, 18:55:38 and 18:55:43. At 18:55:48.994, zigbee-herdsman raises the timeout exception that others have posted before.

Interestingly enough, it seems like the `last_seen` time is still getting updated, even though the plug doesn't seem to be responding to state updates.

Here are all of the versions/about page info of all components in the critical path:

* Home Assistant OS:
  * Core: 2024.4.2
  * Supervisor: 2024.04.0
  * OS: 12.1
* Zigbee2MQTT (running the Docker image in k8s):
  * Z2M version: [1.36.1](https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.36.1) commit: [ffc2ff1](https://github.com/Koenkk/zigbee2mqtt/commit/ffc2ff1)
  * Coordinator type: EZSP v12
  * Coordinator revision: 7.3.1.0 build 176
  * `zigbee_herdsman_converters_version`: 19.11.2
  * `zigbee_herdsman_version`: 0.40.3
* vernemq MQTT broker version: 1.13.0-alpine
* SMLight SLZB-06M networked Zigbee coordinator:
  * Core firmware version: v2.0.18
  * Zigbee firmware version: 20231030
* Living Room Repeater plug (Third Reality 3RSP02028BZ, intermediate router, LQI to coordinator 114/169)
  * Firmware: v1.00.74
  * IEEE address: `0x282c02bfffebe5b2`
  * Network address: `0xF16C`
* Bedroom Plant Light (Third Reality 3RSP02028BZ, intransigent fool of a smart plug, LQI to Living Room Repeater plug 50):
  * Firmware: v1.00.74
  * IEEE address: `0x282c02bfffe8cb11`
  * Network address: `0x39A8`

mainmind83, 2024-09-05:
I had these same messages after [CC1352P2 firmware update](https://www.mainmind.com/blog/actualizar-firmware-modulo-zigbee/) (CC1352P2_CC2652P_launchpad_coordinator_20240710.hex) on both [TubesZB](https://tubeszb.com/product/cc2652p2-based-zigbee-to-poe-coordinator-2023) and SONOFF Universal Zigbee 3.0 USB Dongle Plus (ZBDongle-P) It may take some time for the network to stabilize after the update.

Instead of waiting, I turned off all the devices in the entire house to force start completely. A couple of them appeared "offline" but had LQI value, sending several commands (on, off...) they appear again without problem