Koenkk/zigbee2mqtt #20528: UNSUPPORTED_ATTRIBUTE with unsupported device (generated definition?)

Cossid, 2024-01-03:
### What happened?

I have an unsupported device in which I have an incomplete, but semi-functioning external component driver.  I decided to give generated definition (new in 1.35.0) a try, but promptly get a crash when I remove my custom component definition.

```
Zigbee2MQTT:error 2024-01-02 18:28:47: Error: Read 0x000d6f000ebf4d6d/2 lightingColorCtrl(["colorCapabilities"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')
    at Endpoint.checkStatus (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:321:28)
    at Endpoint.read (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:463:22)
    at getClusterAttributeValue (/app/node_modules/zigbee-herdsman-converters/src/lib/utils.ts:452:9)
    at Array.extenderOnOffLight (/app/node_modules/zigbee-herdsman-converters/src/lib/generateDefinition.ts:92:35)
    at addExtenders (/app/node_modules/zigbee-herdsman-converters/src/lib/generateDefinition.ts:45:42)
    at generateDefinition (/app/node_modules/zigbee-herdsman-converters/src/lib/generateDefinition.ts:50:13)
    at findDefinition (/app/node_modules/zigbee-herdsman-converters/src/index.ts:289:35)
    at Object.findByDevice (/app/node_modules/zigbee-herdsman-converters/src/index.ts:261:22)
    at Device.resolveDefinition (/app/lib/model/device.ts:36:31)
    at Zigbee.start (/app/lib/zigbee.ts:67:13)
 ```

I see there are other reports for `UNSUPPORTED_ATTRIBUTE`, but they are getting device specific fixes by the looks of it, so I'm assuming this is a separate issue (#20354 , #20515 , #20527 for example).  Please close if it is covered by something else.  I'm not specifically getting anything about power_on_behavior here.

### What did you expect to happen?

Not crash the entire instance at the least?

### How to reproduce it (minimal and precise)

Add (specific?) unsupported devices.

### Zigbee2MQTT version

1.35.0

### Adapter firmware version

20220219

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus ZBDongle-P

### Debug log

```
Zigbee2MQTT:error 2024-01-02 18:28:47: Error: Read 0x000d6f000ebf4d6d/2 lightingColorCtrl(["colorCapabilities"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')
    at Endpoint.checkStatus (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:321:28)
    at Endpoint.read (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:463:22)
    at getClusterAttributeValue (/app/node_modules/zigbee-herdsman-converters/src/lib/utils.ts:452:9)
    at Array.extenderOnOffLight (/app/node_modules/zigbee-herdsman-converters/src/lib/generateDefinition.ts:92:35)
    at addExtenders (/app/node_modules/zigbee-herdsman-converters/src/lib/generateDefinition.ts:45:42)
    at generateDefinition (/app/node_modules/zigbee-herdsman-converters/src/lib/generateDefinition.ts:50:13)
    at findDefinition (/app/node_modules/zigbee-herdsman-converters/src/index.ts:289:35)
    at Object.findByDevice (/app/node_modules/zigbee-herdsman-converters/src/index.ts:261:22)
    at Device.resolveDefinition (/app/lib/model/device.ts:36:31)
    at Zigbee.start (/app/lib/zigbee.ts:67:13)
 ```

Koenkk, 2024-01-03:
Fixed in the dev branch!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

closed 2024-01-03 by Koenkk

Cossid, 2024-01-03:
I switched to the latest-dev docker which was built after your response, but I still get the exact same error/stack trace.

By the change log, it looks like you refactored powerOnBehavior, but the message I'm getting doesn't explicitly mention that attribute.  Perhaps the error message could be improved to include the name of the unsupported attribute?

Koenkk, 2024-01-03:
what shows up in the debug log now?

Cossid, 2024-01-03:
Exact same message and callstack and crash.  Same line numbers for each call.

Cossid, 2024-01-03:
Here is the full log with network settings redacted

```
Using '/app/data' as data directory
Zigbee2MQTT:debug 2024-01-03 13:10:42: Loaded state from file /app/data/state.json
Zigbee2MQTT:info  2024-01-03 13:10:42: Logging to console and directory: '/app/data/log/2024-01-03.13-10-42' filename: log.txt
Zigbee2MQTT:debug 2024-01-03 13:10:42: Removing old log directory '/app/data/log/2024-01-02.11-36-25'
Zigbee2MQTT:info  2024-01-03 13:10:42: Starting Zigbee2MQTT version 1.35.0 (commit #b5a90e9)
Zigbee2MQTT:info  2024-01-03 13:10:42: Starting zigbee-herdsman (0.30.0)
Zigbee2MQTT:debug 2024-01-03 13:10:42: Using zigbee-herdsman with settings: <<REDACTED>>
Zigbee2MQTT:error 2024-01-03 13:10:43: Failed to start zigbee
Zigbee2MQTT:error 2024-01-03 13:10:43: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
Zigbee2MQTT:error 2024-01-03 13:10:43: Exiting...
Zigbee2MQTT:error 2024-01-03 13:10:43: Error: Read 0x000d6f000ebf4d6d/2 lightingColorCtrl(["colorCapabilities"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')
    at Endpoint.checkStatus (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:321:28)
    at Endpoint.read (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:463:22)
    at getClusterAttributeValue (/app/node_modules/zigbee-herdsman-converters/src/lib/utils.ts:452:9)
    at Array.extenderOnOffLight (/app/node_modules/zigbee-herdsman-converters/src/lib/generateDefinition.ts:92:35)
    at addExtenders (/app/node_modules/zigbee-herdsman-converters/src/lib/generateDefinition.ts:45:42)
    at generateDefinition (/app/node_modules/zigbee-herdsman-converters/src/lib/generateDefinition.ts:50:13)
    at findDefinition (/app/node_modules/zigbee-herdsman-converters/src/index.ts:289:35)
    at Object.findByDevice (/app/node_modules/zigbee-herdsman-converters/src/index.ts:261:22)
    at Device.resolveDefinition (/app/lib/model/device.ts:36:31)
    at Zigbee.start (/app/lib/zigbee.ts:67:13)
```

Where the entire container crashes and stops

Looks like it's still on commit #b5a90e9 , but the build time was from today.  I'm guessing it's just it's just the converter has updated?  It's for sure the running latest-dev from docker hub and is up-to-day.

Cossid, 2024-01-03:
Sorry, forgot I had switched back to stable.

Here is the log from dev, which looks identical, except for the commit line

```
Using '/app/data' as data directory
Zigbee2MQTT:debug 2024-01-03 13:22:36: Loaded state from file /app/data/state.json
Zigbee2MQTT:info  2024-01-03 13:22:36: Logging to console and directory: '/app/data/log/2024-01-03.13-22-36' filename: log.txt
Zigbee2MQTT:debug 2024-01-03 13:22:36: Removing old log directory '/app/data/log/2024-01-02.11-37-31'
Zigbee2MQTT:info  2024-01-03 13:22:36: Starting Zigbee2MQTT version 1.35.0-dev (commit #e9aee4c)
Zigbee2MQTT:info  2024-01-03 13:22:36: Starting zigbee-herdsman (0.30.0)
Zigbee2MQTT:debug 2024-01-03 13:22:36: Using zigbee-herdsman with settings: <<REDACTED>>
Zigbee2MQTT:error 2024-01-03 13:22:36: Failed to start zigbee
Zigbee2MQTT:error 2024-01-03 13:22:36: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
Zigbee2MQTT:error 2024-01-03 13:22:36: Exiting...
Zigbee2MQTT:error 2024-01-03 13:22:36: Error: Read 0x000d6f000ebf4d6d/2 lightingColorCtrl(["colorCapabilities"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')
    at Endpoint.checkStatus (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:321:28)
    at Endpoint.read (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:463:22)
    at getClusterAttributeValue (/app/node_modules/zigbee-herdsman-converters/src/lib/utils.ts:452:9)
    at Array.extenderOnOffLight (/app/node_modules/zigbee-herdsman-converters/src/lib/generateDefinition.ts:92:35)
    at addExtenders (/app/node_modules/zigbee-herdsman-converters/src/lib/generateDefinition.ts:45:42)
    at generateDefinition (/app/node_modules/zigbee-herdsman-converters/src/lib/generateDefinition.ts:50:13)
    at findDefinition (/app/node_modules/zigbee-herdsman-converters/src/index.ts:289:35)
    at Object.findByDevice (/app/node_modules/zigbee-herdsman-converters/src/index.ts:261:22)
    at Device.resolveDefinition (/app/lib/model/device.ts:36:31)
    at Zigbee.start (/app/lib/zigbee.ts:67:13)
```

Koenkk, 2024-01-04:
Found the issue, pushed a fix!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

Cossid, 2024-01-04:
Still getting the exact same error
```
Using '/app/data' as data directory
Zigbee2MQTT:debug 2024-01-04 08:17:15: Loaded state from file /app/data/state.json
Zigbee2MQTT:info  2024-01-04 08:17:15: Logging to console and directory: '/app/data/log/2024-01-04.08-17-15' filename: log.txt
Zigbee2MQTT:debug 2024-01-04 08:17:15: Removing old log directory '/app/data/log/2024-01-02.11-39-59'
Zigbee2MQTT:info  2024-01-04 08:17:15: Starting Zigbee2MQTT version 1.35.0-dev (commit #99a57bf)
Zigbee2MQTT:info  2024-01-04 08:17:15: Starting zigbee-herdsman (0.30.0)
Zigbee2MQTT:debug 2024-01-04 08:17:15: Using zigbee-herdsman with settings: <<REDACTED>>
Zigbee2MQTT:error 2024-01-04 08:17:16: Failed to start zigbee
Zigbee2MQTT:error 2024-01-04 08:17:16: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
Zigbee2MQTT:error 2024-01-04 08:17:16: Exiting...
Zigbee2MQTT:error 2024-01-04 08:17:16: Error: Read 0x000d6f000ebf4d6d/2 lightingColorCtrl(["colorCapabilities"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')
    at Endpoint.checkStatus (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:321:28)
    at Endpoint.read (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:463:22)
    at getClusterAttributeValue (/app/node_modules/zigbee-herdsman-converters/src/lib/utils.ts:452:9)
    at Array.extenderOnOffLight (/app/node_modules/zigbee-herdsman-converters/src/lib/generateDefinition.ts:92:35)
    at addExtenders (/app/node_modules/zigbee-herdsman-converters/src/lib/generateDefinition.ts:45:42)
    at generateDefinition (/app/node_modules/zigbee-herdsman-converters/src/lib/generateDefinition.ts:50:13)
    at findDefinition (/app/node_modules/zigbee-herdsman-converters/src/index.ts:289:35)
    at Object.findByDevice (/app/node_modules/zigbee-herdsman-converters/src/index.ts:261:22)
    at Device.resolveDefinition (/app/lib/model/device.ts:36:31)
    at Zigbee.start (/app/lib/zigbee.ts:67:13)
```

The commit updated, but the line numbers didn't suggesting maybe the converter didn't get updated in the most recent build?  I'm assuming the generateDefinition line number should have changed from 92 to 93 if the converter was updated.  The startup log doesn't list what version of converter is currently present.

Koenkk, 2024-01-04:
Please wait a few hours, I will publish a new version soon.

Koenkk, 2024-01-04:
- What device is `Entry Smoke Alarm`, can you provide the data/database.db of it?
- Can you provide a bit more debug logging of what happens before this?

Cossid, 2024-01-04:
There are two similar devices, different models of the old HaloSmartLabs combo smoke/co2, ones is the `halo`, the other the `halo+` (which includes emergency weather scanning/reporting, but doesn't really work anymore)

database.db for the halo+:
```json
{"id":6,"type":"Router","ieeeAddr":"0x000d6f000ebf4b82","nwkAddr":30968,"manufId":4609,"manufName":"HaloSmartLabs","powerSource":"Mains (single phase)","modelId":"haloWX","epList":[1,2,3,4,5],"endpoints":{"1":{"profId":260,"epId":1,"devId":3,"inClusterList":[0,1,3,1026,1027,1029,1280,1282],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"stackVersion":0}},"ssIasZone":{"attributes":{"iasCieAddr":"0x00124b0026b73d6f","zoneState":1,"zoneId":23}},"genPowerCfg":{"attributes":{"batteryPercentageRemaining":200,"batteryVoltage":41}},"msTemperatureMeasurement":{"attributes":{"measuredValue":2023}},"msPressureMeasurement":{"attributes":{"measuredValue":998}},"msRelativeHumidity":{"attributes":{"measuredValue":2589}},"genOnOff":{"attributes":{}},"genLevelCtrl":{"attributes":{}},"lightingColorCtrl":{"attributes":{}},"ssIasWd":{"attributes":{"maxDuration":240}}},"binds":[{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0x00124b0026b73d6f","endpointID":1},{"cluster":1026,"type":"endpoint","deviceIeeeAddress":"0x00124b0026b73d6f","endpointID":1},{"cluster":1027,"type":"endpoint","deviceIeeeAddress":"0x00124b0026b73d6f","endpointID":1},{"cluster":1029,"type":"endpoint","deviceIeeeAddress":"0x00124b0026b73d6f","endpointID":1},{"cluster":0,"type":"endpoint","deviceIeeeAddress":"0x00124b0026b73d6f","endpointID":1},{"cluster":1280,"type":"endpoint","deviceIeeeAddress":"0x00124b0026b73d6f","endpointID":1},{"cluster":1282,"type":"endpoint","deviceIeeeAddress":"0x00124b0026b73d6f","endpointID":1}],"configuredReportings":[{"cluster":1,"attrId":32,"minRepIntval":3600,"maxRepIntval":65000,"repChange":0,"manufacturerCode":null},{"cluster":1,"attrId":33,"minRepIntval":3600,"maxRepIntval":65000,"repChange":0,"manufacturerCode":null},{"cluster":1026,"attrId":0,"minRepIntval":10,"maxRepIntval":3600,"repChange":100,"manufacturerCode":null},{"cluster":1027,"attrId":0,"minRepIntval":10,"maxRepIntval":3600,"repChange":5,"manufacturerCode":null},{"cluster":1029,"attrId":0,"minRepIntval":10,"maxRepIntval":3600,"repChange":100,"manufacturerCode":null}],"meta":{}},"2":{"profId":260,"epId":2,"devId":1027,"inClusterList":[3,6,8,768],"outClusterList":[],"clusters":{"lightingColorCtrl":{"attributes":{"currentHue":169,"currentSaturation":254,"currentX":10916,"currentY":0,"colorMode":0,"options":0}},"genOnOff":{"attributes":{"onOff":0}},"genLevelCtrl":{"attributes":{"currentLevel":254}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b0026b73d6f","endpointID":1},{"cluster":8,"type":"endpoint","deviceIeeeAddress":"0x00124b0026b73d6f","endpointID":1},{"cluster":768,"type":"endpoint","deviceIeeeAddress":"0x00124b0026b73d6f","endpointID":1}],"configuredReportings":[{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":3600,"repChange":0,"manufacturerCode":null},{"cluster":8,"attrId":0,"minRepIntval":1,"maxRepIntval":3600,"repChange":1,"manufacturerCode":null}],"meta":{}},"3":{"profId":260,"epId":3,"devId":1026,"inClusterList":[3,1280],"outClusterList":[],"clusters":{"ssIasZone":{"attributes":{"iasCieAddr":"0x00124b0026b73d6f","zoneState":1,"zoneStatus":0,"zoneId":23,"zoneType":551}}},"binds":[{"cluster":1280,"type":"endpoint","deviceIeeeAddress":"0x00124b0026b73d6f","endpointID":1}],"configuredReportings":[],"meta":{}},"4":{"profId":260,"epId":4,"devId":64768,"inClusterList":[64768,64769,64770],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"5":{"profId":260,"epId":5,"devId":64769,"inClusterList":[64771],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":1,"stackVersion":0,"hwVersion":1,"zclVersion":1,"interviewCompleted":true,"meta":{"configured":-1073125925},"lastSeen":1704391558233,"defaultSendRequestWhen":"immediate"}
```
and prettified:
```json
{
    "id": 6,
    "type": "Router",
    "ieeeAddr": "0x000d6f000ebf4b82",
    "nwkAddr": 30968,
    "manufId": 4609,
    "manufName": "HaloSmartLabs",
    "powerSource": "Mains (single phase)",
    "modelId": "haloWX",
    "epList": [1, 2, 3, 4, 5],
    "endpoints": {
        "1": {
            "profId": 260,
            "epId": 1,
            "devId": 3,
            "inClusterList": [0, 1, 3, 1026, 1027, 1029, 1280, 1282],
            "outClusterList": [25],
            "clusters": {
                "genBasic": {
                    "attributes": {
                        "stackVersion": 0
                    }
                },
                "ssIasZone": {
                    "attributes": {
                        "iasCieAddr": "0x00124b0026b73d6f",
                        "zoneState": 1,
                        "zoneId": 23
                    }
                },
                "genPowerCfg": {
                    "attributes": {
                        "batteryPercentageRemaining": 200,
                        "batteryVoltage": 41
                    }
                },
                "msTemperatureMeasurement": {
                    "attributes": {
                        "measuredValue": 2023
                    }
                },
                "msPressureMeasurement": {
                    "attributes": {
                        "measuredValue": 998
                    }
                },
[... cut ...]
I messed around with a custom driver a while ago, but had a ton of trouble with the light, and just ignored it.  Generated Definition picks up the batteryPercentage (but not voltage), temperature, humidity, pressure, and light (light doesn't work), but misses the smoke, co2, and tamper, and whatever other entities I don't understand.

As for what triggers those errors, just hitting the refresh button next to the binding in the Exposes tab in the UI for each of those entities, though I think it also gives the error on initial setup/query of the device at boot as well.

Koenkk, 2024-01-04:
>  though I think it also gives the error on initial setup/query of the device at boot as well.

Could you provide the debug log of this?

Regarding the definition, adding this to it might fix your problems:

```
endpoint: (device) => {
            return {default: 2};
        },
```

Cossid, 2024-01-04:
> > though I think it also gives the error on initial setup/query of the device at boot as well.
> 
> Could you provide the debug log of this?

I stand corrected, nothing from startup or idle functionality, only when trying to interact with certain attributes of the light related to startup/power on (changing startup color temp, power state).  Others light settings don't throw an error, but also do nothing.

Example of trying to change the startup color temp:
```
error 2024-01-04 12:53:52: Publish 'set' 'color_temp_startup' to '0x000d6f000ebf4b82' failed: 'Error: Write 0x000d6f000ebf4b82/4 lightingColorCtrl({"startUpColorTemperature":454}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'
debug 2024-01-04 12:53:52: Error: Write 0x000d6f000ebf4b82/4 lightingColorCtrl({"startUpColorTemperature":454}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')
    at Endpoint.checkStatus (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:321:28)
    at Endpoint.write (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:390:22)
    at Object.convertSet (/app/node_modules/zigbee-herdsman-converters/src/converters/toZigbee.ts:1180:13)
    at Publish.onMQTTMessage (/app/lib/extension/publish.ts:249:36)
```

Cossid, 2024-01-04:
> Regarding the definition, adding this to it might fix your problems:
> 
> ```
> endpoint: (device) => {
>             return {default: 2};
>         },
> ```

I'll look into that a bit later as I have more time.  I'm still not super comfortable with device definitions.

Cossid, 2024-01-04:
> Regarding the definition, adding this to it might fix your problems:
> 
> ```
> endpoint: (device) => {
>             return {default: 2};
>         },
> ```

Well, that changed the results significantly, but I wouldn't say fixed.

I no longer get errors immediately when attempted to refresh the values.

The generated definition is:
```json
const {batteryPercentage, temperature, pressure, humidity, light} = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
    zigbeeModel: ['haloWX', 'halo+', 'SABDA1'],
    model: 'haloWX',
    vendor: 'HaloSmartLabs',
    description: 'Conbination Smoke/CO2 detector',
    extend: [batteryPercentage(), temperature(), pressure(), humidity(), light({"colorTemp":{"range":[150,500]},"color":{"modes":["xy","hs"],"enhancedHue":true}})],
};

module.exports = definition;
```
which I modified to
```json
const {batteryPercentage, temperature, pressure, humidity, light} = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
    zigbeeModel: ['haloWX', 'halo+', 'SABDA1'],
    model: 'haloWX',
    vendor: 'HaloSmartLabs',
    description: 'Conbination Smoke/CO2 detector',
    endpoint: (device) => {
        return {default: 2};
    },
    extend: [batteryPercentage(), temperature(), pressure(), humidity(), light({"colorTemp":{"range":[150,500]},"color":{"modes":["xy","hs"],"enhancedHue":true}})],
};

module.exports = definition;
```

When doing that, I no longer get errors refreshing the state of the light, but I then do for refreshing anything non-light (because of the forced endpoint I'm sure).  Beyond that, the light still doesn't work, though it is sometimes turning on/off, but never adjusting, but it's really hard to diagnose when as there are a lot of timeout errors when publishing set states like
```
error 2024-01-04 13:52:33: Publish 'set' 'state' to '0x000d6f000ebf4b82' failed: 'Error: Command 0x000d6f000ebf4b82/2 genOnOff.on({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 1226 - 2 - 59 - 6 - 11 after 10000ms)'
debug 2024-01-04 13:52:33: Error: Command 0x000d6f000ebf4b82/2 genOnOff.on({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 1226 - 2 - 59 - 6 - 11 after 10000ms)
    at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35)
    at listOnTimeout (node:internal/timers:569:17)
    at processTimers (node:internal/timers:512:7)
```

It does turn on & off, but everything is delayed by the publish timeouts.  Anything related to color or color temperature times out and does reflect the change

```
error 2024-01-04 13:54:43: Publish 'set' 'color' to '0x000d6f000ebf4b82' failed: 'Error: Command 0x000d6f000ebf4b82/2 lightingColorCtrl.moveToColor({"transtime":0,"colorx":45494,"colory":19418}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 1226 - 2 - 62 - 768 - 11 after 10000ms)'
debug 2024-01-04 13:54:43: Error: Command 0x000d6f000ebf4b82/2 lightingColorCtrl.moveToColor({"transtime":0,"colorx":45494,"colory":19418}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 1226 - 2 - 62 - 768 - 11 after 10000ms)
    at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35)
    at listOnTimeout (node:internal/timers:569:17)
    at processTimers (node:internal/timers:512:7)
```

This is why I had such trouble trying to create a custom definition, it's a very unusual device.

Koenkk, 2024-01-05:
Try also adding `meta: {disableDefaultResponse: true},` to the definition.

Cossid, 2024-01-07:
That removes the delay/timeout when trying to referesh/get states.  For the light, on/off and brightness work, but nothing related to color/color temperature seem to do anything.