Koenkk/zigbee2mqtt #20532: [Feature request]: Naming convention for exposes and default categorization in Home Assistant

slugzero, 2024-01-03:
### Is your feature request related to a problem? Please describe

I'd like to put this up for discussion before implementing a test for it and creating a PR:

_As a user:_ Every time I update zigbee2mqtt, a bunch of new entities pops up in Home Assistant, mostly device specific configuration options which I rarely need. These should be hidden by default. Home Assistant provides the entity categories "config" and "diagnostic" for that purpose. For example, right now I have the `power_outage_count` entity for the MCCGQ11LM and several seldomly used configuration entities for the BTH-RA like `valve_adapt_status` or `display_ontime`.

_As a developer:_ If I want these device specific options to be treated properly, I cannot do this inside the zigbee_herdsman_converters repo. Instead, I have to also modify `homeassistant.ts` in zigbee2mqtt with a separate PR. (And if the configuration is in a device specific cluster, I might need a 3rd one for the cluster definition in zigbee-herdsman repo. And a 4th for the documentation)

### Describe the solution you'd like

1. I propose to introduce the following naming convention for adding _new_ exposes: Device specific configuration exposes (i.e. exposes that are not yet in the list of existing exposes and will most likely not be required for other devices) are prefixed with `config_`,  and device specific diagnostic exposes are prefixed with `diagnostic_` (I am open for different naming).

2. Exposes with prefix `config_` or `diagnostic_` are added to the respective categories in Home Assistant by default like this: https://github.com/slugzero/zigbee2mqtt/commit/6bbe28c3d6563aa3baf9430a247099e2e4e8aeef

3. This could also help to let the list of exposes on https://www.zigbee2mqtt.io/supported-devices/ grow more slowly, because exposes starting with `config_` or `diagnostic_` could be filtered out easily.

### Describe alternatives you've considered

- Keep adding the exposes to `homeassistant.ts` myself for the devices I own :-) This would be required for all _existing_ exposes anyway, because I guess we do not want to rename dozens of exposes.
- Add some sort of additional flag to the exposes for configuration/diagnostic exposes. This solution could also cover existing exposes without renaming. However, this would require a bigger code change. It is also possible to combine this with the proposed solution at a leter point in time so, I'd suggest to leave this open for now and start with the small solution for new exposes first.

### Additional context

Proposed implementation is here: https://github.com/slugzero/zigbee2mqtt/commit/6bbe28c3d6563aa3baf9430a247099e2e4e8aeef

sjorge, 2024-01-03:
It's would probably be better to have a optional extra property in the expose information to tag things as config and diagnostic, that way other things .e.g. the frontend can also sue that information to for example hide things or group things.

I don't thing using a prefix is a great idea, we already have a bunch of reserved suffixes for example and sometimes dealing with those is a pain. 

Adding an extra field to the expose information would be less impact for other people who already relay on certain attribute names.

osnwt, 2024-01-04:
https://github.com/Koenkk/zigbee2mqtt/issues/20376

slugzero, 2024-01-04:
Fair point, @sjorge, thanks for the feedback. To make sure I got you right - you mean adding a `withCategory()` function like this, right?

https://github.com/slugzero/zigbee-herdsman-converters/commit/7a4850beaf9d6452cb0952e9a1d4487c7871ebab

sjorge, 2024-01-04:
Yes, i meant something like that. Probably make the variable type an enum to limit the options and make tsc complain if a other value is used.

slugzero, 2024-01-05:
ok, I opened a PR. It also adds a bit more validation according to the expose access

Koenkk, 2024-01-05:
@slugzero thanks for looking into this. The thing I don't like about https://github.com/Koenkk/zigbee-herdsman-converters/pull/6837 is that we are now putting HA specific stuff in zigbee-herdsman-converters. I'm wondering that maybe we can always set category to diagnostic (for read only) or config (for read/write) and create a list with exceptions to this?

sjorge, 2024-01-05:
I'd like the category hint to also be available in the generic exposes data, it makes it easier to filter attribute based on that.

slugzero, 2024-01-05:
@Koenkk, @sjorge I think in general, a distinction in primary exposes (for the user) and secondary exposes (for maintenance/introspection) would be useful for other purposes, too (e.g. filtering on zigbee2mqtt.io or in the frontend). However I do agree that the config/diagnostic distinction from Home Assistant is probably not required. How about having only a single category, (e.g. "secondary", open for better names), and doing the config/diagnostic translation directly in `homeassistant.ts`?

Koenkk, 2024-01-06:
@slugzero I'm fine with that, however I'm not sure about the category names yet

slugzero, 2024-01-06:
@Koenkk Yes, so am I. Configuration and diagnostics are different things, but if they are treated the same programmatically, people will mix them up and just choose either of these. But I haven't found a good term to summarize both.

You probably have seen my updated PR, where I named it 'system' (as opposed to 'user'), which is maybe a bit too generic. Other terms I could think of are 'admin', 'management', 'internal', or a combination of these, e g. 'sysMgmt' 

For reference, Here's also the update in`homeassistant.ts`:
https://github.com/slugzero/zigbee2mqtt/commit/416dd9c56815147c9af7eea8de282159f1f40937

Koenkk, 2024-01-06:
Maybe the `diagnostic`/`config` category for Home Assistant is not that bad after all, it's at least quite clear.

sjorge, 2024-01-06:
> Maybe the `diagnostic`/`config` category for Home Assistant is not that bad after all, it's at least quite clear.

I don't use home assistant and I think the names are pretty good even for just plain extend payload. It clearly describes what the attribute is force. e.g. `power_outage_count` and `device_temperature` for Xiaomi, Aqara, Maji devices, it offers no 'useful' information for things the device/sensor collects. But they can help with diagnosing issues. e.g. is the in wall device overheating because device_temperature is a very high value, did my sensor drop and come back (power_outage_count) so stuff might be missing, ... that is useful but not what you have the device for...

I'm just not sure what a good naming would be for the normal properties, we can leave it blank I suppose.

slugzero, 2024-01-06:
Alright, I reverted it back to `diagnostic`/`config`. I also left the `validateCategory()` in. I would be also fine with removing it, but I think it would help people to use the categories in a meaningful way. (I don't think a read-only config expose makes sense, but it is somewhat arguable whether an expose that is primarily for diagnostics could also be settable).

Wrt the normal properties, I would also suggest to just leave it blank.