Koenkk/zigbee2mqtt #20533: Almost all devices disappeared

acca84, 2024-01-03:
### What happened?

Hello, I'm using zigbee2mqtt docker image on my qnap nas, it is automatically updated.

Yesterday, I had between 10 and 20 devices, today I only have 3. 
I don't understand why but I didn't removed it. 

What can I do to identify the origin of this problem?
Thanks and happy new year!

### Zigbee2MQTT version

1.35

### Adapter firmware version

6.7.8.0 build 373

### Adapter

Sonoff zb  bridge

pawol, 2024-01-03:
1.35 is a piece of sh.t. Perform revert onto 1.34

cpaucar3, 2024-01-05:
My container was crashing on 1.35, rolled back to 1.34 to stop the crashing.

Koenkk, 2024-01-05:
Please provide logging, without it I cannot do anything.

See [this](https://www.zigbee2mqtt.io/guide/usage/debug.html) on how to enable debug logging.

closed 2024-08-03 by github-actions[bot]