Koenkk/zigbee2mqtt #20535: Zigbee plugs don't execute On/Off instructions

FredM17, 2024-01-03:
### What happened?

I have 2 NOUS Zigbee A1Z connected sockets that rarely respond to On or Off instructions whether through automations, the device management screen (Settings / Devices and services) or Zigbee2MQTT.

### What did you expect to happen?

I expected the sockets to obey the instructions.

### How to reproduce it (minimal and precise)

It is happeming regularly on my config.

### Zigbee2MQTT version

 1.35.0-1

### Adapter firmware version

 1.35.0

### Adapter

Sonoff Zigbee 3.0 USB Dongle Plus

### Debug log

Here's the error message I get in Zigbee2MQTT when I send an Off instruction to one of the sockets:
Error 2024-01-03 11:24:27Publish 'set' 'state' to 'Fibre_&_HA' failed: 'Error: Command 0xa4c13805404a1c08/1 genOnOff. on({}, {"sendWhen": "immediate", "timeout":10000, "disableResponse":false, "disableRecovery":false, "disableDefaultResponse":false, "direction":0, "srcEndpoint":null, "reservedBits":0, "manufacturerCode":null, "transactionSequenceNumber":null, "writeUndiv":false}) failed (Data request failed with error: 'MAC channel access failure' (225))'
LQI of 'Fibre_&_HA' socket: 153

Koenkk, 2024-01-04:
`MAC channel access failure' (225))` means there is too many interference, see the [faq](https://www.zigbee2mqtt.io/guide/faq/#common-error-codes) for more info.

closed 2024-01-04 by Koenkk

FredM17, 2024-01-04:
Hi, Thanks, I got the solution from another forum, just added a usb cable to distance the Sonoff dongle from the computer.

Thanks