Koenkk/zigbee2mqtt #20536: Aqara E1 QBKG39LM 2 way (without 0-line)

nork1n, 2024-01-03:
### What happened?

In the new version of z2m 1.35.0-1, relay decoupling can be selected in the interface and there are no errors for this, but now it is impossible to turn on the lamp using the keys, because There are errors about timeout. I think that this is also why the "flip indicator light" function does not work. Now it can be turned on or off - the LEDs on the case do not work as they should.

### Zigbee2MQTT version

1.35.0-1

### Adapter firmware version

20230507

### Adapter

zigstar-lan-E72

### Debug log

Zigbee2MQTT:error 2024-01-03 15:13:50: Publish 'set' 'mode_switch' to '0x54ef441000182d6d' failed: 'Error: Write 0x54ef441000182d6d/1 aqaraOpple({"4":{"value":1,"type":33}}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":4447,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 53905 - 1 - 164 - 64704 - 4 after 10000ms)'
Zigbee2MQTT:error 2024-01-03 15:14:13: Publish 'set' 'power_outage_memory' to '0x54ef441000182d6d' failed: 'Error: Write 0x54ef441000182d6d/1 aqaraOpple({"513":{"value":1,"type":16}}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":4447,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 53905 - 1 - 165 - 64704 - 4 after 10000ms)'
Zigbee2MQTT:error 2024-01-03 15:14:36: Publish 'set' 'flip_indicator_light' to '0x54ef441000182d6d' failed: 'Error: Write 0x54ef441000182d6d/1 aqaraOpple({"240":{"value":1,"type":32}}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":4447,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 53905 - 1 - 166 - 64704 - 4 after 10000ms)'
Zigbee2MQTT:error 2024-01-03 15:14:59: Publish 'set' 'operation_mode' to '0x54ef441000182d6d' failed: 'Error: Write 0x54ef441000182d6d/1 aqaraOpple({"512":{"value":0,"type":32}}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":4447,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 53905 - 1 - 167 - 64704 - 4 after 10000ms)'
Zigbee2MQTT:error 2024-01-03 15:15:22: Publish 'set' 'operation_mode' to '0x54ef441000182d6d' failed: 'Error: Write 0x54ef441000182d6d/2 aqaraOpple({"512":{"value":0,"type":32}}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":4447,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 53905 - 2 - 168 - 64704 - 4 after 10000ms)'

vlavrynovych, 2024-01-13:
I had a similar issue even after the update to 1.35.1. 
Here is my error:
```
Publish 'set' 'operation_mode' to 'Switch X' failed: 'Error: Write 0x54ef44100017f6b0/1 aqaraOpple({"512":{"value":1,"type":32}}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":4447,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'Timeout' (9999))'
```

@nork1n , you can try to **re-connect your device** - it helped in my case. So, just start pairing mode and press a button (or what is needed for your device)

nork1n, 2024-01-13:
@vlavrynovych Yes, I tried this, but it doesn't solve the problem. At the moment, in the interface the keys are decoupled from the relay and when you press the keys there is no click, but when you turn off the lamp from the remote control, this switch disappears from the network, as if it had been de-energized, while physically the relays on the switch are in the on position.

nork1n, 2024-01-22:
> I have problems with that switch too. Looks like problem appeared after ota update of that switch. I cant even turn on/off switch from web gui. Previous firmware for switch works great with z2mqtt. Other my aqara switches (1 and 2 gang with neutral) except that model works great after ota update.
> 
> `Publish 'set' 'state' to 'Switch-bathroom' failed: 'Error: Command 0x54ef4410000f39ef/2 genOnOff.on({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'No network route' (205))'`
>
--------------------
It seems to me that I also updated, but unfortunately I don’t remember which version was earlier and whether the update was successful. Now if you look through OTA it shows the date of creation of the firmware and there is no version: QBKG39LM  07-01-2022. What to do now? throw it away or you can somehow downgrade the firmware version? At the moment I have the following error:
```Failed to ping 'Switch-smroom' (attempt 1/1, Read 0x54ef441000182d6d/1 genBasic(["zclVersion"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 16555 - 1 - 133 - 0 - 1 after 10000ms)) ```
I have an Yeelight lamp connected to this switch and when you turn it off via the remote control, the key behaves as if the relay was turned off on it, although no one touches the relay and the status is “decoupled” in the interface.

Smosia, 2024-01-23:
Force delete from z2mqtt and rejoin again works for me. Try it.

nork1n, 2024-01-23:
> Force delete from z2mqtt and rejoin again works for me. Try it.
> 
> P.s. current version of firmware for switch: 23 Previous one: 21. You can check verion in status tab of switch.

Thank you, my firmware is 23 ((I tried forced deletion and re-pairing, but this did not solve the problem.

nork1n, 2024-01-23:
Today I bought a new Akara WS-EUK01 key, connected it to HA and to my surprise or regret, the situation is exactly the same, the relay is disconnected and when the lamp is turned off - the key goes offline. Automations with the relay uncoupled are not executed.

closed 2024-08-22 by github-actions[bot]