Koenkk/zigbee2mqtt #20545: IKEA on/off button triggers an Aqara H1 switch

wralach, 2024-01-03:
### What happened?

I have the following devices added in the zigbee2mqtt addon on Home Assistant:
- H1 Switch as described in [https://www.zigbee2mqtt.io/devices/WS-EUK04.html#xiaomi-ws-euk04](https://www.zigbee2mqtt.io/devices/WS-EUK04.html#xiaomi-ws-euk04)
- IKEA on/off switch E1743 as in [https://www.zigbee2mqtt.io/devices/E1743.html#ikea-e1743](https://www.zigbee2mqtt.io/devices/E1743.html#ikea-e1743)

The IKEA button uses the Aqara H1 switch as router.

When the Ikea button is clicked, the Aqara switch (left button, which is connected to the wire) is triggered, too. The logs below were produced after only clicking on the IKEA button.

I resetted the IKEA button and it worked fine for a while. Note: at that time the IKEA button was connected to a different router in the Zigbee network. Once the ‘issue’ re-appeared and the IKEA button started turning on the lights of the Aqara switch, I noticed, that IKEA uses the Aqara H1 as router.

Worth mentioning: I created a new HA instance during the past week and basically did a fresh set-up. However, the issues appeared also on the new instance.
I am using HA core 2023.12.4 and zigbee2mqtt in the most recent version.

### What did you expect to happen?

When pushing the IKEA button (which is connected to the Aqara H1 switch as router), the Aqara H1 switch should not execute any action.

### How to reproduce it (minimal and precise)

Connect both, the IKEA button + Aqara H1 switch.
Ensure, that the IKEA button uses the Aqara H1 switch as router.

### Zigbee2MQTT version

1.35.0

### Adapter firmware version

20230507

### Adapter

Silicon_Labs_Sonoff_Zigbee_3.0_USB_Dongle_Plus_0001-if00-port0

### Debug log

Debug 2023-12-29 10:05:03Received Zigbee message from 'IKEA Traedfri On-Off Button', type 'commandOn', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 0
Info 2023-12-29 10:05:03MQTT publish: topic 'zigbee2mqtt/IKEA Traedfri On-Off Button', payload '{"action":"on","battery":60,"linkquality":112,"update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null}'
Info 2023-12-29 10:05:03MQTT publish: topic 'zigbee2mqtt/IKEA Traedfri On-Off Button', payload '{"action":"","battery":60,"linkquality":112,"update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null}'
Info 2023-12-29 10:05:03MQTT publish: topic 'zigbee2mqtt/IKEA Traedfri On-Off Button/action', payload 'on'
Debug 2023-12-29 10:05:03Received Zigbee message from 'Aqara H1 Switch', type 'attributeReport', cluster 'genOnOff', data '{"245":126896389,"onOff":1}' from endpoint 1 with groupID 0
Info 2023-12-29 10:05:03MQTT publish: topic 'zigbee2mqtt/Aqara H1 Switch', payload '{"action":null,"consumption":0.02,"current":0,"device_temperature":35,"energy":0.02,"flip_indicator_light":null,"led_disabled_night":null,"linkquality":120,"operation_mode_left":null,"operation_mode_right":null,"power":7.93,"power_outage_count":0,"power_outage_memory":null,"state":"OFF","state_left":"ON","state_right":"ON","voltage":234.07}'

Koenkk, 2024-01-04:
Duplicate of https://github.com/Koenkk/zigbee2mqtt/issues/16918 (this is an issue in the Aqara firmware which cannot be fixed from z2m)

closed 2024-01-04 by Koenkk

viharm, 2024-01-13:
@Koenkk, thanks for investigating this issue across the various issues reported here. Since Z2M is such a significant project, will Aqara take feedback from you, to look into this?

Koenkk, 2024-01-15:
@viharm I don't know (and don't have any contacts with them). I think it's unlikely 😄

eligao, 2024-02-07:
I had a same problem some time ago, 
with [Mi smart plug](https://www.zigbee2mqtt.io/devices/ZNCZ02LM.html#xiaomi-zncz02lm) + the same [IKEA E1743 TRADFRI ON/OFF switch](https://www.zigbee2mqtt.io/devices/E1743.html#ikea-e1743)
I upgraded the E1743 firmware to 24.4.6 and re-joined it from another router device (a TRADFRI bulb) and the problem was solved. Not sure which step helped though.

adynis, 2024-12-27:
I just got a firmware update on my [WS-EUK04](https://www.zigbee2mqtt.io/devices/WS-EUK04.html#aqara-ws-euk04) ; And.... I'm still testing but ... I think the issue in this ticket is solved.

I did the firmware update by an Aqara hub G3; I had no release notes shown.
Before I had version: 0.0.0_1123 (with built date in Z2M: 11-10-2021)
Now I have version: to 0.0.0_1125 (with build date in Z2M: 07-25-2024 )

I'm testing now the symfonisk remote and I have no trigger on the switch relay anymore when I press the "play" button. If I "join" the remote to another H1 switch which is not yet upgraded to the new firmware, I immediately have the issue. If I join it to the "upgraded one" , I can normally use the remote ... at least until now after ... ~30 mins of testing.

jmccrohan, 2024-12-27:
@adynis This is excellent news, thanks. Do you happen to have the URL to the WS-EUK04 firmware file for those of us without a Aqara hub?

adynis, 2024-12-27:
I have no idea how to take the URL ... I got the update through Aqara app having my Hub G3. If you know how, tell me and I try (since I only upgraded one switch in my room to test it ... before the rest of the house)

jmccrohan, 2024-12-27:
@adynis Guide here if you're able to follow it: https://www.zigbee2mqtt.io/advanced/more/tuya_xiaomi_ota_url.html

adynis, 2024-12-27:
uh... that doesn't look so easy to me :-) I can't promise anything . If someone has the environment ready maybe coud do it faster than me.

chinery, 2025-01-21:
Any chance anyone managed to get the firmware update for the Aqara switch? My WS-EUK03 has started showing the same behaviour where it will toggle the relay when acting as a router.

mosiph, 2025-06-28:
I had this issue for a year now. I had to run the Philips Hue and IKEA remotes on separate Zigbee networks to work around the issue.

Recently I upgraded all the Aqara wall switches to the latest firmware (manually through M1S hub) and pulled all the Philips Hue and IKEA remotes onto the same Zigbee2MQTT network. The issue went away. Definitely there was some fix regarding this issue in the latest Aqara FW.