Koenkk/zigbee2mqtt #20546: SWITCH-ZR02 supported but "power_on_behavior"returns "Status 'UNSUPPORTED_ATTRIBUTE' "

antunesls, 2024-01-03:
### What happened?

trying to change an exposed attribute "power_on_behavior" get an error.
see log: 

Publish 'get' 'power_on_behavior' to 'Tomada_Umidificador_Leo' failed: 'Error: Read 0x00124b0026b8130c/1 genOnOff(["startUpOnOff"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'

### What did you expect to happen?

Expected parameter update

### How to reproduce it (minimal and precise)

Go to exposed attributes and select power_on_behavior setting to some value.

### Zigbee2MQTT version

1.35.0

### Adapter firmware version

20230507

### Adapter

ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus (ZBDongle-P)

pixeldesbois, 2024-01-05:
Hello,
Happy new Year !!!
 I have the same problem with https://www.zigbee2mqtt.io/devices/SWITCH-ZR02.html
Cordially
Stéphane

antunesls, 2024-01-05:
Hi @pixeldesbois , 

Thanks, 4U2.

Please, use the Dev Branch, its already fixed there.

closed 2024-01-06 by Koenkk