Koenkk/zigbee2mqtt #20550: Issues with Moes BHT-002-GCLZB

derekoharrow, 2024-01-03:
### What happened?

I am getting two errors in the Home Assistant log file for my Moes thermostat (https://www.zigbee2mqtt.io/devices/BHT-002-GCLZB.html#moes-bht-002-gclzb).

Firstly, it is reporting the climate mode of "cool" as invalid.

Secondly, it doesn't like the state length as it's over 255 characters (it's 388)

Here's the HA log file extract:
```
homeassistant.exceptions.InvalidStateError: Invalid state with length 388. State max length is 255 characters.
2024-01-03 21:17:47.734 ERROR (MainThread) [homeassistant.components.mqtt.climate] Invalid modes mode: cool
2024-01-03 21:17:47.734 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'preset' when rendering '{{ value_json.preset }}'
2024-01-03 21:17:47.870 ERROR (MainThread) [homeassistant.components.mqtt.climate] Invalid modes mode: cool
2024-01-03 21:17:47.871 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'preset' when rendering '{{ value_json.preset }}'
2024-01-03 21:17:47.886 ERROR (MainThread) [homeassistant.helpers.entity] Failed to set state for sensor.ensuite_floor_program, fall back to unknown
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1125, in _async_write_ha_state
    hass.states.async_set(
  File "/usr/src/homeassistant/homeassistant/core.py", line 1831, in async_set
    state = State(
            ^^^^^^
  File "/usr/src/homeassistant/homeassistant/core.py", line 1408, in __init__
    validate_state(state)
  File "/usr/src/homeassistant/homeassistant/core.py", line 228, in validate_state
    raise InvalidStateError(
homeassistant.exceptions.InvalidStateError: Invalid state with length 388. State max length is 255 characters.
2024-01-03 21:17:48.001 ERROR (MainThread) [homeassistant.components.mqtt.climate] Invalid modes mode: cool
```

And here's the state as reported by Zigbee2MQTT:
```
{
    "child_lock": "UNLOCK",
    "current_heating_setpoint": 18,
    "heat": "OFF",
    "last_seen": "2024-01-03T21:21:46+00:00",
    "linkquality": 47,
    "local_temperature": 19,
    "local_temperature_calibration": 0,
    "max_temperature": 40,
    "min_temperature_limit": 5,
    "program": {
        "saturday_p1_temperature": null,
        "saturday_p2_temperature": null,
        "saturday_p3_temperature": null,
        "saturday_p4_temperature": null,
        "sunday_p1_temperature": null,
        "sunday_p2_temperature": null,
        "sunday_p3_temperature": null,
        "sunday_p4_temperature": null,
        "weekdays_p1_temperature": null,
        "weekdays_p2_temperature": null,
        "weekdays_p3_temperature": null,
        "weekdays_p4_temperature": null
    },
    "running_state": "idle",
    "sensor": "OU",
    "system_mode": "cool",
    "deadzone_temperature": null,
    "max_temperature_limit": null
}
```

### Zigbee2MQTT version

1.35.0-dev commit: e9aee4c

### Adapter firmware version

20221226

### Adapter

zzh!

derekoharrow, 2024-01-03:
Oh, and just to add, my device is a BHT-002-GB (Electric Floor Heating) and not the -GC that z2m detects it as, in case that makes any difference. Ta

zeminator, 2024-01-04:
Just noticed that all my BHT-002-GCLZB devices are getting wrong system_mode when they are report something back to Z2M... Also I will get now   "system_mode": "cool"... Thankly heating is still working but any of the visual displays can't show the real state to me, in Z2M or in HA.

Issues is most likely started on [1.35.0](https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.35.0)

osmeest, 2024-01-09:
Not sure if I should create a new ticket or if it is related but...
I noticed that local temperature calibration of BHT-002 is broken. Setting anything different than 0 makes it report a funky local temperature.

kastmgnru, 2024-01-14:
It's the same for me. And there are two identical thermostats. One of them is fine, but the other one has system_mode set to cool. Rebooting/resetting/etc. does not help.
`..........
    "sensor": "AL",
    "system_mode": "cool",
}`

DorianStrasser, 2024-01-28:
I'm chiming in here with both problems. Has anyone made any progress? Is there anything I can do to help?

zeminator, 2024-02-07:
@Koenkk can you check this... 

Also https://github.com/Koenkk/zigbee2mqtt/issues/15654

closed 2024-09-05 by github-actions[bot]

TurnrDev, 2024-10-01:
> Not sure if I should create a new ticket or if it is related but... I noticed that local temperature calibration of BHT-002 is broken. Setting a value with a decimal part makes it report a funky local temperature. And with the slider in the GUI, it is not always easy to pick an integer value.

I too have noticed issues with non-integer values. If I set it to anything non-integer it then gets set to either the int max or int min and gives gibberish values for local temperature