Koenkk/zigbee2mqtt #20551: IKEA Tradfri LED2104R3 still showing as Not supported in homeassistant addon

kamermcw, 2024-01-03:
### What happened?

I have 8 new IKEA Tradfri LED2104R3 bulbs added to homeassistant, but got the Not supported message. After some investigation I found the exact issue:
"Debug log not showing special characters on new device." #18809
It should have been solved by "Update ikea.ts with new definition for LED2104R3" #6635

### What did you expect to happen?

I upgraded to 1.35.0-1 two days ago and expected this to be solved. Instead it still reported the device Not supported.

TRADFRI bulb GU10 WW 345lm translates to \u001TRADFRI bulb GU10 WW 345lm

### Zigbee2MQTT version

1.35.0

### Adapter firmware version

7.1.1.0 build 273

### Adapter

Nabu Casa SkyConnect

### Debug log

info  2024-01-03 13:03:44: Successfully interviewed '0xbc026efffe7e6284', device has successfully been paired
warn  2024-01-03 13:03:44: Device '0xbc026efffe7e6284' with Zigbee model 'TRADFRI bulb GU10 WW 345lm' and manufacturer name 'IKEA of Sweden' is NOT supported, please follow https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html
info  2024-01-03 13:03:44: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":{"description":"Automatically generated definition","exposes":[{"features":[{"access":7,"description":"On/off state of the switch","label":"State","name":"state","property":"state","type":"binary","value_off":"OFF","value_on":"ON","value_toggle":"TOGGLE"}],"type":"switch"},{"access":1,"description":"Link quality (signal strength)","label":"Linkquality","name":"linkquality","property":"linkquality","type":"numeric","unit":"lqi","value_max":255,"value_min":0}],"model":"\u001aTRADFRI bulb GU10 WW 345lm","options":[{"access":2,"description":"State actions will also be published as 'action' when true (default false).","label":"State action","name":"state_action","property":"state_action","type":"binary","value_off":false,"value_on":true}],"supports_ota":false,"vendor":"IKEA of Sweden"},"friendly_name":"0xbc026efffe7e6284","ieee_address":"0xbc026efffe7e6284","status":"successful","supported":false},"type":"device_interview"}'
info  2024-01-03 13:03:44: Configuring '0xbc026efffe7e6284'

Koenkk, 2024-01-04:
Should be fixed now.

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

kamermcw, 2024-02-03:
Tested with version 1.35.2-1 today. Still not supported.

Zigbee2MQTT:warn  2024-02-03 15:22:11: Device '0xbc026efffe7e6284' with Zigbee model 'TRADFRI bulb GU10 WW 345lm' and manufacturer name 'IKEA of Sweden' is NOT supported, please follow https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html

Koenkk, 2024-02-03:
could you provide the data/database.db entry of your device?

kamermcw, 2024-02-03:
Is this what you mean by database entry?

{"id":30,"type":"Router","ieeeAddr":"0xbc026efffe7e6284","nwkAddr":52815,"manufId":4476,"manufName":"IKEA of Sweden","powerSource":"Mains (single phase)","modelId":"\u001aTRADFRI bulb GU10 WW 345lm","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":257,"inClusterList":[0,3,4,5,6,8,4096,64636],"outClusterList":[25],"clusters":{"genOnOff":{"attributes":{"onOff":1}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0xe0798dfffe12fe8e","endpointID":1}],"configuredReportings":[{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":65000,"repChange":1,"manufacturerCode":null}],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[33],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":1,"stackVersion":113,"hwVersion":1,"dateCode":"20230406","swBuildId":"1.0.25","zclVersion":8,"interviewCompleted":true,"meta":{"configured":1324213189},"lastSeen":1707002028920}

Koenkk, 2024-02-04:
yes, fixed it!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

closed 2024-02-04 by Koenkk

Toomblr, 2024-02-05:
I'm still experiencing the same issue. My Zigbee2MQTT version: `1.35.2-dev commit: [dfda631]`

[image]