Koenkk/zigbee2mqtt #20552: TRADFRI drivers  ICPSHC24-10EU-IL-1 and  ICPSHC24-30EU-IL-1 - not published state actions as 'action'

antti365, 2024-01-03:
### What happened?

Changing state action in the frontend does not lead to success - action states are not published.

Same problem here... https://github.com/Koenkk/zigbee2mqtt/issues/20408#issue-2057845726.

Almost the same problems are observed with the Tradfri light bulb
LED1623G12 - it can send an action once, and then not send it 3 times, especially if only the brightness changes and not on/off.

### Zigbee2MQTT version

1.34.0 commit: 56589dc

### Adapter firmware version

20210319

### Adapter

CC2652

### Debug log

debug 2024-01-03 23:31:18: Received MQTT message on 'Zigbee/bridge/request/device/options' with data '{"id":"0xb4e3f9fffe77cff9","options":{"state_action":true},"transaction":"hkm97-14"}'
info  2024-01-03 23:31:19: Changed config for device 0xb4e3f9fffe77cff9
info  2024-01-03 23:31:19: MQTT publish: topic 'Zigbee/bridge/response/device/options', payload '{"data":{"from":{"description":"xxx","legacy":false,"state_action":false},"id":"0xb4e3f9fffe77cff9","restart_required":false,"to":{"description":"xxx","legacy":false,"state_action":true}},"status":"ok","transaction":"hkm97-14"}'

debug 2024-01-03 23:31:28: Received MQTT message on 'Zigbee/driver_02/set' with data '{"state":"ON"}'
debug 2024-01-03 23:31:28: Publishing 'set' 'state' to 'driver_02'
info  2024-01-03 23:31:29: MQTT publish: topic 'Zigbee/driver_02', payload '{"brightness":0,"last_seen":1704317489043,"level_config":{"on_level":"previous"},"linkquality":81,"power_on_behavior":"off","state":"OFF","update":{"installed_version":587753009,"latest_version":304371058,"state":"idle"}}'
info  2024-01-03 23:31:29: MQTT publish: topic 'Zigbee/driver_02', payload '{"brightness":178,"last_seen":1704317489043,"level_config":{"on_level":"previous"},"linkquality":81,"power_on_behavior":"off","state":"ON","update":{"installed_version":587753009,"latest_version":304371058,"state":"idle"}}'

debug 2024-01-03 23:31:31: Received MQTT message on 'Zigbee/driver_02/set' with data '{"state":"OFF"}'
debug 2024-01-03 23:31:31: Publishing 'set' 'state' to 'driver_02'
info  2024-01-03 23:31:31: MQTT publish: topic 'Zigbee/driver_02', payload '{"brightness":178,"last_seen":1704317491617,"level_config":{"on_level":"previous"},"linkquality":78,"power_on_behavior":"off","state":"ON","update":{"installed_version":587753009,"latest_version":304371058,"state":"idle"}}'
info  2024-01-03 23:31:31: MQTT publish: topic 'Zigbee/driver_02', payload '{"brightness":178,"last_seen":1704317491617,"level_config":{"on_level":"previous"},"linkquality":78,"power_on_behavior":"off","state":"OFF","update":{"installed_version":587753009,"latest_version":304371058,"state":"idle"}}'

Koenkk, 2024-01-04:
The `state_action` is only applicable for devices which can be physically switched, e.g. a power plug with a button or a wall switch.

antti365, 2024-01-04:
> The `state_action` is only applicable for devices which can be physically switched, e.g. a power plug with a button or a wall switch.

Like this? I'm confused... Why then do some other devices, such as light bulbs, publish state action? And why then do drivers and light bulbs have the appropriate settings in the front-end settings and the state_action option is indicated in the documentation for these devices? (see screenshots)
[image]
[image]
[image]

Koenkk, 2024-01-04:
Because all on/off devices can support it, if you want this you can setup reporting for the `genOnOff`/`onOff` attribute through the z2m frontend -> device -> reporting tab.

antti365, 2024-01-04:
Thank you, now I managed to set everything up! The question is closed.

antti365, 2024-01-06:
@Koenkk Forced to open again... If the light bulb receives not only state, but also brightness at the same time, the action is not published, if only state is published. In my opinion this is not entirely correct.

debug 2024-01-06 13:43:43: Received MQTT message on 'Zigbee/bulb_01/set' with data '{"state":"ON","brightness":254}'
debug 2024-01-06 13:43:43: Publishing 'set' 'brightness' to 'bulb_01'
info  2024-01-06 13:43:45: MQTT publish: topic 'Zigbee/bulb_01', payload '{"brightness":254,"last_seen":1704541425582,"linkquality":96,"power_on_behavior":"off","state":"OFF","update":{"installed_version":587810353,"latest_version":587347505,"state":"idle"}}'
info  2024-01-06 13:43:46: MQTT publish: topic 'Zigbee/bulb_01', payload '{"brightness":254,"last_seen":1704541426069,"linkquality":126,"power_on_behavior":"off","state":"OFF","update":{"installed_version":587810353,"latest_version":587347505,"state":"idle"}}'
info  2024-01-06 13:43:46: MQTT publish: topic 'Zigbee/bulb_01', payload '{"brightness":254,"last_seen":1704541426069,"linkquality":126,"power_on_behavior":"off","state":"ON","update":{"installed_version":587810353,"latest_version":587347505,"state":"idle"}}'
debug 2024-01-06 13:53:43: Received MQTT message on 'Zigbee/bulb_01/set' with data '{"state":"OFF"}'
debug 2024-01-06 13:53:43: Publishing 'set' 'state' to 'bulb_01'
info  2024-01-06 13:53:43: MQTT publish: topic 'Zigbee/bulb_01', payload '{"brightness":254,"last_seen":1704542023991,"linkquality":27,"power_on_behavior":"off","state":"ON","update":{"installed_version":587810353,"latest_version":587347505,"state":"idle"}}'
info  2024-01-06 13:53:43: MQTT publish: topic 'Zigbee/bulb_01', payload '{"brightness":254,"last_seen":1704542023991,"linkquality":27,"power_on_behavior":"off","state":"OFF","update":{"installed_version":587810353,"latest_version":587347505,"state":"idle"}}'
debug 2024-01-06 13:53:44: Received Zigbee message from 'bulb_01', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
info  2024-01-06 13:53:44: MQTT publish: topic 'Zigbee/bulb_01', payload '{"action":"off","brightness":254,"last_seen":1704542024005,"linkquality":27,"power_on_behavior":"off","state":"OFF","update":{"installed_version":587810353,"latest_version":587347505,"state":"idle"}}'
debug 2024-01-06 14:16:40: Received Zigbee message from 'bulb_01', type 'attributeReport', cluster 'genLevelCtrl', data '{"currentLevel":254}' from endpoint 1 with groupID 0
info  2024-01-06 14:16:40: MQTT publish: topic 'Zigbee/bulb_01', payload '{"brightness":254,"last_seen":1704543400462,"linkquality":33,"power_on_behavior":"off","state":"OFF","update":{"installed_version":587810353,"latest_version":587347505,"state":"idle"}}'

closed 2024-08-04 by github-actions[bot]