Koenkk/zigbee2mqtt #20553: ADEO indoor siren LDSENK06

phanou9961, 2024-01-03:
### Link

https://www.leroymerlin.fr/produits/electricite-et-domotique/alarme-telesurveillance/alarme-maison/sirene-alarme/sirene-alarme-interieure-sur-batterie-100-db-lexman-85170909.html

### Database entry

{"id":9,"type":"Router","ieeeAddr":"0x5c0272fffed418f0","nwkAddr":59314,"manufId":4727,"manufName":"ADEO","powerSource":"DC Source","modelId":"LDSENK06","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":1027,"inClusterList":[0,3,1280,1282,2821],"outClusterList":[3,25],"clusters":{"ssIasZone":{"attributes":{"iasCieAddr":"0xe0798dfffeb31d40","zoneState":1}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":41,"stackVersion":6,"hwVersion":2,"dateCode":"20200430-15","swBuildId":"2.09","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1704324875833,"defaultSendRequestWhen":"immediate"}

### Comments

Hi everyone, 

I've tried to add manually with the converters this devices and trying to follow the documentation but with no success. 
If someone can give me some help. 

thanks in advance.

### External defintion

```shell
const {identify} = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
    zigbeeModel: ['LDSENK06'],
    model: 'LDSENK06',
    vendor: 'ADEO',
    description: 'Automatically generated definition',
    extend: [identify()],
};

module.exports = definition;
```

gapiob, 2024-01-22:
Hi,
I try to add this js :
`const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['LDSENK06'], // The model ID from: Device with modelID 'lumi.sens' is not supported.
    model: 'LDSENK06', // Vendor model number, look on the device for a model number
    vendor: 'ADEO', // Vendor of the device (only used for documentation and startup logging)
    description: 'ENKI LEXMAN Wireless smart indoor siren 1', // Description of the device, copy from vendor site. (only used for documentation and startup logging)
    fromZigbee: [fz.ias_siren],
        toZigbee: [tz.warning],        
        exposes: [e.warning()],
        configure: async (device, coordinatorEndpoint, logger) => {
            device.defaultSendRequestWhen = 'immediate';
            device.save();
            await device.getEndpoint(1).unbind('genPollCtrl', coordinatorEndpoint);
        },
};`

but it's only partialy work.
commandes are take into account, but after a little time out, alarm stop, and an error message is raised. all others component works perfecly.

`Debug 2024-01-22 15:33:31Received MQTT message on 'zigbee2mqtt/Sirene 1/set' with data '{"warning":{"mode":"fire"}}'
Debug 2024-01-22 15:33:31Publishing 'set' 'warning' to 'Sirene 1'
Error 2024-01-22 15:33:51Publish 'set' 'warning' to 'Sirene 1' failed: 'Error: Command 0x04cd15fffe602f63/1 ssIasWd.startWarning({"startwarninginfo":37,"warningduration":10,"strobedutycycle":0,"strobelevel":1}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 12973 - 1 - 4 - 1282 - 11 after 10000ms)'
Debug 2024-01-22 15:33:51Error: Command 0x04cd15fffe602f63/1 ssIasWd.startWarning({"startwarninginfo":37,"warningduration":10,"strobedutycycle":0,"strobelevel":1}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 12973 - 1 - 4 - 1282 - 11 after 10000ms) at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35) at listOnTimeout (node:internal/timers:569:17) at processTimers (node:internal/timers:512:7)`

phanou9961, 2024-02-01:
Thanks for your work, I'll try this and see how it's working by me side and let you know.

forever80-dev, 2024-03-01:
Hello,

I'm also interested too. I see the same latency problems.

Sincerely

closed 2024-09-28 by github-actions[bot]