Koenkk/zigbee2mqtt #20558: Future incompatibility with HA core  2024.2.0

robitsrl, 2024-01-04:
### What happened?

I have several device using Zigbee2MQTT that are using Entity category `config`.
this will be deprecated in the future HA core 2024.2.0 as read in the HA logs:

Entity category `config` is invalid for sensors, ignoring. This stops working from HA Core 2024.2.0

### What did you expect to happen?

NEW RELEASE FOR ZIGBBETOMQTT

### Zigbee2MQTT version

1.35.0-1

### Adapter firmware version

N/A

### Adapter

Silicon Labs Multiprotocol ITEAD_SONOFF_Zigbee_3.0_USB_Dongle_Plus_V2

Koenkk, 2024-01-04:
For what sensor do you get these/

robitsrl, 2024-01-04:
Aquara Xiaomi WXKG15LM
SONOFF SNZB-06P
SONOFF SNZB-06P

Koenkk, 2024-01-04:
What are the errors in the HA log?

bernardesarthur, 2024-01-05:
From what I see, this error is the HA log:

`Entity category config is invalid for sensors, ignoring. This stops working from HA Core 2024.2.0
`

robitsrl, 2024-01-05:
Yes I confirm the error is:

Entity category config is invalid for sensors, ignoring. This stops working from HA Core 2024.2.0 

I wrote to MQTT developer and they told me to use logger component to find out what sensor.
I do not install logger for now but i have download MQTT diagnostic and post it to MQTT developer and they confirm that the cause come from those sensor integrations

bckthomas, 2024-02-09:
Hi 2024.2.0 is out and I have sensors with Entity category `config` :

- Ikea TRADFRI control outlet (E1603/E1702/E1708)
- Lidl Silvercrest smart plug (EU, CH, FR, BS, DK) (HG06337)
- MHCOZY 4 channel relay (TYWB 4ch-RF)

closed 2024-09-08 by github-actions[bot]