Koenkk/zigbee2mqtt #20560: Some effects no longer work

OUARZA, 2024-01-04:
### What happened?

Although the effects are present in the Zigbee2MQTT drop-down menu, they have no effect. This is the case for candle, fireplace, colorloop. They have already worked.
Is this linked to the transition to V2 of the Philips API?
The OKAY, Breathe effect works.
There are no mistakes.
https://www.zigbee2mqtt.io/devices/9290012573A.html#philips-9290012573a
https://www.zigbee2mqtt.io/devices/915005106701.html#philips-915005106701

### What did you expect to happen?

I expect the effects offered by Philips to work.

### How to reproduce it (minimal and precise)

By selecting an effect from the drop-down menu.

### Zigbee2MQTT version

 1.34.0 commit: 56589dc

### Adapter firmware version

0x26780700

### Adapter

ConBee2/RaspBee2

### Debug log

Log made on equipment 0x00178801102e455a
Test done with the Candle and Colorloop effect which do not work and the breathe effect which does work.

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/13827612/log.txt)

OUARZA, 2024-05-18:
Hello, I still have the problem. Do you have an idea ?

glassbase, 2024-11-08:
Colorloop does not work on my Philips 7299355PH light strip and 9290012573A bulbs. It worked on ZHA (I just recently migrated to z2m)

posixx, 2024-12-21:
Same here. Some of my lily spots do not work with colorloop effect (1741830P7 and 1746230P7). The strange thing is: the 1742830P7 (newer model) does work!

closed 2025-02-27 by github-actions[bot]