Koenkk/zigbee2mqtt #20563: frient EMIZB-141 divisor

Dommoe, 2024-01-04:
### What happened?

Hello everyone,

I buy a frient EMIZB-141.
So far, it works on my electricity meter as well. I only have the problem that a standard electricity meter counts 1000 pulses. But my electricity meter has 10000 pulses. I've read that you have to change the divisor. Unfortunately, I can't find this in the settings. When I try to change the manual under the developer console, I get the error message that the divisor can only be read. Can I change the divisor manually?

Sorry for my English 
[image]

Thank you in advance

### Zigbee2MQTT version

1.35.0

### Adapter firmware version

20230507

### Adapter

 zStack3x0

joschaschultze, 2024-01-04:
Its under "details" the "Pulse Configuration". You have to change this value to 10.000.

Dommoe, 2024-01-04:
If I set this value to 10000, then I get values around 400000 watts. It also says that it is a percentage value. I have now entered -90, then the values seem to be approximately correct. 

[image]

The energy sensor is also not working properly. But I don't know exactly why. If I change there in the settings, it also counts too fast. 

[image]

MaxD16, 2024-01-09:
Hello,

I have the same exact issue. I can not find the option to change pulses to 10.000 pulses/kW.

FrankTolksdorf, 2024-01-10:
have the same exact issue. I can not find the option to change pulses to 10.000 pulses/kW.

thedotor, 2024-01-11:
I have added this device today, running Zigbee2MQTT 1.35.1-1 and missing the "Pulse Configuration" mine needs to be 4000 pulses/kW.
Thank you

closed 2024-09-15 by github-actions[bot]