Koenkk/zigbee2mqtt #20566: Zigbee HA error zigbee2mqtt_bridge_*

william-sy, 2024-01-04:
### What happened?

Warnings during startup of HA

### What did you expect to happen?

No warnings 

### How to reproduce it (minimal and precise)

Not sure, Just reboot?  seems like every one might have them :) 

### Zigbee2MQTT version

1.35.0

### Adapter firmware version

20230507

### Adapter

sonoff P

### Debug log

```
2024-01-04 11:48:05.562 WARNING (Recorder) [homeassistant.components.recorder.db_schema] State attributes for switch.zigbee2mqtt_bridge_permit_join exceed maximum size of 16384 bytes. This can cause database performance issues; Attributes will not be stored
2024-01-04 11:48:05.562 WARNING (Recorder) [homeassistant.components.recorder.db_schema] State attributes for select.zigbee2mqtt_bridge_log_level exceed maximum size of 16384 bytes. This can cause database performance issues; Attributes will not be stored
2024-01-04 11:48:05.563 WARNING (Recorder) [homeassistant.components.recorder.db_schema] State attributes for sensor.zigbee2mqtt_bridge_permit_join_timeout exceed maximum size of 16384 bytes. This can cause database performance issues; Attributes will not be stored
2024-01-04 11:48:05.563 WARNING (Recorder) [homeassistant.components.recorder.db_schema] State attributes for sensor.zigbee2mqtt_bridge_version exceed maximum size of 16384 bytes. This can cause database performance issues; Attributes will not be stored
2024-01-04 11:48:05.966 WARNING (Recorder) [homeassistant.components.recorder.db_schema] State attributes for switch.zigbee2mqtt_bridge_permit_join exceed maximum size of 16384 bytes. This can cause database performance issues; Attributes will not be stored
2024-01-04 11:48:05.966 WARNING (Recorder) [homeassistant.components.recorder.db_schema] State attributes for select.zigbee2mqtt_bridge_log_level exceed maximum size of 16384 bytes. This can cause database performance issues; Attributes will not be stored
2024-01-04 11:48:05.966 WARNING (Recorder) [homeassistant.components.recorder.db_schema] State attributes for sensor.zigbee2mqtt_bridge_permit_join_timeout exceed maximum size of 16384 bytes. This can cause database performance issues; Attributes will not be stored
2024-01-04 11:48:05.967 WARNING (Recorder) [homeassistant.components.recorder.db_schema] State attributes for sensor.zigbee2mqtt_bridge_version exceed maximum size of 16384 bytes. This can cause database performance issues; Attributes will not be stored
```

julianfs, 2024-01-04:
I also see these log entries, but haven't experienced any issues so far.

duczz, 2024-01-04:
me too, see https://github.com/Koenkk/zigbee2mqtt/issues/20488

william-sy, 2024-01-04:
Ah yes duplicate, fair enough. 

Did set `legacy_entity_attributes: false` , stopped z2m & ha, cleared mqtt and started the containers again. errors gone

closed 2024-01-04 by william-sy