Koenkk/zigbee2mqtt #20569: Zigbee2MQTT can receive commands from ZBMINIL2 but cannot send

cdemi, 2024-01-04:
### What happened?

I have about 3 ZBMINIL2 in various places and they have been up and running for almost a year, I think. Recently, and one by one (not all at once), they have started showing as Offline and whenever I send a command to them, they all time out:

```
Publish 'set' 'state' to 'Main Bathroom Lights' failed: 'Error: Command 0xf082c0fffe1304b0/1 genOnOff.off({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 65462 - 1 - 104 - 6 - 11 after 10000ms)'

Publish 'get' 'state' to 'Main Bathroom Lights' failed: 'Error: Read 0xf082c0fffe1304b0/1 genOnOff(["onOff"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 65462 - 1 - 105 - 6 - 1 after 10000ms)'
```

If I press the physical switch, Zigbee2MQTT, receives the message and correctly updates the lqi (142) and  state (and shows them as online) but I cannot turn it on or off from Zigbee2MQTT.

### Zigbee2MQTT version

1.35.0

### Adapter firmware version

20221226

### Adapter

Electrolama zig-a-zig-ah! (zzh!)

Thrasher2020, 2024-01-04:
I would roll back to 1.34.0 and wait until all the issues in 1.35.0 have been ironed out, I had similar issues with devices dropping off/extreme latency etc. Reverted and all back on.

marazmarci, 2024-01-05:
I own a ZBMINI-L (not L2), and it exhibited similar behavior several times in the last year. I was able to solve the problem with the following steps:
1. Enable pairing in Z2M
2. Cut the power from the Sonoff ZBMINI-L (eg. by turning off the breaker)
3. Wait for 5-10 seconds
4. Turn its power back on

But first, I'd narrow down if the issue is caused by upgrading to 1.35.0 (by downgrading to 1.34.0), or if it is just a random coincidence that your devices started behaving weirdly at the same time with the 1.35.0 upgrade.

Neibafsama, 2024-01-15:
I also have the same problem with my ZBmini-L2 device.
I receive the data well _(state when the wall switch is on/off, quality of the link)_, but impossible to control it remotely.

I receive the message:
_" Publish 'set' 'state' to 'Zbmini_l2' failed: 'Error: Command 0x9035eafffe0fed98/1 genOnOff.on({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 25079 - 1 - 100 - 6 - 11 after 10000ms)' "_

salopette, 2024-01-23:
same problem too.

msaphoto06, 2024-03-26:
Same problem

cazxtc, 2024-06-10:
Same problem

closed 2024-12-31 by github-actions[bot]