Koenkk/zigbee2mqtt #20575: Conbee3 and Aqara WS-EUK03/WS-EUK04 - missing attributes

vezyr, 2024-01-04:
### What happened?

I have several Aqara switches, WS-EUK03 and WS-EUK04 models. After changing the coordinator from Conbee 2 to Conbee 3 and reconfiguring (repairing) all devices, I noticed that some attributes are no longer being reported.

Attributes:
- power
- energy
- device temperature
- power outage count

are missing (null), for example:

```
MQTT publish: topic 'zigbee2mqtt/room_1_light_main', payload '{"action":null,"device_temperature":null,"energy":null,"flip_indicator_light":"ON","led_disabled_night":false,"linkquality":190,"operation_mode":"control_relay","power":null,"power_outage_count":null,"power_outage_memory":false,"state":"ON","update":{"installed_version":2839,"latest_version":2839,"state":"idle"},"update_available":false}'
```

After enabling debug log, I have noticed information about missing converters for these switches, for example:
```
No converter available for 'WS-EUK03' with cluster 'seMetering' and type 'attributeReport' and data '{"currentSummDelivered":[0,1]}'
No converter available for 'WS-EUK03' with cluster 'haElectricalMeasurement' and type 'attributeReport' and data '{"activePower":0}'
No converter available for 'WS-EUK04' with cluster 'seMetering' and type 'attributeReport' and data '{"currentSummDelivered":[0,33]}'
```

### What did you expect to happen?

Devices should report the state of attributes.

### Zigbee2MQTT version

1.35.0

### Adapter firmware version

0x264e0900

### Adapter

ConBee3

Koenkk, 2024-01-04:
@chrishae I think the Conbee 3 fw is missing a hack for Xiaomi devices (the Conbee 2 fw has it).  The trick is that reply with the xiaomi manufacturerCode on interview. [Zstack version of this hack](https://github.com/Koenkk/Z-Stack-firmware/blob/4e888d3310eaf24c19699b00a29e24d512e49e8c/coordinator/Z-Stack_3.x.0/firmware.patch#L678)

ChrisHae, 2024-01-05:
Correct. There are a bunch of hacks and fixes (from conbee2 fw) that will come to the conbee3 firmware in the next releases.

vezyr, 2024-01-13:
I've checked the newest firmware for Conbee 3 (from yesterday, ConBeeIII_0x264f0900). Now it's even worse. Theoretically my WS-EUK04 paired and configured correctly, but after a few seconds, the lights turned off, and WS-EUK04 stopped working in zigbee2mqtt (I can't change the state, etc). Additionally, the LED indicators on the device started flashing very quickly (a few times per second).
I'll check out the next versions as they come out.

thk-socal, 2024-01-13:
Agree...my mostly stable Conbee III Z2M network has gone to crap with the 0x264f0900 firmware change. When I query my devices for status or other things the Inovelli switches will reboot. Major issues with new firmware.

DAVIZINH0, 2024-01-15:
Hello, I bought yesterday a conbee3 and my intention is migrate from conbee2 to a 3 the next days. but i find this topic and maybe i will wait :-) thanks for the report!!!!!

ChrisHae, 2024-01-15:
Yes, the problems are known and still beeing worked on. I'm confident that they are beeing fixed in a short time.

DAVIZINH0, 2024-01-16:
> Yes, the problems are known and still beeing worked on. I'm confident that they are beeing fixed in a short time.

where can i follow the progress with the firmware? or see the issues?, exist a github from the conbee firmware? or something similar?

thanks a lot!

ChrisHae, 2024-01-17:
@DAVIZINH0 Not yet. The developer currently thinks about how to deploy this. Most likey we will open a new issue in an existing repository. I will update you.

AndrisJrs, 2024-01-26:
I am having exactly the same issue on Sonoff Zigbee P dongle.

vezyr, 2024-02-09:
I've updated Conbee3 to the newest firmware version (0x26500900, from yesterday). I've tested WS-EUK04 and connection to the Conbee3 is stable again.
Unfortunately the issue with missing attributes' values is still present, but now there aren't any "No converter available" entries at all.

thk-socal, 2024-02-11:
Let me pull the latest firmware and try my other issues. Will report back.

thk-socal, 2024-02-11:
Latest firmware is significantly improved for my network. So far only Inovelli and Hue devices. (0x2650900)

DAVIZINH0, 2024-03-17:
Hello,
I finally migrate mi network from conbee 2 to conbee 3.
First 24 hours is aparently well and better in coverage with the conbee 3.
BUT after this 24 hours, some aqara devices apears offline, temperature sensors, movement, buttons. All aqara (xiaomi)

In one post someone tell that the conbee3 dont have the aqara hacks to solve this problem in the firmware.
Its posible to report this to conbee staff???. if this is solved in the previous devices, maybe this will be easy.
With this problem, the conbee3 its unsable for me :-(

ChrisHae, 2024-03-18:
> Hello, I finally migrate mi network from conbee 2 to conbee 3. First 24 hours is aparently well and better in coverage with the conbee 3. BUT after this 24 hours, some aqara devices apears offline, [...]

Wich FW version of ConBee3 do you use? As I know there where some fixes regarding Aqara devices in the newest FW (26500900)

https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Update-deCONZ-manually

DAVIZINH0, 2024-03-18:
> Wich FW version of ConBee3 do you use? As I know there where some fixes regarding Aqara devices in the newest FW (26500900) [...]

Hello, and thanks for reply

Yes, my fw version is the Last (26500900) 8 of february 24.

Aparently all works fine in the fkrst 24hours. But yesterday I lost 8 devices (aqara thermometers and button). And not apears after a Bull reboot of HA.
Maybe this night i put the conbee2 again to test if the device apears.
Or Maybe o will repair this 8 devices.

I need some time this week to think o it :-)

Jopyth, 2024-04-07:
Does anyone know whether the missing Xiaomi hacks are now implemented in the Firmware from 08-Feb-2024? (There is no mention regarding such fix in [the wiki](https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Firmware-Changelog#conbee-iii-silabs).)

I have tried finding details on this, because I experienced a problem that is similar to the one described [here](https://github.com/zigpy/zha-device-handlers/issues/2630) (I was using ZHA with Conbee III before). Now I am switched to Zigbee2MQTT and I saw comments on issues with other zigbee coordinators (namely [this issue](https://github.com/Koenkk/zigbee2mqtt/discussions/16218) which itself links to [that one](https://github.com/Koenkk/zigbee2mqtt/discussions/16573#discussioncomment-4920977)) that made me believe the problem with the switch described in that issue comes from pairing with the (missing) Xiaomi manufacturer code.

DAVIZINH0, 2024-04-07:
> Does anyone know whether the missing Xiaomi hacks are now implemented in the Firmware from 08-Feb-2024? (There is no mention regarding such fix in [the wiki](https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Firmware-Changelog#conbee-iii-silabs).)
> 
> I have tried finding details on this, because I experienced a problem that is similar to the one described [here](https://github.com/zigpy/zha-device-handlers/issues/2630) (I was using ZHA with Conbee III before). Now I switched to Zigbee2MQTT and saw comments on issues with other zigbee coordinators (namely [this issue](https://github.com/Koenkk/zigbee2mqtt/discussions/16218) which itself links to [that one](https://github.com/Koenkk/zigbee2mqtt/discussions/16573#discussioncomment-4920977)) that made me believe the problem with the switch described in that issue comes from (not) pairing with the Xiaomi manufacturer code.

in this post you can see a developer of conbee 3 firmware.
https://forum.phoscon.de/t/issues-after-migration-to-conbee-iii/4416/47
you can ask to him

closed 2024-11-04 by github-actions[bot]